Sloeber/arduino-eclipse-plugin #1468: The compilation of SoftwareSerial.h Library of esp32 failed.

xiezhoubin, 2022-05-19:
I can compile successfully in Arduino ide.
[image]

xiezhoubin, 2022-05-19:
However, an error will be reported in sloeber, indicating that the library file cannot be found. The library path has been added correctly, because I have tried other libraries in the library path and can compile successfully, but the library "SerialwareSerial.h" cannot.
The link to this library is：https://github.com/dok-net/espsoftwareserial
[image]
[image]

jantje, 2022-05-19:
There are changes in this area in the nightly. Please try with the current nightly which you can find here
https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/2287689867

xiezhoubin, 2022-05-20:
What does this operation mean? I don't quite understand. Can you solve the compilation error problem?

jantje, 2022-05-20:
Did you check wether softwareserial is in the libraries folder in your project?
If not; did you try to add it via the menu: add library to project?

closed 2022-05-21 by xiezhoubin