Sloeber/arduino-eclipse-plugin #1469: Esp32 compilation error

xiezhoubin, 2022-05-20:
[image]
[image]

xiezhoubin, 2022-05-20:
[image]

jantje, 2022-05-20:
There is nothing I can do about download issues. If you do not have the json file (because of failed download or failed processing) you are likely to experience path issues as on the first image you provided

Some textual description on what you did and what the problem seems to be like would be great.
Without that I can only do guesses.
This is why we added the content below to the new issue template:

**Before creating a bug report check for existing issues**  
Make sure you are on the latest and greatest Sloeber version. We do not support older version.  
Here is a list of [known limitations](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue++label%3A%22status%3A+known+limitation%22+)   
Here is a list of [common issues ](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue+label%3A%22Hot+issue%22+)   
Here is the list of all [open issues](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue+is%3Aopen)  
Here is a list of issues that have been [fixed but are not yet available in a release](https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=label%3A%22status%3A+fixed+in+nightly%22+)  

---

**Describe your environment**   
Do you use the Sloeber product or plugin?   
Sloeber version number  
OS: [e.g. windows 10]   
The url of the json that you used to download the platform  
A image of the project properties arduino showing the board settings   

**Describe the bug**   
A clear and concise description of what the bug is.

**To Reproduce**   
Steps to reproduce the behavior:   
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

provide the name of the board and the platform version you are using.  
Provide the json url and a screenshot of project properties->arduino  

**Expected behavior**   
A clear and concise description of what you expected to happen.  

**Data to back up your claim**   
If applicable, add screenshots, console output and/or logs to help explain your problem.   

**Additional context**   
Add any other context about the problem here.

closed 2022-05-21 by xiezhoubin