Sloeber/arduino-eclipse-plugin #1471: Failed to compile after renaming the project folder name!

xiezhoubin, 2022-05-21:
Before change.
[image]

After change.
[image]

Change the name of the ino file to match the name of the project folder.
[image]

Re open the software, but there is still a compilation error.
[image]

xiezhoubin, 2022-05-21:
ide version
[image]

jantje, 2022-05-21:
did you go to project properties->sloeber and select apply and close?

xiezhoubin, 2022-05-21:
Still report an error. You try that too?
[image]
[image]

xiezhoubin, 2022-05-21:
I tried to change the name of the project folder back to the original name, and the compilation can pass again.

closed 2022-05-23 by xiezhoubin