Sloeber/arduino-eclipse-plugin #1472: There is no "Color Theme" option.

xiezhoubin, 2022-05-21:
[image]

jantje, 2022-05-23:
As far as I remember sloeber never came with a color theme.
So I do not see what is wrong/wanted.
But maybe this:
[image]

erdemuncuoglu, 2022-05-23:
There is an eclipse color theme  plugin but I never 100% satisfied of it. I messed my Eclipse's visual settings and only recovered by removing it recently. I even used to gave more than one chance in the past with no luck. It is probably because of "me turned to much knobs" there but anyway I stick to Dark theme right now.

Here is the link if anyone interested. [Eclipse Color Themes](http://www.eclipsecolorthemes.org/)

jantje, 2022-05-23:
It is not my area of expertise: but IMHO the color themes plugin has become a part of eclipse itself.
By adding the old color theme plugin to an eclipse that has it integrated you are bound for a nightmare.
My general advice for installing eclipse plugins: try to do without them ( do this for each install because things change all the time) if you fully miss functionality check for a plugin.
If you partially miss functionality consider whether you want to invest into the "add a plugin route" or live with it till it makes your installed package.