Sloeber/arduino-eclipse-plugin #1473: If I want to add "build_flags", what should I do? just like in platformio.

xiezhoubin, 2022-05-21:
[image]
[image]
[image]

xiezhoubin, 2022-05-22:
I seem to have found it, but how can I add multiple "build_flags"?

[image]

xiezhoubin, 2022-05-22:
If I do this, I will save an error. What's the reason?
[image]
[image]

erdemuncuoglu, 2022-05-22:
If this helps...
[image]

xiezhoubin, 2022-05-22:
> 如果这有帮助... [image]

I have found it, but how can I add multiple "build_flags"?

erdemuncuoglu, 2022-05-22:
Like doing on command line. 
```-D TEST_FLAG -D OTHER_FLAG```

xiezhoubin, 2022-05-22:
> 就像在命令行上做的一样。 `-D TEST_FLAG -D OTHER_FLAG`

Single can.
[image]

Two will report a lot of mistakes.
[image]

erdemuncuoglu, 2022-05-22:
```-D a=1``` may conflict with default ones. Try this? ```-D _a=1``` or any longer name.

xiezhoubin, 2022-05-22:
> `-D a=1`可能与默认的冲突。试试这个？`-D _a=1`或任何其他名称。

succeed. Thank you.

xiezhoubin, 2022-05-22:
> 如果我这样做，我将保存一个错误。什么原因？ [image] [image]

Can this problem be solved?

xiezhoubin, 2022-05-22:
[image]
[image]

jantje, 2022-05-22:
>Can this problem be solved?

only if we get a java stacktrace or are able to reproduce the issue.

xiezhoubin, 2022-05-22:
> [image] [image]

If "testa==1" can be saved.
[image]

xiezhoubin, 2022-05-22:
> > 这个问题能解决吗？
> 
> 仅当我们获得 java stacktrace 或能够重现该问题时。

You can try. It's easy to try to reproduce.

jantje, 2022-05-23:
As I have never seen this I would say it is not easy to reproduce
You can try to provide the stacktrace, it is easy, it should be in the eclipse log file.
https://wiki.eclipse.org/FAQ_Where_can_I_find_that_elusive_.log_file%3F

erdemuncuoglu, 2022-05-23:
BTW I tried it with ESP8266 and it ran smoothly even I could see the warning message in console.
Platform: 
 - Eclipse  : 4.23.0.I20220308-0310
 - Sloeber : nightly plugin 4.4.1.202205080021

closed 2022-05-23 by xiezhoubin

jantje, 2022-05-23:
As this is a open source project only supported by volunteers I would really appreciate you to get the stacktrace.
This way:
1) We improve the quality of sloeber
2) other people do not have to go through this 
3) I do not have to go through another issue.

I have been helping you so I would appreciate you helping me by trying to provide a stack trace so I can investigate/fix the issue at hand.