Sloeber/arduino-eclipse-plugin #1476: There is a problem with the ROM size occupied by the esp8266 / esp32 calculation program.

xiezhoubin, 2022-05-22:
Will exceed the actual size.

[image]
[image]

xiezhoubin, 2022-05-22:
[image]
[image]

xiezhoubin, 2022-05-23:
Can this problem be fixed?

jantje, 2022-05-23:
This arduino way size calculation is pretty new and only tested with avr boards.
As it involves some higher levels of magic -and so does esp stuff- I'm actually surprised the results are so close.
I'll see if I can reproduce and fix.

xiezhoubin, 2022-05-23:
I hope you can fix this bug. Thank you very much.

jantje, 2022-05-24:
I looked into this issue and it seems the esp8266 does something which I do not understand.
To get the size the arduino way Sloeber runs the command 
${recipe.size.pattern}
The output of this command is parsed with the regular expressions (in a awk file) 
${recipe.size.regex}
${recipe.size.regex.data}
${recipe.size.regex.eeprom}
When I ran a test with the board below and the default ino sketch
[image]
${recipe.size.pattern} resulted into 
`"C:\eclipse\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-size" -A "L:\test\runtime-EclipseApplication_nl\test\Release/test.elf" 
`
When running that command I get following result.
[image]

The first field in the result is calculated with the regex ${recipe.size.regex} this regex is defined in platform.txt as
recipe.size.regex=^(?:\.irom0\.text|\.text|\.text1|\.data|\.rodata|)\s+([0-9]+).*
Sloeber interprets this as: for "all the lines that hold .irom0.text or .text or .text1 or .data or .rodata and one or more numbers" sum the second column
If you look at the output of the command you'll see that .irom0.text is 231500 which is already way bigger than the value Arduino IDE provides.
I tried to look how I could get the 10644 reported by Arduino IDE but I did not find a way.

So basically I do not know how to calculate the values. Which fields are needed and how do I do the calculation with these fields?

Please provide input on how to calculate the required values

PS sloeber regular expression behaviour is different from Arduino IDE behaviour as Sloeber uses awk and I assume Arduino IDE uses java regex
PPS I also notice arduino IDE does not provide a second line. I assume that is because ${recipe.size.regex.eeprom} is not defined.

jantje, 2022-05-25:
oeps problem in previus post. Arduino IDE says 260089 not 10644 
If I add  .irom0.text .text .text1 .data  .rodata I get 260089 
So something is wrong when converting from java regex to awk

jantje, 2022-05-25:
I have found the root cause 😄 

### Workaround
1) Replace the following line in the platform.txt
`recipe.size.regex=^(?:\.irom0\.text|\.text|\.text1|\.data|\.rodata|)\s+([0-9]+).*`
with
`recipe.size.regex=^(?:\.irom0\.text|\.text|\.text1|\.data|\.rodata)\s+([0-9]+).*`

2) delete the platform.sloeber.txt file

3) delete the build folders of all the project configs (default is Release) that need the workaround

### Explanation
The | means or and the | at the end of the list is apparently ignored by java regex but the sloeber converted AWK command results in "all lines are good lines".
As the output already contains the total the result is 2 times total

xiezhoubin, 2022-05-25:
> 我找到了根本原因😄
> 
> ### 解决方法
> 1. 将 platform.txt 中的以下行替换
>    `recipe.size.regex=^(?:\.irom0\.text|\.text|\.text1|\.data|\.rodata|)\s+([0-9]+).*`
>    为
>    `recipe.size.regex=^(?:\.irom0\.text|\.text|\.text1|\.data|\.rodata)\s+([0-9]+).*`
> 2. 删除 platform.sloeber.txt 文件
> 3. 删除需要解决方法的所有项目配置的构建文件夹（默认为 Release）
> 
> ### 解释
> 该| 表示或和| java regex 显然忽略了列表末尾的内容，但 sloeber 转换的 AWK 命令导致“所有行都是好行”。 由于输出已经包含总数，因此结果是总数的 2 倍

This method is feasible. Will awk be updated in the next version so that there is no need to modify the package file, otherwise it is still a little troublesome.
[image]
[image]

jantje, 2022-05-25:
I already made a fix but it needs to be tested before I release it and as there is not automated test for this .....
So it is comming

xiezhoubin, 2022-05-25:
> I already made a fix but it needs to be tested before I release it and as there is not automated test for this ..... So it is comming

OK, esp32 also has such a calculation error and needs to be repaired.

jantje, 2022-05-28:
Though the build fails this should be fixed.
I was not aware the test I changed was part of the regression testing and now it is build to fail when things change so I can see only the things changed I wanted to change :-(

jantje, 2022-05-30:
Should be fixed in the nightly

closed 2022-05-30 by jantje