Sloeber/arduino-eclipse-plugin #1479: An error occurs when formatting and saving.

xiezhoubin, 2022-05-24:
[image]

Exception Stack Trace:
org.eclipse.jface.text.BadLocationException
	at org.eclipse.jface.text.TreeLineTracker.fail(TreeLineTracker.java:1054)
	at org.eclipse.jface.text.TreeLineTracker.lineByOffset(TreeLineTracker.java:268)
	at org.eclipse.jface.text.TreeLineTracker.getLineNumberOfOffset(TreeLineTracker.java:1122)
	at org.eclipse.jface.text.AbstractLineTracker.getLineNumberOfOffset(AbstractLineTracker.java:155)
	at org.eclipse.jface.text.AbstractDocument.getLineOfOffset(AbstractDocument.java:872)
	at org.eclipse.core.internal.filebuffers.SynchronizableDocument.getLineOfOffset(SynchronizableDocument.java:334)
	at org.eclipse.cdt.internal.ui.editor.CDocumentProvider.createTrailingWhitespaceEdit(CDocumentProvider.java:1065)
	at org.eclipse.cdt.internal.ui.editor.CDocumentProvider.performSaveActions(CDocumentProvider.java:993)
	at org.eclipse.cdt.internal.ui.editor.CDocumentProvider.commitWorkingCopy(CDocumentProvider.java:892)
	at org.eclipse.cdt.internal.ui.editor.CDocumentProvider$2.execute(CDocumentProvider.java:924)
	at org.eclipse.ui.editors.text.TextFileDocumentProvider$DocumentProviderOperation.run(TextFileDocumentProvider.java:133)
	at org.eclipse.ui.actions.WorkspaceModifyDelegatingOperation.execute(WorkspaceModifyDelegatingOperation.java:71)
	at org.eclipse.ui.actions.WorkspaceModifyOperation.lambda$0(WorkspaceModifyOperation.java:110)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2315)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2340)
	at org.eclipse.ui.actions.WorkspaceModifyOperation.run(WorkspaceModifyOperation.java:131)
	at org.eclipse.ui.internal.editors.text.WorkspaceOperationRunner.run(WorkspaceOperationRunner.java:76)
	at org.eclipse.ui.internal.editors.text.WorkspaceOperationRunner.run(WorkspaceOperationRunner.java:66)
	at org.eclipse.ui.editors.text.TextFileDocumentProvider.executeOperation(TextFileDocumentProvider.java:461)
	at org.eclipse.ui.editors.text.TextFileDocumentProvider.saveDocument(TextFileDocumentProvider.java:764)
	at org.eclipse.ui.texteditor.AbstractTextEditor.performSave(AbstractTextEditor.java:4996)
	at org.eclipse.ui.texteditor.AbstractTextEditor.doSave(AbstractTextEditor.java:4796)
	at org.eclipse.ui.texteditor.AbstractTextEditor$TextEditorSavable.doSave(AbstractTextEditor.java:7162)
	at org.eclipse.ui.Saveable.doSave(Saveable.java:213)
	at org.eclipse.ui.internal.SaveableHelper.doSaveModel(SaveableHelper.java:333)
	at org.eclipse.ui.internal.SaveableHelper.lambda$1(SaveableHelper.java:196)
	at org.eclipse.ui.internal.SaveableHelper.lambda$3(SaveableHelper.java:271)
	at org.eclipse.jface.operation.ModalContext.runInCurrentThread(ModalContext.java:434)
	at org.eclipse.jface.operation.ModalContext.run(ModalContext.java:352)
	at org.eclipse.ui.internal.WorkbenchWindow.lambda$7(WorkbenchWindow.java:2345)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:74)
	at org.eclipse.ui.internal.WorkbenchWindow.run(WorkbenchWindow.java:2343)
	at org.eclipse.ui.internal.SaveableHelper.runProgressMonitorOperation(SaveableHelper.java:278)
	at org.eclipse.ui.internal.SaveableHelper.runProgressMonitorOperation(SaveableHelper.java:260)
	at org.eclipse.ui.internal.SaveableHelper.saveModels(SaveableHelper.java:207)
	at org.eclipse.ui.internal.SaveableHelper.savePart(SaveableHelper.java:150)
	at org.eclipse.ui.internal.WorkbenchPage.saveSaveable(WorkbenchPage.java:3799)
	at org.eclipse.ui.internal.WorkbenchPage.saveEditor(WorkbenchPage.java:3812)
	at org.eclipse.ui.internal.handlers.SaveHandler.execute(SaveHandler.java:82)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:97)
	at jdk.internal.reflect.GeneratedMethodAccessor61.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:317)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:251)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:488)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.executeCommand(KeyBindingDispatcher.java:308)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.press(KeyBindingDispatcher.java:580)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.processKeyEvent(KeyBindingDispatcher.java:647)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher.filterKeySequenceBindings(KeyBindingDispatcher.java:439)
	at org.eclipse.e4.ui.bindings.keys.KeyBindingDispatcher$KeyDownFilter.handleEvent(KeyBindingDispatcher.java:96)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.filterEvent(Display.java:1280)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1062)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1087)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1072)
	at org.eclipse.swt.widgets.Widget.sendKeyEvent(Widget.java:1114)
	at org.eclipse.swt.widgets.Widget.sendKeyEvent(Widget.java:1110)
	at org.eclipse.swt.widgets.Widget.wmChar(Widget.java:1539)
	at org.eclipse.swt.widgets.Control.WM_CHAR(Control.java:4868)
	at org.eclipse.swt.widgets.Canvas.WM_CHAR(Canvas.java:345)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4746)
	at org.eclipse.swt.widgets.Canvas.windowProc(Canvas.java:340)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5015)
	at org.eclipse.swt.internal.win32.OS.DispatchMessage(Native Method)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3630)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)

jantje, 2022-05-24:
This seems to have to do something with the formatting and removal of trailing whitespace from changed lines.
Are there special characters in this file?

jantje, 2022-05-25:
C/C++ is old and does not support modern stuff like long file names, spaces in paths names and unicode properly? So Chinese giving problems during compilation seems "normal" to me as this is generic C/C++ old tools running.

The formatting however is eclipse/java code and can/should support this, however I have no clue how much eclipse has penetrated the Chinese market and as such I have nu clue how bug free the code is or how much support you can get.

The biggest problem for you is finding someone who can help.
Sloeber has nothing to do with the formatter so I can't really help you.
From the stacktrace these components are involved CDT, eclipse. core and jface
If you send the problem to all these components you will get backlash for cross posting.
If you send it to one component you can increase your chances of success by "translating the problem to their environment".
In this case that is probably the easiest for CDT so I would start there.
To do so 
1. make a .cpp file (just copy your .ino file to a .cpp file in sloeber)
2. Format it with a CDT provided formatter (one called build-in)
3. Make it so that a format of the file causes the problem
4. Add at the top of the file a comment that this file causes a java error when in file format (I guess that would be UTF-8 in your case) and formatter selected in point 2
5. Contact the CDT team and ask to take a look (create bug report/post on the CDT dev list)

closed 2022-05-25 by xiezhoubin