Sloeber/arduino-eclipse-plugin #1480: Errors will be reported when compiling multiple versions of package.

xiezhoubin, 2022-05-25:
I clearly chose the package of version 1.0.6, but according to the print information of the console, I can see the package of version 2.0.6 used by the program.
[image]
[image]
[image]

Exception Stack Trace:
org.eclipse.core.runtime.CoreException: No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project
	at org.eclipse.core.internal.expressions.TypeExtensionManager.getProperty(TypeExtensionManager.java:131)
	at org.eclipse.core.expressions.TestExpression.evaluate(TestExpression.java:104)
	at org.eclipse.core.expressions.CompositeExpression.evaluateAnd(CompositeExpression.java:54)
	at org.eclipse.core.expressions.AndExpression.evaluate(AndExpression.java:36)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.failsEnablement(RegistryPageContributor.java:284)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.isApplicableTo(RegistryPageContributor.java:227)
	at org.eclipse.ui.internal.dialogs.PropertyPageContributorManager.contribute(PropertyPageContributorManager.java:114)
	at org.eclipse.ui.internal.dialogs.PropertyDialog.createDialogOn(PropertyDialog.java:64)
	at org.eclipse.ui.dialogs.PreferencesUtil.createPropertyDialogOn(PreferencesUtil.java:217)
	at org.eclipse.ui.dialogs.PreferencesUtil.createPropertyDialogOn(PreferencesUtil.java:145)
	at io.sloeber.ui.actions.OpenPreferencesHandler.execute(OpenPreferencesHandler.java:46)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:283)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:97)
	at jdk.internal.reflect.GeneratedMethodAccessor65.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:317)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:251)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:173)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:156)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:488)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:213)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:438)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:449)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:475)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4243)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1063)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4060)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3632)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:401)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)

Code:

#include <Arduino.h>  // 导入Arduino库文件
#include <WiFi.h>     // 导入WiFi库文件
#include <esp_now.h>  // 导入esp_now库文件

#define debugState true             // 串口调试打印状态，true--开启调试打印，false--关闭调试打印
#define debugSerial Serial          // 调试打印的串口
#define debugSerialBuadrate 115200  // 调试打印的串口波特率

#define espnowChannel 0                                        // ESP-NOW频道为0，即使用当前所在Station或Softap的频道。否则，必须设置为Station或Softap的所在频道
#define espnowEncryptState false                               // ESP-NOW对等方发送/接收的数据是否加密，false--不加密，true--加密
#define numberOfReceivingTerminals 2                           // 接收端数量
uint8_t broadcastAddresses[numberOfReceivingTerminals][6] = { // 接收端设备的MAC地址
  { 0x58, 0xBF, 0x25, 0x80, 0x79, 0x54 }, { 0x58, 0xBF, 0x25, 0x81, 0xF5, 0xE0 }
};

#define sendDataIntervalTime 3000  // 发送数据的间隔时间，3000ms
uint32_t lastSendDataTime = 0; // 上一次发送数据的时间

// 发送数据的结构体，最大内容为250字节，发送端与接收端必须保持一致！
struct structMessage {
  uint8_t a;
  uint16_t b;
  uint32_t c;
  int d;
  boolean e;
  String f;
  float g;
};

structMessage myData; // 创建发送数据的结构体
esp_now_peer_info_t peerInfo; // 创建对等点结构体

/**
   函数功能：发送数据后的回调函数
   参数1：[mac_addr] [const uint8_t*] 接收端设备的MAC地址
   参数2：[sendStatus] [esp_now_send_status_t] 发送状态，ESP_NOW_SEND_SUCCESS或ESP_NOW_SEND_FAIL
   返回值：无
   注意事项：无
*/
void onDataSent(const uint8_t *mac_addr, esp_now_send_status_t sendStatus) {
#if debugState
  char macStr[30] = { 0 };
  snprintf(macStr, sizeof(macStr), "0x%02X,0x%02X,0x%02X,0x%02X,0x%02X,0x%02X", mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
  debugSerial.print(F("接收端设备的MAC地址为: "));
  debugSerial.print(macStr);
  debugSerial.print(F("，发送结果为: "));
#endif

  if (sendStatus == ESP_NOW_SEND_SUCCESS) { // 判断发送状态
#if debugState
    debugSerial.println(F("成功【√】"));
#endif
  } else {
#if debugState
    debugSerial.println(F("失败【×】"));
#endif
  }
}

/**
   函数功能：ESP-NOW初始化
   参数：无
   返回值：无
   注意事项：无
*/
void espnowInit() {
  WiFi.mode(WIFI_STA); // 将WIFI模式设置为Station模式
  // WiFi.disconnect(); // 断开当前WIFI连接

  if (esp_now_init() != ESP_OK) { // 初始化ESP-NOW
#if debugState
    debugSerial.println(F("ESP-NOW初始化失败！"));
#endif
    return;
  }

  esp_now_register_send_cb(onDataSent); // 设置发送数据的回调函数

  peerInfo.channel = espnowChannel; // 设置ESP-NOW频道
  peerInfo.encrypt = espnowEncryptState; // 设置ESP-NOW对等方发送/接收的数据是否加密

  for (uint8_t i = 0; i < numberOfReceivingTerminals; i++) {
    memcpy(peerInfo.peer_addr, broadcastAddresses[i], 6); // 将接收端设备的MAC地址添加到对等点结构体中
    if (esp_now_add_peer(&peerInfo) != ESP_OK) { // 将对等点结构体添加到对等点列表中
#if debugState
      debugSerial.println(F("添加对等点列表失败！"));
#endif
      return;
    }
  }
}

/**
   函数功能：发送ESP-NOW数据
   参数：无
   返回值：无
   注意事项：无
*/
void sendDataTask() {
  if (millis() - lastSendDataTime >= sendDataIntervalTime) { // 判断是否到达发送时间
    lastSendDataTime = millis(); // 更新上一次的发送时间

    // 设置要发送的数据
    myData.a = random(0, 256);
    myData.b = random(0, 1000);
    myData.c = random(0, 10000);
    myData.d = random(-10000, 10000);
    myData.e = random(0, 2);
    myData.f = "谢周斌你好呀";
    myData.g = 1.23;

    esp_err_t result = esp_now_send(0, (uint8_t*) &myData, sizeof(myData)); // 第1个参数为0则表示向添加到对等点列表的所有对等方发送数据
    if (result == ESP_OK) { // 判断发送数据结果
#if debugState
      debugSerial.println(sizeof(myData));
      debugSerial.println(F("数据已发送，请查看发送结果！"));
#endif
    } else {
#if debugState
      debugSerial.println(F("数据发送失败，请检查！"));
#endif
    }
  }
}

/**
   函数功能：程序初始化时调用，仅执行一次
   参数：无
   返回值：无
   注意事项：无
*/
void setup() {
  delay(1000); // 延迟1000ms，等待各设备上电稳定

#if debugState
  debugSerial.begin(debugSerialBuadrate); // 初始化调试打印的串口
#endif

  espnowInit(); // ESP-NOW初始化

#if debugState
  debugSerial.println(F("\r\n【ESP32-NOW一对多发送数据测试】程序初始化完成！"));
#endif
}

/**
   函数功能：程序主循环函数，重复运行
   参数：无
   返回值：无
   注意事项：无
*/
void loop() {
  sendDataTask(); // 发送数据任务
}

xiezhoubin, 2022-05-25:
[image]

jantje, 2022-05-25:
>I clearly chose the package of version 1.0.6, but according to the print information of the console, I can see the package of version 2.0.6 used by the program.

I think this should say
I clearly chose the package of version 1.0.6, but according to the print information of the console, I can see the libraries of version 2.0.6 used by the program.

Did you try the reattach libraries functionality in the sloeber menu?

xiezhoubin, 2022-05-25:
> > 我明明选择了1.0.6版本的包，但是根据控制台的打印信息，可以看到程序使用的2.0.6版本的包。
> 
> 我想这应该说 我明明选择了1.0.6版本的包，但是根据控制台的打印信息，可以看到程序使用的2.0.6版本的库。
> 
> 您是否尝试过 sloeber 菜单中的重新附加库功能？

Solved. You really need to click the "reattach libraries" option in the "sloeber" menu. I thought it would be automatic every time I changed a different package version.
[image]

jantje, 2022-05-25:
>I thought it would be automatic every time I changed a different package version.

It is complicated, it really is. It should automatically update, but as it is complicated in some cases it won't and therefore there is this special command.
Note that running this command may cause problems for certain libraries/packages and therefore the command is not run all the time.

closed 2022-05-25 by xiezhoubin