Sloeber/arduino-eclipse-plugin #1481: Sloeber Passing Invalid Argument to esptool.py in Ubuntu 20.04

emintz, 2022-05-29:
I checked the open issues.

**Describe your environment**   

Ubuntu 20.04 on a large Intel-based system

Do you use the Sloeber product or plugin?   Product

Sloeber version number  
  Sloeber	4.4.0.202107110422	io.sloeber.product
  Sloeber, the Eclipse Arduino IDE (plugin version)	4.4.0.202107110422	io.sloeber.feature.feature.group	Sloeber.io

OS: [e.g. windows 10]   
Ubuntu 20.04 LTE, patched < 1 week ago

The url of the json that you used to download the platform  

Sloeber: I downloaded Sloeber .zip file from the website, unpacked it, and moved the Sloeber directory into my home directory. The .zip file is: V4.4.0_linux64.2021-07-04_06-48-31.zip. I still have the file.

ESP32 Plugin: I cloned the github repository and installed the package per instructions. The package is installed as a private package. Its path is: /homd/git/arduino-esp32

I changed platform.sloeber.txt as suggested here: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1416

Here's a diff of the original and changed file:

eric@mighty-wurlitzer:~/git/arduino-esp32$ diff platform.sloeber.txt platform.sloeber.txt.original
217c217
< tools.esptool_py.upload.pattern=python3 "${tools.esptool_py.path}/${tools.esptool_py.cmd}" ${tools.esptool_py.pattern_args}
---
> tools.esptool_py.upload.pattern=python3 "${tools.esptool_py.path}/${tools.esptool_py.cmd}" ${upload.pattern_args}
eric@mighty-wurlitzer:~/git/arduino-esp32$ 

My third-party index URLs are:

https://downloads.arduino.cc/packages/package_index.json
https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json
https://raw.githubusercontent.com/jantje/ArduinoLibraries/master/library_jantje_index.json
https://arduino.esp8266.com/stable/package_esp8266com_index.json
https://downloads.arduino.cc/libraries/library_index.json
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

The last URI is the one I added.

A image of the project properties arduino showing the board settings 

Here are the project settings:

  Configuration: Release [Active]
  Platform folder: /home/eric/git/arduino-esp32
  Board: Node32s
  Upload protocol: Default
  Port: /dev/ttyUSB0
  Partition Scheme: Default
  Flash Frequency: 80 MHz
  Core Debug Level: None
  Upload Speed: 921600

**Describe the bug**   
A clear and concise description of what the bug is.

Cannot upload an empty sketch to the ESP32 Board. Sketch compiles fine

**To Reproduce**   
Steps to reproduce the behavior:   

1. Create an empty project
2. Click the Arduino checkmark button to compile the sketch. The sketch compiles successfully.
3. Click the Arduino --> (right pointing arrow) to upload the sketch.
4. The "Build before upload?" pop-up appears.
5. Click "Yes"
6. Sketch compiles successfully again.
7. The following commands appear on the console
Starting upload
Uploading project "ESP32Test" with "esptool_py"
Launching: python3 /home/eric/git/arduino-esp32/tools/esptool/esptool.py 
9. Upload fails as follows:

[image]
Output:
esptool.py v3.3-dev
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,get_security_info,version}
               ...

esptool.py v3.3-dev - Espressif chips ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,get_security_info,version}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    merge_bin           Merge multiple raw binary files into a single file for
                        later flashing
    get_security_info   Get some security-related data
    version             Print esptool version

optional arguments:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2}, -c {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,usb_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset,no_reset_stub}, -a {hard_reset,soft_reset,no_reset,no_reset_stub}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use
                        with care)
  --connect-attempts CONNECT_ATTEMPTS
                        Number of attempts to connect, negative or 0 for
                        infinite. Default: 7.
The execution of command "python3" is done.

1. Go to '...'
9. Click on '....'
11. Scroll down to '....'
12. See error

provide the name of the board and the platform version you are using.  
Provide the json url and a screenshot of project properties->arduino  

**Expected behavior**   
Sketch uploaded to the microcontroller.

**Data to back up your claim**   
Please let me know what else is needed. 

**Additional context**   

Please let me know what information is reqired.

I apologize if I have misses something obvious, and thank you so much for developing this project. I also use it with the Arduino development environment, and I love it.

Please stay safe and well

jantje, 2022-05-30:
That is a very well documented issue. Congrats for that.
Reading through this a couple of times I do think this is a duplicate of #1416
Note that platform.sloeber.txt is considered a temporary file and can be regenerated at any time, though I do not think this is why you have the problem. Anyway it is safer to modify the platform.txt and delete platform.sloeber.txt (if it is recreated you know your changes are taken into account as described in the workaround https://github.com/Sloeber/arduino-eclipse-plugin/issues/1364#issuecomment-890544323 )
Also note that the safest way to make sure Sloeber rereads the platform.txt files is by going to  project properties->Sloeber (Arduino in your version) and select apply close.
Another way around this is to get the latest nightly. At this point that is here https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/2402582489

emintz, 2022-06-04:
Thank you. 

You are 100% correct. Applying the workaround fixed it. The trick is to read it through to the end because I run on Linux.

I look forward to happy developing. Thank you once again.

-- Eric

closed 2022-06-04 by emintz