Sloeber/arduino-eclipse-plugin #1482: Sloeber does not find ESP 32 platform libraries after update of ESP platform libraries to version 2.0.3

owenl1971, 2022-06-03:
I have tested this problem with the ESP example code - ESP Now Basic Master.
The code complies in Sloeber if the platform is set to ESP32 - version 1.0.6, but not if the platform is set to 2.0.3
compile fails because the library

` #include <esp_now.h>`

is not found

the same code complies ok in the arduino environment with either of the 1.0.6 and 2.0.3 platforms selected.

Looking at the platform library I note that the 2.0.3 library has a different structure to the 1.0.6 library
in 1.0.6 esp_now. is found here

> /Sloeber43/arduinoPlugin/packages/esp32/hardware/esp32/1.0 (copy).6/tools/sdk/include/esp32

while in the 2.0.3. the same library is here

> /Sloeber43/arduinoPlugin/packages/esp32/hardware/esp32/2.0.3/tools/sdk/esp32/include/esp_wifi/include

jantje, 2022-07-22:
I just see this issue now.

The location on disk `/Sloeber43/arduinoPlugin/packages/esp32/hardware/esp32/1.0 (copy).6/` does not match version 1.0.6.
So it looks to me like version 1.0.6 has not been installed using the sloeber boards manager.
To fix this delete the `1.0 (copy).6` folder and install the version 1.0.6 using the sloeber boards manager

owenl1971, 2022-07-24:
I have cleaned and reinstall the hardware package and confirm they are at the following locations.

Version 1.0.6 - /Desktop/Sloeber43/arduinoPlugin/packages/esp32/hardware/esp32/1.0.6
Version 2.0.4 - /Desktop/Sloeber43/arduinoPlugin/packages/esp32/hardware/esp32/2.0.4

when version 1.0.6 is selected everything works, when 2.0.4 compilation fails becasue the headers can't be found. I am using esp_now.h for reference.

In version 1.0.6 the ESP_Now library is located here
/Desktop/Sloeber43/arduinoPlugin/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp32$ 

In 2.0.4 it is located here
/Desktop/Sloeber43/arduinoPlugin/packages/esp32/hardware/esp32/2.0.4/tools/sdk/esp32/include/esp_wifi/include$

jantje, 2022-08-04:
I tried to reproduce the issue here in the latest stable sloeber and esp32 2.0.4 and sloeber compiles fine and the indexer finds esp_now.h at the location you specified it should be found.
I needed to compile the project to get Sloeber to find the code (esp_now.h is not a arduino library that needs to be discovered and attached so this is actually a simple case)
Then I switched to esp version 1.0.4 and it compiled fine again and sloeber found the esp_now.h file at the location you specified.

In other words : I can not reproduce the issue

jantje, 2022-09-16:
closing as to long without new content.
Please reopen with new content when you disagree

closed 2022-09-16 by jantje