Sloeber/arduino-eclipse-plugin #1483: The MAC system cannot be installed and used.

xiezhoubin, 2022-06-04:
[image]
[image]
[image]
[image]

jantje, 2022-06-05:
I'm sorry. I don't know what you are trying to do and I do not understand most of what is visible in those pictures.

xiezhoubin, 2022-06-06:
> I'm sorry. I don't know what you are trying to do and I do not understand most of what is visible in those pictures.

This means that the MAC system cannot open the installation package, and an error "sloeber is damaged" will be reported. I have tried several versions without success.

jantje, 2022-06-06:
That is most likely a Mac security feature. Have you seen the install comments here http://eclipse.baeyens.it/stable.php?OS=MacOS

dblewis76, 2022-07-06:
I have tried to install V4.4.0 on MacOS Monterey also. After downloading, moving the ZIP file out of the Download folder and using the Mac native Archive Utility to unpack and running the sudo command as stated here  http://eclipse.baeyens.it/stable.php?OS=MacOS, when I start Sloeber, I get these errors:
[image]
[image]
[image]
[image]

Expanding one of these errors shows:
[image]

What do I need to do to fix this? I there a specific version of MacOS that V4.4.0 works with?

jantje, 2022-07-22:
I think I have read somewhere you need to upgrade your java.

jantje, 2022-09-16:
closing as to long without activity
Please reopen if you have new info

closed 2022-09-16 by jantje