Sloeber/arduino-eclipse-plugin #1486: Migration from 4.3.3 to 4.4.0 workspace issues

ElsomaUwe, 2022-06-27:
---

using Windows 10 on a Lenovo Yoga Laptop
working with Arduino Due projects
using sloeber product
was upgrading from 4.3.3. to 4.4.0
**Using same/old workspace**
[image]

**Describe the bug**   
1) eclipse / sloeber crashes sometimes and does not react
2) compiler/toolchain does not find "something", after changing board to another board and then changing it back, it works
3) darkest dark theme does not work any longer in old workspace. In a new workspace it works.

jantje, 2022-06-28:
>    eclipse / sloeber crashes sometimes and does not react

This has been reported before. It is a difficult migration and a migration is hard to test.
One properly migrated this should be fixed

>    compiler/toolchain does not find "something", after changing board to another board and then changing it back, it works

The "changing board to another board and then changing it back" procedure is the best known to fix "something" issues.

>   darkest dark theme does not work any longer in old workspace. In a new workspace it works.

Sloeber does not do anything with the Eclipse theme's.

jantje, 2022-07-22:
Can this issue be closed?

ElsomaUwe, 2022-07-25:
If there is no major issue with it for all users that migrate to new version, this item can be closed. 
At my PC it is still a migrate old projects in new Version.

jantje, 2022-07-25:
I'm considering to remove the auto upgrade code and make it a manual process. That should fix the hanging/ not reacting.

jantje, 2023-01-16:
I removed the auto upgrade which should stop the hanging. It is currently in the nightly

closed 2023-01-16 by jantje