Sloeber/arduino-eclipse-plugin #1489: No valid system found for "avr-gcc"

ghollf, 2022-07-11:
Problem installing tools on MBA M1 / Eclipse 2022-06 / OpenJDK11

!SUBENTRY 1 io.sloeber.core 4 0 2022-07-11 01:40:20.148
!MESSAGE Failed to install Arduino AVR Boards
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 01:40:20.148
!MESSAGE No valid system found for "avr-gcc".
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 01:40:20.148
!MESSAGE No valid system found for "avrdude".

TRY 1 io.sloeber.core 4 0 2022-07-11 02:40:25.388
!MESSAGE Failed to install esp32
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 02:40:25.388
!MESSAGE No valid system found for "riscv32-esp-elf-gcc".
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 02:40:25.388
!MESSAGE No valid system found for "xtensa-esp32-elf-gcc".
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 02:40:25.388
!MESSAGE No valid system found for "xtensa-esp32s2-elf-gcc".
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 02:40:25.388
!MESSAGE No valid system found for "xtensa-esp32s3-elf-gcc".
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 02:40:25.388
!MESSAGE No valid system found for "esptool_py".
!SUBENTRY 2 io.sloeber.core 4 0 2022-07-11 02:40:25.388
!MESSAGE No valid system found for "mklittlefs".

ghollf, 2022-07-11:
os.arch=aarch64
os.name=Mac OS X
os.version=12.4

I think Arduino tools will not provide native support, so aarch64 should be treated as x86_64.

jantje, 2022-07-11:
When the build is finished you can try to install from here https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/2651591737

jantje, 2022-07-22:
@ghollf did you find some time to check whether the fix fixed your problem?

ghollf, 2022-07-22:
Not yet. I have switched back to a PC temporarily.

jantje, 2022-09-16:
As this "should be fixed" and nobody else seems to have the issue I'm closing the issue.
Please reopen if you find the time to test and the test fails

closed 2022-09-16 by jantje