Sloeber/arduino-eclipse-plugin #1491: Indexer in V4.4.0 is Seriously Flakey with ESP32 Code

gfvalvo, 2022-07-12:
With just about any ESP32 code, will usually get tons of fake “… unresolved” errors even though the code compiles totally error-free. I’ve played with all the check boxes under Window --> Preferences --> C/C++ --> Indexer. I’ve tried deleting the errors in the “Problems” tab, Clean Project, Recompile, Build All, Close Project, Close Sloeber, etc.

Then for no apparent reason, most of the fake errors might go away. Or, they may stay permanently.

Can’t find the right Kabuki Dance to make it work all the time.

MasteringTheMess, 2022-07-14:
Had the same issue with AVR and STM32, so seems not to be platform specific.
This worked ONCE for me:

close sloeber, delete indexer settings in the projects .prefs file, restart sloeber, select project->Indexer->Rebuild Index
[(https://stackoverflow.com/questions/11357009/eclipse-indexing-not-working/41760020#41760020)]

jantje, 2022-07-14:
please try the nightly to see if the problem still exists
https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/2651591737

gfvalvo, 2022-07-14:
> please try the nightly to see if the problem still exists
> https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/2651591737

Jantje, I will try. But, I'm not a tool chain guy and I don't know how to "build" anything. Do I just click on "windows" and download the .zip? Then, click on "sloeber-ide.exe"?

[image]

[image]

gfvalvo, 2022-07-14:
OK, thanks. That's how I install all versions of the Sloeber Product ... in their own directory right under root, each with their workspace directory (also right under root).

MasteringTheMess, 2022-07-18:
Sorry, but after a while now it is back to the original state where the indexer stopped working.
This is very clearly indicated by missing syntax coloring and "jump to declaration" not working again.
Rebuilding the index does not work, clicking the command does nothing at all.
Even the trick above (deleting some contents in .prefs does not remedy the situation...

jantje, 2022-07-18:
I'm sorry to hear this.
Are there errors/warning in the log file?

MasteringTheMess, 2022-07-18:
hmmm, where would I find that logfile?
I found a build log deep inside the .metadata folder...

MasteringTheMess, 2022-07-18:
hmmm, that error appears also when the index is rebuilt properly :/

MasteringTheMess, 2022-07-19:
now it happened again.
(I am on STM32,  not ESP32. but I think it might not be platform specific)

sloeber was open, I closed and reopened the project (which closed all open files as well) and after reopening the project and a file, syntax coloring did not happen, call hierarchy, reference searching etc. don't work. Second file too. Rebuilding index did not start, F5 also with no effect.

But there is nothing in the error log, only the bespoke and unrelated property tester message.

MasteringTheMess, 2022-07-19:
found a menu option on sourcefile right-click -> index - create parser log

here's the output:
[eclipse_parser_log.log](https://github.com/Sloeber/arduino-eclipse-plugin/files/9138665/eclipse_parser_log.log)

this is interesting:

```
Unresolved includes (from headers in index):
   file:/C:/Users/xXxXxXx/Documents/sloeber_projects/PowerSupply/Controller.h is not indexed
   file:/C:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/Arduino.h is not indexed
   file:/C:/Users/xXxXxXx/Documents/Arduino/libraries/SafeString/src/loopTimer.h is not indexed
   file:/C:/Users/xXxXxXx/Documents/Arduino/libraries/MemoryFree-master/MemoryFree.h is not indexed
```

nevertheless nothing happens when I try

- rebuild index
- update with modified files
- freshen all files
- re-resolve unresolved includes

jantje, 2022-07-19:
Are you sure this is not a casing issue Arduino.h versus arduino.h or something like that?

MasteringTheMess, 2022-07-19:
nope, not sure!
... but as it builds and I am on Windows I'd guess casing is not the root evil here...

I will try in a xubuntu VM

MasteringTheMess, 2022-07-20:
just a thought on this: if it builds without problem on Linux, can I rule out a casing issue?

jantje, 2022-07-20:
Yes

gfvalvo, 2022-08-03:
So, is there a solution? This bug really limits the usefulness of Sloeber when working with ESP32 code. The "Open Declaration" function is probably the feature I use most. As things are now, that fails often even though the code compiles totally error-free.

jantje, 2022-08-03:
The root cause has still not been identified.

MasteringTheMess, 2022-08-04:
a clumsy and not always working workaround is to close and re-open sloeber...

gfvalvo, 2022-09-02:
> a clumsy and not always working workaround is to close and re-open sloeber...
Even that rarely works.

jantje, 2022-09-02:
Have you tried to trigger the discovery?
project properties->pre processor includes->providers->arduino compiler settings->Command to get compiler specs. change it (add a space)
Select Apply and close
As you are there also activate Allocate console in console view.
Next time it happens see whether the console provides some debugging info
[image]

jantje, 2022-09-16:
Today I tried this with https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
Using version 2.0.4
I didn't see any problems.
[image]

MasteringTheMess, 2022-10-18:
Despite the title beeing for ESP32, I experience the same indexer misbehaviour for STM32.
I have Sloeber v 4.4.1.202207151257

Created a new project, add some very simple code (read inputs and write to serial)
(first hickup, that has nothing to do with this issue: SrcWrapper.h was not added automatically)
I saw that syntax-coloring was halfway done, but it wouldn't compile because of a missing semicolon. After correcting that and rebuilding (I always do a "build clean" before a build) Syntax coloring was completely gone.
Rebuilding the index, refreshing all files, re-resolving unresolved includes, nothing of these helped.

I then did the steps you wrote a few weeks back to trigger the discovery:

> Have you tried to trigger the discovery? project properties->pre processor includes->providers->arduino compiler settings->Command to get compiler specs. change it (add a space) Select Apply and [...]

And well, it worked!
This is a one-time experience for now, since I did not succeed in reproducing a state where the indexer misbehaves...

MasteringTheMess, 2022-10-19:
today (computer was in hibernation over night, sloeber and the project left open) I closed the project and reopened it.
No syntax coloring
variables within the while seem to be somehow indexed (can jump to the declaration with F3)
inculded stuff (Serial, pinMode, INPUT_PULLUP) seem not to be indexed, nothing happens when hitting F3
rebuilding (even clean build) did not help
after a rebuild, some code changes, upload, another rebuild the following errors appeared:

```
eclipse.buildId=unknown
java.version=17.0.3
java.vendor=Eclipse Adoptium
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_GB
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective

org.eclipse.core.expressions
Error
Wed Oct 19 09:10:50 CEST 2022
No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project

org.eclipse.core.runtime.CoreException: No property tester contributes a property org.eclipse.embedcdt.managedbuild.cross.core.isGnuMcu to type class org.eclipse.core.internal.resources.Project
	at org.eclipse.core.internal.expressions.TypeExtensionManager.getProperty(TypeExtensionManager.java:131)
	at org.eclipse.core.expressions.TestExpression.evaluate(TestExpression.java:104)
	at org.eclipse.core.expressions.CompositeExpression.evaluateAnd(CompositeExpression.java:54)
	at org.eclipse.core.expressions.AndExpression.evaluate(AndExpression.java:36)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.failsEnablement(RegistryPageContributor.java:284)
	at org.eclipse.ui.internal.dialogs.RegistryPageContributor.isApplicableTo(RegistryPageContributor.java:227)
	at org.eclipse.ui.internal.dialogs.PropertyPageContributorManager.contribute(PropertyPageContributorManager.java:114)
	at org.eclipse.ui.internal.dialogs.PropertyDialog.createDialogOn(PropertyDialog.java:64)
	at org.eclipse.ui.dialogs.PropertyDialogAction.createDialog(PropertyDialogAction.java:169)
	at org.eclipse.ui.dialogs.PropertyDialogAction.run(PropertyDialogAction.java:151)
	at org.eclipse.ui.internal.ide.actions.ProjectPropertyDialogAction.run(ProjectPropertyDialogAction.java:81)
	at org.eclipse.jface.action.Action.runWithEvent(Action.java:474)
	at org.eclipse.jface.action.ActionContributionItem.handleWidgetSelection(ActionContributionItem.java:580)
	at org.eclipse.jface.action.ActionContributionItem.lambda$4(ActionContributionItem.java:414)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:89)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4251)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1066)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4068)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3645)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1155)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1046)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:155)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:644)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:338)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:551)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:156)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:152)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:203)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:136)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:402)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:255)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:659)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:596)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1467)
```

```
eclipse.buildId=unknown
java.version=17.0.3
java.vendor=Eclipse Adoptium
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_GB
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective
[... 169 lines cut ...]
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_GB
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective

org.eclipse.jface
Warning
Wed Oct 19 09:23:04 CEST 2022
A conflict occurred for CTRL+U:
Binding(CTRL+U,
	ParameterizedCommand(Command(io.sloeber.actions.UploadprojectAction,Upload Sketch,
		Upload selected project to your board,
[... cut ...]
		org.eclipse.ui.internal.WorkbenchHandlerServiceHandler@3063be68,
		,,true),null),
	org.eclipse.ui.defaultAcceleratorConfiguration,
	org.eclipse.ui.contexts.window,,,user)
Binding(CTRL+U,
	ParameterizedCommand(Command(io.sloeber.actions.UploadProjectAction,Upload Sketch,
		Upload selected project to your board,
		Category(io.sloeber.commands,Sloeber,Sloeber Plugin commands,true),
		org.eclipse.ui.internal.WorkbenchHandlerServiceHandler@3a3bc0da,
		,,true),null),
	org.eclipse.ui.defaultAcceleratorConfiguration,
	org.eclipse.ui.contexts.window,,,user)
```

MasteringTheMess, 2022-10-19:
the only "good" thing, it seems that I can reproduce the non-indexed state now...
closing the project and reopening leaves me without syntax coloring
closing sloeber and reopening restores syntax coloring

this has worked a few times now

jantje, 2022-10-20:
>the only "good" thing, it seems that I can reproduce the non-indexed state now...

This is great. I have been able to reproduce the problem.
Now I can have a look at what causes this:
A first look: based on the observation that the reindex of the project is very fast I think the "do not index flag" does not get reset.
Which gets confirmed by the log file of my test (below the end of the log file index)
```
!ENTRY io.sloeber.core 8 0 2022-10-20 14:26:36.609
!MESSAGE Do not index esp32New

!ENTRY io.sloeber.core 8 0 2022-10-20 14:26:40.939
!MESSAGE Ignoring configuration change during project creation esp32New

!ENTRY io.sloeber.core 8 0 2022-10-20 14:26:45.173
!MESSAGE Ignoring configuration change during project creation esp32New

!ENTRY io.sloeber.core 8 0 2022-10-20 14:26:46.672
!MESSAGE pospone index esp32New
```
As sloeber runs on workspace level; obviously a restart of the workspace fixes this.

MasteringTheMess, 2022-12-01:
Any news on this issue?
It happens quite often and always restarting the workspace a bit of an annoying workaround...

jantje, 2022-12-01:
I think this issue should have improved maybe even fixed because of the changes in #1523.
It would be great if you could test.
Safest way to get it is from the actions here at github https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/3560037094

gfvalvo, 2022-12-01:
> Safest way to get it is from the actions here at github https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/3560037094
Is that a new official release? How to install it in Windows for testing?

MasteringTheMess, 2022-12-01:
Don't know if that's the best/ideal way, but I just backup the existing sloeber install folder as well as the workspace(s), then unzip in the install folder and run.

gfvalvo, 2022-12-01:
Oh, run "sloeber-ide.exe" from inside of "windows.zip\Sloeber\". That's easy. Thanks.

MasteringTheMess, 2022-12-01:
I'm on 4.4.1.202211272123 now, but the problem persists.

Directly after opening sloeber, it seems to be fine.

- I then close the project and reopen.
- Part of the indexing is gone, apparently everything that is loaded from libraries or arduino core (pinmode(); println(); INPUT_PULLUP; EthernetServer; ...).
- Stuff from the "local" files is indexed correctly.
- I can restore indexing by switching the workspace to itself :) (looks to me that this is the same as close and reopen sloeber).

I can reproduce this behavior very consistently...

jantje, 2022-12-01:
Ok.
I will start looking into this right now.

jantje, 2022-12-01:
I see this is a starting/racing condition (hard to fix stuff)
Basically the project.isOpen() returns true but the project.HasNature() retuns with an error the project is not yet open :-(
```
org.eclipse.core.internal.resources.ResourceException(/opoen_close)[368]: java.lang.Exception: Resource '/opoen_close' does not exist.
	at org.eclipse.core.internal.resources.ResourceException.provideStackTrace(ResourceException.java:42)
	at org.eclipse.core.internal.resources.ResourceException.<init>(ResourceException.java:38)
	at org.eclipse.core.internal.resources.Resource.checkExists(Resource.java:330)
	at org.eclipse.core.internal.resources.Resource.checkAccessible(Resource.java:204)
	at org.eclipse.core.internal.resources.Project.checkAccessible(Project.java:150)
	at org.eclipse.core.internal.resources.Project.hasNature(Project.java:536)
	at io.sloeber.core.api.Sketch.isSketch(Sketch.java:81)
	at io.sloeber.core.api.SloeberProject.getSloeberProject(SloeberProject.java:821)
	at io.sloeber.core.toolchain.SloeberConfigurationVariableSupplier.getSloeberProject(SloeberConfigurationVariableSupplier.java:25)
	at io.sloeber.core.toolchain.SloeberConfigurationVariableSupplier.getVariable(SloeberConfigurationVariableSupplier.java:32)
	at org.eclipse.cdt.managedbuilder.internal.envvar.ExternalExtensionEnvironmentSupplier.getVariable(ExternalExtensionEnvironmentSupplier.java:68)
	at org.eclipse.cdt.managedbuilder.internal.dataprovider.BuildEnvironmentContributor.getVariable(BuildEnvironmentContributor.java:56)

```
The error org.eclipse.core.internal.resources.ResourceException(/opoen_close)[368]: java.lang.Exception: **Resource '/opoen_close' does not exist.**
Where you can see that earlier the isOpen has been cheked.
[image]
I'm going to try to change the isSketch by checking on the .sProject file.

jantje, 2022-12-01:
>I'm going to try to change the isSketch by checking on the .sProject file.

Did that but it didn't fix the problem.

some more debugging got me to a possible workaround.
Can you go to project properties->Sloeber->other options->share in version control
Close sloeber and retry

jantje, 2022-12-01:
There is more going on. Forget about the workaround

MasteringTheMess, 2022-12-01:
I will be afk in a few minutes but happy to assist again tomorrow morning

jantje, 2022-12-02:
I have been working on this. I'm not sure no other things will pop up so some serious usage/testing will be needed to raise my comfort level.
If the build succeeded the version is available at https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/3597718536

MasteringTheMess, 2022-12-02:
Hi jantje, sorry for the late answer.
I just "installed" the latest build and at least my standard procedure to reproduce the broken index results in a good index.
tested multiple times with different projects

I will continue working with this versions, although my normal usage is quite trivial and is not much more than typing/editing code, using F3 for declaration lookups, searching for occurences or plain text and browsing the project explorer.
ah, and maybe playing around in the menus a bit.
Nothing fancy...
But if you have anything specific in mind that I can do to help you testing, let me know!

cheers and thank you!

MasteringTheMess, 2022-12-02:
One thing that I notice so far, but I am quite sure that this was already with the / a previous version:
The Sloeber Menu is missing in the sloeber perspective.
But it is there in the C/C++ perspective

MasteringTheMess, 2022-12-05:
I change project properties (compile options -  show warnings and alt. size command).
Closing and reopening project broke index again.
Switching workspace or restarting sloeber did not fix it :/

I then opened another project, for this project index and syntax coloring is OK
Now, the index for the first project is also OK, but syntax coloring isn't. I also get strange error messages from code analysis:

[image]
But "go to declaration" works fine

... and there seems to be no way to change this. Closing / opening the project / sloeber, manual rebuild of the index, changing the options back to how it was before, clean build of project, ... it remains in this state.

jantje, 2022-12-05:
>I change project properties (compile options - show warnings and alt. size command).
Closing and reopening project broke index again.
Switching workspace or restarting sloeber did not fix it :/

I tried to reproduce this but didn't get any problems.

[image]
My 2 cent
pinMode uses uint8_t as a parameter and this is unknown (therefore the ?) and as such the indexer can not map the arguments.
This clearly points towards a failed discovery because uint8_t is defined in a file on a folder that is not on the include path and needs to be discovered by querying the compiler.
The workaround described here should fix that https://github.com/Sloeber/arduino-eclipse-plugin/issues/1491#issuecomment-1235841293

As I can't reproduce and I have seen racing conditions while investigating this issue I'm currently assuming this is a racing condition that is not reproducible on my (slower old machines.
Activating the discovery console is the first step I would do

MasteringTheMess, 2022-12-06:
I'm not using ESP32, but STM32

Triggering the discovery as described brought back a nice, working index.
But, same as you, I cannot reproduce this behavior.

here's the log (of the relevant time span)

[image]
[image]
[image]

How do I activate the discovery console?

jantje, 2022-12-06:
>How do I activate the discovery console?

[image]

In your error log there are multiple errors I would consider serious and may cause the issue you have.
I guess this is a upgraded install. If so please try the product from the zip downloaded from the actions

MasteringTheMess, 2022-12-07:
Allright, will try with a fresh install then

MasteringTheMess, 2022-12-09:
Fresh install of latest "actions" download, worked fine for a while, I then changed some project option (sloeber->Arduino board->Optimize from -O0 to -O3)
Index and syntax coloring were immediately broken after "apply and close"
Neither restarting sloeber, nor triggering the discovery helped. (Indexer is terminated extremely fast after triggering the discovery)

heres the rror log

[image]

jantje, 2022-12-09:
I see some concerning logs but nothing related to the behaviour you describe.
Does the build work?
Can you provide the discovery console output?

MasteringTheMess, 2022-12-12:
well, besides the fact that I don't know where to look for that output (I think it should be displayed in the "normal" console view?), I currently have the problem that there is no text whatsoever displayed in the console - or the progress view - or any of the bottom views:

[image]

This is a fresh "install" of sloeber but with an existing workspace... Can that cause problems?
Shall I go for a completely "virgin" environment?
Is there more to it than newly downloaded product .zip and new workspace? Some registry magic to undo or sthg. similar?

MasteringTheMess, 2022-12-12:
after a restart, the other views seem to be restored, but the console output is still missing.

jantje, 2022-12-12:
You have to activate the discovery console  like shown here
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1491#issuecomment-1339833656

Normally a fresh install should be fine but someone in eclipse thought it was a good idea to activate oomph by default so that may cause problems

MasteringTheMess, 2022-12-12:
Hmm, strange that box is checked...
I triggered the discovery again and now it worked!

I had a correct index, changed some sloeber setting in properties -> sloeber -> Optimize from -O3 to -O0 and after that the index and syntax coloring was broken.

Here's what is in the console now:

```
12:23:54 **** Running scanner discovery: Arduino Compiler Settings ****
"C:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\tools\\xpack-arm-none-eabi-gcc\\10.3.1-2.3\\bin\\arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb "@C:\\Users\\xXxXxXx\\Documents\\Nucleo-Tests\\sloeberNucleo-Tests\\Test_L4R5\\Release/sketch/build.opt" -c -O0 -g -Wall -Wextra -std=gnu++17 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IC:\\Users\\xXxXxXx\\Documents\\Nucleo-Tests\\sloeberNucleo-Tests\\Test_L4R5" -IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/avr -IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32 -IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/LL -IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb -IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/OpenAMP -IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/hid -IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/cdc "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/STM32L4xx_HAL_Driver/Inc" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/STM32L4xx_HAL_Driver/Src" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/STM32L4xx" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Middlewares/OpenAMP/virtual_driver" -DSTM32L4xx -DARDUINO=10812 -DARDUINO_NUCLEO_L4R5ZI -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"NUCLEO_L4R5ZI\"" "-DVARIANT_H=\"variant_NUCLEO_L4R5ZI.h\"" -DSTM32L4R5xx -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/Core/Include/" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0\\system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/DSP/Include" "-IC:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\tools\\CMSIS\\5.7.0/CMSIS/DSP/PrivateInclude" -D__IN_ECLIPSE__=1 -E -P -v -dD -D__IN_ECLIPSE__ C:/Users/xXxXxXx/Documents/Nucleo-Tests/sloeberNucleo-Tests/.metadata/.plugins/org.eclipse.cdt.managedbuilder.core/spec.cpp 
arm-none-eabi-g++: error: C:\Users\xXxXxXx\Documents\Nucleo-Tests\sloeber_Nucleo-Tests\Test_L4R5\Release/sketch/build.opt: No such file or directory
Using built-in specs.
COLLECT_GCC=C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3\bin\arm-none-eabi-g++
Target: arm-none-eabi
Configured with: /Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/win32-ia32/sources/gcc/configure --prefix=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/win32-ia32/install/arm-none-eabi-gcc --infodir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/win32-ia32/install/arm-none-eabi-gcc/share/doc/info --mandir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/win32-ia32/install/arm-none-eabi-gcc/share/doc/man --htmldir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/win32-ia32/install/arm-none-eabi-gcc/share/doc/html --pdfdir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/win32-ia32/install/arm-none-eabi-gcc/share/doc/pdf --build=i686-pc-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --with-pkgversion='xPack GNU Arm Embedded GCC i386' --enable-languages=c,c++ --enable-mingw-wildcard --enable-plugins --enable-lto --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/win32-ia32/install/arm-none-eabi-gcc/arm-none-eabi --with-native-system-header-dir=/include --with-multilib-list=rmprofile --disable-build-format-warnings --with-syst [... cut ...] e/Templates/gcc/ -IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/Include -IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/PrivateInclude -D__IN_ECLIPSE__=1 -E -P -v -dD -D__IN_ECLIPSE__ C:/Users/xXxXxXx/Documents/Nucleo-Tests/sloeberNucleo-Tests/.metadata/.plugins/org.eclipse.cdt.managedbuilder.core/spec.cpp" terminated with exit code 1. Build might be incomplete.

12:23:54 Build Finished. 0 errors, 0 warnings. (took 187ms)
```

jantje, 2022-12-12:
Can you create the file
C:\Users\xXxXxXx\Documents\Nucleo-Tests\sloeber_Nucleo-Tests\Test_L4R5\Release/sketch/build.opt
and run the discovery again?

MasteringTheMess, 2022-12-12:
hmm, that file already exists (created 6 minutes after the above project settings change the corrupted the output)
Probably caused by some action I did to get back a working index/discovery

content of the file build.opt:
```

-fmacro-prefix-map=C:\\Sloeber\\arduinoPlugin\\packages\\STMicroelectronics\\hardware\\stm32\\2.3.0=.```

jantje, 2022-12-12:
I think this makes sense
C:\Users\xXxXxXx\Documents\Nucleo-Tests\sloeber_Nucleo-Tests\Test_L4R5\Release/sketch/build.opt is probably created in the prebuild actions of the build
I assume that changing the compilation options deletes the folder C:\Users\xXxXxXx\Documents\Nucleo-Tests\sloeber_Nucleo-Tests\Test_L4R5\Release/
And then the discovery fails
After a build the file is back but no discovery is run anymore

MasteringTheMess, 2022-12-12:
> I assume that changing the compilation options deletes the folder

yes, after clicking "apply & close" a lot happens in the GUI and among these happenings the release folder disappears.

I deleted the whole release folder and triggered discovery via the 
[above method](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1491#issuecomment-1235841293)

nothing happened, maybe some very quick actions that my eye didn't catch...

I then did a clean build, and triggered the discovery again.
Now the indexer runs for a longer time and after triggering the discovery index and syntax coloring is correct again.
But no output in the console from the discovery :/

as of yet I still have not found clearly what has to be done to recover the index... mostly a combination of a clean build and triggering the discovery works.
But sometimes it doesn't, I then restart sloeber and repeat the above steps in varying order until it eventually works again...

jantje, 2022-12-12:
>But no output in the console from the discovery :/

You have to jukebox the consoles. It is not really user friendly and error prone.

A fool proof full functionality supporting workaround won't be easy.
If you do not use the build.opt functionality I would comment out the line `build.opt.path=${build.path}/sketch/${build.opt.name}` in platform.txt (delete sloeber.txt; project properties->sloeber->apply close.

And this problem needs it's own issue as it is different form the original issue.

MasteringTheMess, 2022-12-12:
well, it seems that the STM32 implementation needs that specific line...
If I comment it out, the compiler complains a lot:

```
'Starting C++ compile'
"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb "@" -c -O3 -g -Wall -Wextra -std=gnu++17 -ffunction-sections -fdata-sections -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -MMD "-IC:\Users\ansay\Documents\_fs-Laser\sloeber_fs-Laser\Test_L4R5" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/avr" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/LL" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/OpenAMP" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/hid" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/cdc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Drivers/STM32L4xx_HAL_Driver/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Drivers/STM32L4xx_HAL_Driver/Src" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/STM32L4xx" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/OpenAMP" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/OpenAMP/virtual_driver" -DSTM32L4xx -DARDUINO=10812 -DARDUINO_NUCLEO_L4R5ZI -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"NUCLEO_L4R5ZI\"" "-DVARIANT_H=\"variant_NUCLEO_L4R5ZI.h\""  -DSTM32L4R5xx -DUSBCON  -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED  "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/Core/Include/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/Include" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\CMSIS\5.7.0/CMSIS/DSP/PrivateInclude" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\cores\arduino" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\variants\STM32L4xx\L4R5Z(G-I)T_L4R7ZIT_L4S5ZIT_L4S7ZIT" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\libraries\SPI\src" -I"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\libraries\SrcWrapper\src" -MMD -MP -MF"Test_L4R5.cpp.d" -MT"Test_L4R5.cpp.o" -D__IN_ECLIPSE__=1 "..\Test_L4R5.cpp" -o "Test_L4R5.cpp.o"
'Starting C compile'
"C:\Sloeber\arduinoPlugin\packages\STMicroelectronics\tools\xpack-arm-none-eabi-gcc\10.3.1-2.3/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_FULL_LL_DRIVER -mthumb "@" -c -O3 -g -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections --param max-inline-insns-single=500 -MMD "-IC:\Users\ansay\Documents\_fs-Laser\sloeber_fs-Laser\Test_L4R5" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/avr" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/LL" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/OpenAMP" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/hid" "-IC:/Sloeber/arduinoPlugin/packages/STMicroelectronics/hardware/stm32/2.3.0/cores/arduino/stm32/usb/cdc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Drivers/STM32L4xx_HAL_Driver/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Drivers/STM32L4xx_HAL_Driver/Src" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/STM32L4xx" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\Sloeber\arduinoPlugin\packages\STMicroelectronics\hardware\stm32\2.3.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\Sloeber\arduinoPlugin\packa [... cut ...]
arm-none-eabi-g++: error: : No such file or directory
make[1]: *** [libraries\SrcWrapper\src\stm32\subdir.mk:70: libraries\SrcWrapper\src\stm32\analog.cpp.o] Error 1
make: *** [makefile:106: all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make -j8 all" terminated with exit code 2. Build might be incomplete.

14:16:44 Build Failed. 4 errors, 0 warnings. (took 6s.982ms)
```

Shall I open a new issue? If so, I need assistance in finding an appropriate title and description since I don't know enough about the specifics of what goes wrong here...

MasteringTheMess, 2022-12-13:
> there is no file name in the error so this is probably caused by a double space :-(

Is it possible to tell in which command/file the double space could be? I can search for it, but due to lack of knowledge don't know where to start looking...

jantje, 2022-12-13:
I provided a workaround in the issue #1534 which is about the opt file problem.
This issue is in regards the indexer being deactivated after a close open project

closed 2022-12-13 by jantje

gfvalvo, 2023-01-10:
This issue is marked "fixed in the nightly". Yet it persists in Stable Product v4.4.1. Does this Stable version pre-date the "nightly" in which the fix was implemented?

jantje, 2023-01-10:
Things marked fixed in nightly are not yet released.
You can download the "nightly" from the actions above. Scroll down for artifacts.
Here is a link to the latest nightly at the time of writing
https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/3659904477

gfvalvo, 2023-01-17:
> Here is a link to the latest nightly at the time of writing
> https://github.com/Sloeber/arduino-eclipse-plugin/actions/runs/3659904477

WOW, this works a lot better than previous versions. Thanks!!!!

I notice it also seems to fix [This Problem](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1364)

The only "Fake Errors" I'm still getting now are those associated with the data structures and functions in the file "esp_camera.h". The code compiles error-free but the Indexer simply doesn't "see" anything in that file. Perhaps it's because of the file's location? This is were it is in the installation directory:

D:\SloeberNightly230109\arduinoPlugin\packages\esp32\hardware\esp32\2.0.6\tools\sdk\esp32\include\esp32-camera\driver\include

Is it possible to also make the Indexer look there?

Thanks again.

[6 more replies]