Sloeber/arduino-eclipse-plugin #1492: updates fail

MasteringTheMess, 2022-07-14:
today, eclipse popped a little window that read that some updates are available.

after clicking on it, a few seconds later I got an error message:

> An error occurred while collecting items to be installed
> session context was:(profile=DefaultProfile, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
> No repository found containing: osgi.bundle,io.sloeber.application,4.4.1.202207120021
> No repository found containing: osgi.bundle,io.sloeber.core,4.4.1.202207120021
> No repository found containing: osgi.bundle,io.sloeber.core.nl1,4.4.1.202207120021
> No repository found containing: osgi.bundle,io.sloeber.ui,4.4.1.202207120021
> No repository found containing: osgi.bundle,io.sloeber.ui.nl1,4.4.1.202207120021
> No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope,1.4.0.202207072214
> No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope.source,1.4.0.202207072214
> No repository found containing: org.eclipse.update.feature,io.sloeber.feature,4.4.1.202207120021
> No repository found containing: binary,io.sloeber.product.executable.win32.win32.x86_64,4.4.1.202207120021
> No repository found containing: org.eclipse.update.feature,org.eclipse.nebula.widgets.oscilloscope.feature,1.4.0.202207072214

is this sloeber related or an eclipse issue?

jantje, 2022-07-14:
>is this sloeber related or an eclipse issue?

probably both :-(
Upgrading in a interdependent plugin world is hard.

MasteringTheMess, 2022-07-14:
> > is this sloeber related or an eclipse issue?
> 
> probably both :-( Upgrading in a interdependent plugin world is hard.

would you recommend using the precompiled nightlys rather then the built-in update function?

jantje, 2022-07-14:
>would you recommend using the precompiled nightlys rather then the built-in update function?

they both have their pro's and cons. If all works fine updating is easy. However ... if the update fails you might be in trouble as updating is so easy you probably didn't make a backup.

The precompiled nightlys are relatively new and are a great addition (thanks @wimjongman to do this)
However if you don't use sloeber naked (I mean you also use other plugins together with sloeber) you will have toinstall these plugins at each upgrade.
Also installed platforms will need to be copied or reinstalled.

There is probably lots more to consider.

wimjongman, 2022-07-14:
You can take a look at the `preferences/install/update/Available software site` this probably contains some old update sites. Try to identify and delete them. Then run the update again.

The following update sites are the only ones that should exist (note that the nightly is optional):

[image]

MasteringTheMess, 2022-07-15:
Hmmm, I removed everything but the ones in your screenshot and ran an update check.

Thereafter some more update sites where added

[image]

One of them was not available

[image]

Thi is weird... it gets added automatically but is not available

MasteringTheMess, 2022-07-15:
> The precompiled nightlys are relatively new and are a great addition (thanks @wimjongman to do this) However if you don't use sloeber naked (I mean you also use other plugins together with sloeber) [...]

at least exporting and importing eclipse preferences worked like a charm for all sloeber relevant settings as well :)

jantje, 2022-07-15:
The babel plugin is only added in case you want to add internationalisation (Sloeber comes in Dutch/Italian so you may want to load the Dutch/Italian eclipse translations)
At the time I did a serious effort to help translate eclipse there was no latest babel url link so I had to add the actual latest address. Seems that now there is a latest url so I should remove the 2019 reference.

This URL management problem is only one of the reasons why I stated

> probably both :-(
> Upgrading in a interdependent plugin world is hard.

BriscoeTech, 2022-07-23:
I was able to fix the babel plugin update sites based on your screenshots in my preferences. 

Now I just need to fix the nebula.widgets.oscilloscope. Any idea where that repository is supposed to be?

[image]

An error occurred while collecting items to be installed
session context was:(profile=DefaultProfile, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope,1.4.0.202207072214
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope.source,1.4.0.202207072214
No repository found containing: org.eclipse.update.feature,org.eclipse.nebula.widgets.oscilloscope.feature,1.4.0.202207072214

jantje, 2022-07-23:
Here
[image]

BriscoeTech, 2022-07-23:
Hmmm, what is strange is I do have that entry in my list, but am still getting that error message

[image]

jantje, 2022-07-23:
> I do have that entry in my list

I know
The message is also shown after validation. So eclipse fins the plugin and then says it doesn't find it.
It must be some bug inside eclipse with multiple update sites containing something wrong I guess
Wim has looked into it seriously and hasn't found it, so I guess it won't be an easy fix.

BriscoeTech, 2022-07-23:
Looks like that specific error message only occurs if I try to update both sloeber and sloeber plugin at the same time

[image]

If I install only sloeber or the sloeber plugin one at a time, it is able to continue and install.

BriscoeTech, 2022-07-23:
Clarification question, when updating from a windows installed version on Slobber v4.4 to a nightly, does the product and the plugin need to be updated, or can you update just the plugin?

jantje, 2022-07-24:
Functionally the plugin is all you need. I however do not know what will happen at the next update  or in other cases.
Also the product is dependent on the plugin so you may get into incompatible plugin version problems

lkapiec, 2022-07-31:
Hi, I have similar issue during installation process of V4.4.1 plugin version file from Marketplace.

```
An error occurred while collecting items to be installed
session context was:(profile=C__Users_Tata_eclipse_embedcpp-2022-06_eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,org.eclipse.nebula.cwt,1.1.0.202207061644
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.cdatetime,1.5.0.202207061644
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope,1.4.0.202207061644
No repository found containing: org.eclipse.update.feature,org.eclipse.nebula.widgets.oscilloscope.feature,1.4.0.202207061644
No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope.source,1.4.0.202207061644
No repository found containing: osgi.bundle,org.hamcrest.core,1.3.0.v20180420-1519
No repository found containing: osgi.bundle,org.junit,4.13.2.v20211018-1956
```

VirtualBox + Win10_21H2_Polish_x64.iso (polish language)

eclipse.vm=C:/Users/Tata/.p2/pool/plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_11.0.15.v20220515-1236/jre/bin\server\jvm.dll

Java only comes from eclipse installation packege (never installed before). I can provide more details if required.

jantje, 2022-08-01:
>Clarification question, when updating from a windows installed version on Slobber v4.4 to a nightly, does the product and the plugin need to be updated, or can you update just the plugin?

You should never install both.
If you want to upgrade an eclipse instance that has the sloeber plugin added to it -> you should install the plugin.
If you want to upgrade a downloaded sloeber -> you should install the product

FYI: I thought I had answered this before but something must have gone wrong.

closed 2024-02-20 by jantje