Sloeber/arduino-eclipse-plugin #1493: Problem with precompiler?

Mike0675, 2022-07-15:
Hi, Im using Sloeber 4.4.0 downloaded from eclipse.baeyens.it.

Im try to compile QNEthernet Library for Teensy 4.1.

Im get :
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c:625:17: error: 'MEMP_IGMP_GROUP' undeclared (first use in this function)

This works in VisualStudio/PlatformIO.

Is it a Problem with the Precompiler? 

Im asked about the Problem at
https://github.com/ssilverman/QNEthernet/issues/16

jantje, 2022-07-15:
Are you sure the code does not compile? I mean, I can understand the indexer gets confused about this code but the build should be fine.
If the build fails; can you provide the output log of the build?

Mike0675, 2022-07-15:
19:10:31 **** Incremental Build of configuration Release for project Alarmanlage_Teensy41 ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c'
'Starting C compile'
"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -c -O1 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD  -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_GERMAN   -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\cores\teensy4" -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\libraries\Wire" -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\libraries\Wire\utility" -I"C:\Sloeber\arduinoPlugin\libraries\Adafruit_MCP23017_Arduino_Library\2.1.0\src" -I"C:\Sloeber\arduinoPlugin\libraries\Adafruit_BusIO\1.11.6" -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\libraries\SPI" -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\libraries\FNET\src" -I"C:\Sloeber\arduinoPlugin\libraries\Functional-Vlpp\1.0.2\src" -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\libraries\SD\src" -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\libraries\SdFat\src" -I"C:\arduino-1.8.19-windows\arduino-1.8.19\hardware\teensy\avr\libraries\Time" -I"C:\Sloeber\arduinoPlugin\libraries\AsyncUDP_Teensy41\1.2.1\src" -I"C:\Sloeber\arduinoPlugin\libraries\AsyncWebServer_Teensy41\1.4.1\src" -I"C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src" -MMD -MP -MF"libraries\QNEthernet\src\lwip\ipv4\igmp.c.d" -MT"libraries\QNEthernet\src\lwip\ipv4\igmp.c.o" -D__IN_ECLIPSE__=1 "C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c"   -o "libraries\QNEthernet\src\lwip\ipv4\igmp.c.o"
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c: In function 'igmp_stop':
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c:179:15: error: 'MEMP_IGMP_GROUP' undeclared (first use in this function)
     memp_free(MEMP_IGMP_GROUP, group);
               ^
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c:179:15: note: each undeclared identifier is reported only once for each function it appears in
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c: In function 'igmp_lookup_group':
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c:258:44: error: 'MEMP_IGMP_GROUP' undeclared (first use in this function)
   group = (struct igmp_group *)memp_malloc(MEMP_IGMP_GROUP);
                                            ^
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c: In function 'igmp_leavegroup_netif':
C:\Sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c:625:17: error: 'MEMP_IGMP_GROUP' undeclared (first use in this function)
       memp_free(MEMP_IGMP_GROUP, group);
                 ^
make: *** [libraries\QNEthernet\src\lwip\ipv4\subdir.mk:69: libraries\QNEthernet\src\lwip\ipv4\igmp.c.o] Error 1
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

19:10:38 Build Failed. 4 errors, 0 warnings. (took 7s.126ms)

jantje, 2022-07-15:
IMHO there are some compile options missing in the compile command and that is why you miss the declarations.
I know some platforms have support for a options file that adds stuff to the command line. I tried to simulate that but (If I remember correctly) decided to simply ignore this construction.
The reason for that is because sloeber allows you to add stuff to the command line

So that may be the root cause. However my search for adding command line options didn't result in a pointer.

jantje, 2022-07-16:
After some investigation I think some new functionality was added in teensyduino that sloeber does not yet support.
Here is a detailed investigation report

PlatformIO teensyDuino and Sloeber all generate a different commandline. 
platformIO succeeds
`arm-none-eabi-gcc -o .pio/build/teensy41/src/lwip/ipv4/igmp.c.o -c -Wall -ffunction-sections -fdata-sections -mthumb -mcpu=cortex-m7 -nostdlib -fsingle-precision-constant -mfloat-abi=hard -mfpu=fpv5-d16 -O2 -DPLATFORMIO=60101 -D__IMXRT1062__ -DARDUINO_TEENSY41 -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=156 -DCORE_TEENSY -DF_CPU=600000000 -DLAYOUT_US_ENGLISH -Iinclude -Isrc -I/my-home-directory/.platformio/packages/framework-arduinoteensy/libraries/Entropy -I/Users/shawn/.platformio/packages/framework-arduinoteensy/cores/teensy4 src/lwip/ipv4/igmp.c`
TeensyDuino succeeds
`"E:\\arduino\\arduuino-1.8.19\\arduino-1.8.19\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10819 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\\arduino\\arduuino-1.8.19\\arduino-1.8.19\\hardware\\teensy\\avr\\cores\\teensy4" "-IC:\\Users\\jan\\Documents\\Arduino\\libraries\\QNEthernet\\src" "-IE:\\arduino\\arduuino-1.8.19\\arduino-1.8.19\\hardware\\teensy\\avr\\libraries\\Entropy" "C:\\Users\\jan\\Documents\\Arduino\\libraries\\QNEthernet\\src\\lwip\\ipv4\\igmp.c" -o nul`
Sloeber fails
```
"E:\arduino\arduuino-1.8.19\arduino-1.8.19\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD  -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10812 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH   -I"E:\arduino\arduuino-1.8.19\arduino-1.8.19\hardware\teensy\avr\cores\teensy4" -I"C:\sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src" -I"E:\arduino\arduino-1.8.15-windows\arduino-1.8.15\hardware\teensy\avr\libraries\Entropy" -MMD -MP -MF"libraries\QNEthernet\src\lwip\ipv4\igmp.c.d" -MT"libraries\QNEthernet\src\lwip\ipv4\igmp.c.o" -D__IN_ECLIPSE__=1 "C:\sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c"   -o "libraries\QNEthernet\src\lwip\ipv4\igmp.c.o"  -DCORE_TEENSY -Iinclude -Isrc
C:\sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c: In function 'igmp_stop':
C:\sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c:178:15: error: 'MEMP_IGMP_GROUP' undeclared (first use in this function)
     memp_free(MEMP_IGMP_GROUP, group);
```
The first thing that strikes me is that Sloeber and platformIO use arm-none-eabi-gcc and teensyDuino uses arm-none-eabi-g++.
I tried to get the platformio command to work but I could not get it to work. When using (note the -i) -I/Users/shawn/.platformio/packages/framework-arduinoteensy/cores/teensy4 src/lwip/ipv4/igmp.c I get a no input file error.
When removing the -I I get the same error.
This is the closest I could get (my current hypothesis is that platformio uses a different arm-none-eabi-gcc)
`"E:\arduino\arduuino-1.8.19\arduino-1.8.19\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -o "libraries\QNEthernet\src\lwip\ipv4\igmp.c.o" -c -Wall -ffunction-sections -fdata-sections -mthumb -mcpu=cortex-m7 -nostdlib -fsingle-precision-constant -mfloat-abi=hard -mfpu=fpv5-d16 -O2 -DPLATFORMIO=60101 -D__IMXRT1062__ -DARDUINO_TEENSY41 -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=156 -DCORE_TEENSY -DF_CPU=600000000 -DLAYOUT_US_ENGLISH -I"C:\sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src" -IE:/arduino/arduino-1.8.15-windows/arduino-1.8.15/hardware/teensy/avr/libraries/Entropy C:/sloeber/arduinoPlugin/libraries/QNEthernet/0.14.0/src/lwip/ipv4/igmp.c`

The teensyDuino command works fine in Sloeber
```
"E:\\arduino\\arduuino-1.8.19\\arduino-1.8.19\\hardware\\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=156 -DARDUINO=10819 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH  -I"E:\arduino\arduuino-1.8.19\arduino-1.8.19\hardware\teensy\avr\cores\teensy4" -I"C:\sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src" "-IE:\\arduino\\arduuino-1.8.19\\arduino-1.8.19\\hardware\\teensy\\avr\\libraries\\Entropy" C:/sloeber/arduinoPlugin/libraries/QNEthernet/0.14.0/src/lwip/ipv4/igmp.c  -o nul
'Finished building: C:\sloeber\arduinoPlugin\libraries\QNEthernet\0.14.0\src\lwip\ipv4\igmp.c'
```
Investigating the platform.txt from Teensy I see Teensyduino from @PaulStoffregen uses the `recipe.preproc.macros` recipe and not the `recipe.c.o.pattern` recipe. I also notice a `recipe.preproc.includes` which is also unknown to me.

Next step is to find out documentation of these new recipes and implement their execution rules.

notes
1) I have no platformio so I use the commandline from https://github.com/ssilverman/QNEthernet/issues/16#issuecomment-1185989132
2) in Sloeber I use arduuino-1.8.19 and arduino-1.8.15-windows but they should point to the same stuff as they are all teensy related and the same teensy version is there.

jantje, 2022-07-16:
I messed up. I took the teensyduino discovery phase command for the actual build command.

I found it :-D
The root cause is that sloeber by default is configured to add `#pragma once` to all header files.
However this is not compatible with QNEthernet because mem_std.h is included explicitly multiple times with different preconditions.

Workaround
1) turn of the option to add pragma once (all installed libraries will need to be removed and reinstalled) which has the drawback that lots of ino examples will not compile properly
2) remove the pragma once from the bottom of the mem_std.h file.

jantje, 2022-07-16:
Closing the issue as a known limitation with a known workaround.

closed 2022-07-16 by jantje