Sloeber/arduino-eclipse-plugin #1494: Entries in 'append to link' have no effect

JojoS62, 2022-07-23:
**Describe your environment**   
Sloeber product   
Sloeber version number  4.4.0
windows 10

**Describe the bug**   
additional link option in project settings-Arduino-append to link are not used.

**To Reproduce**   
Steps to reproduce the behavior:   
1. Go to 'Project/Properties/Arduino'
2. Add some weird name in 'append to link'
3. run build
4. See no error

** Problem

wrong env variable in link command, needs to be 'sloeber.extra.link'

[image]

Credits go to user  frutziputzi in mikrocontroller.net who discovered the reason for this bug.

jantje, 2022-07-23:
Great issue description :-) thanks
Workaround: modify the Arduino combiner command line patters (replace _ with .)

JojoS62, 2022-07-23:
thanks also for the fast response,
Yes, that workaround did it already. But also adding an additional object file became obsolete. Changing from additional objcopy to using an assembler source was more compliant with the standard build system.
And it seems that this feature wasn't used by someone for long time :)

jantje, 2022-07-23:
I don't think this feature was ever used.
When I added the code to add to the c/c++ code line (which I use) I added all the other arduino code commands extensions in one go; for compliance and for "god knows who will ever need this"

closed 2022-09-16 by jantje