Sloeber/arduino-eclipse-plugin #1495: serial terminal does not work after board has been de- and reconnected during open session

MasteringTheMess, 2022-07-27:
I use nigthly 4.4.1.202207151257

scenario is:
sloeber is running, STM32 Nucleo board is connected and serial monitor connected to the corresponding COM-port, everything works fine.

Now I disconnect the board and on reconnect the serial monitor does not display anything at all. Resetting the board does not help.
One has to disconnect the COM-port in serial monitor and reconnect it.

Don't know whether this applies only to STM32 boards, actually I have nothing else to test at hand...

And I know this is quite a minor issue with an existing (and easy) workaround, but as I didn't realize it was a problem with the serial monitor at once, I spent some time looking at the wrong place.
Maybe reporting it here helps other people that run into the same behavior?

jantje, 2022-07-27:
>Now I disconnect the board and on reconnect the serial monitor does not display anything at all. Resetting the board does not help.
One has to disconnect the COM-port in serial monitor and reconnect it.

as far as I am aware this is standard windows behaviour.
I assume arduino ide has the same behaviour

MasteringTheMess, 2022-07-27:
Checked that, yes, Arduino IDE has the same behavior.
Prior to posting, I checked with teraterm serial terminal, where hotplugging the board is supported. No idea how they do it but the fact of having used it for years seems to have influenced my expectancies ;-)

jantje, 2022-07-27:
>I checked with teraterm serial terminal, where hotplugging the board is supported

I'm not claiming Sloeber (terminal) is the best implementation ever. But it works, has some nice Arduino specific features and is not really buggy.

I'll close this as a known limitation. Feel free to fix though I won't be able to help out on the coding for this fix as I wouldn't know how.

closed 2022-07-27 by jantje