Sloeber/arduino-eclipse-plugin #1497: Error in parsing Platform and Boards

tom3f, 2022-09-16:
Eclipse 4.25
Sloeber 4.4.1 Plugin

There is an error when trying to parse this file in Platform and Boards:
`https://raw.githubusercontent.com/sparkfun/Arduino_Boards/main/IDE_Board_Manager/package_sparkfun_index.json
`

[image]

Logfile: 
[log.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/9587224/log.txt)

jantje, 2022-09-16:
I have that json file in my work setup I updated today to 4.4.1 and I installed a sparkfun avr platform version without problems.
So my first thought is that the download may have corrupted the file.
The downloaded file is at [sloeber install]/arduinoPlugin/package_sparkfun_index.json

jantje, 2022-09-16:
Also the log indicates a platform without versions

tom3f, 2022-09-16:
No, the downloaded file is not corrupted and in an older eclipse version there is no problem with parsing in Platforms and Boards. In eclipse 4.25 (clean fresh installation) i copied the board files directly into /arduinoPlugin/packages as quick workaround. 

But maybe there is an incompatibilty with the jre in 4.25:
```
-vm
plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.4.v20220903-1038/jre/bin
```

In the older eclipse installation there was:
```
-vm
plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_18.0.1.v20220515-1614/jre/bin
```

jantje, 2022-09-16:
So you took the newest eclipse and added sloeber to it; then you had the problem
If you take an older eclipse and add sloeber to it; then you do not have the problem

I remember reading something about some json changes in the latest eclipse. It could be this. I should check whether I can reproduce.

magic-icicle, 2024-09-09:
I am facing the same problem with Eclipse 4.32. How can I resolve it?

jantje, 2024-09-09:
>I am facing the same problem with Eclipse 4.32. How can I resolve it?

By creating a new issue and providing the information requested for a new issue.