Sloeber/arduino-eclipse-plugin #1501: Arduino.h not found

Michael580519, 2022-10-01:
Hi to all,
i have Sloeber 4.4.1 with Eclipse 4.24 running on Windows 10, board NodeMCU 1.0 (ESP-12E)  and get the message „Arduino.h not found“ during compile. Of corse it is in the core-path within the project.
18:32:40 **** Incremental Build of configuration Release for project Klimaduino_V2.2E ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\MQTToverWiFi.cpp'
'Starting C++ compile'
"C:\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-IC:\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/include" "-IC:\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/lwip2/include" "-IC:\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\3.0.2/tools/sdk/libc/xtensa-lx106-elf/include" "-ID:\Micha\Projekte\SmartHome\Sloeber\Klimaduino_V2.2E\Release/core" -c -w -Werror=return-type  -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10812 -DARDUINO_ESP8266_NODEMCU_ESP12E -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU_ESP12E\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO  -DESP8266   -I"D:\Micha\Projekte\Arduino\libraries\DHT_sensor_library" -I"C:\Sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src" -I"C:\Sloeber\arduinoPlugin\libraries\PubSubClient\2.8.0\src" -MMD -MP -MF"MQTToverWiFi.cpp.d" -MT"MQTToverWiFi.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\MQTToverWiFi.cpp"   -o "MQTToverWiFi.cpp.o"
In file included from ..\MQTToverWiFi.h:10,
                 from ..\MQTToverWiFi.cpp:10:
C:\Sloeber\arduinoPlugin\libraries\PubSubClient\2.8.0\src/PubSubClient.h:10:10: fatal error: Arduino.h: No such file or directory
   10 | #include "Arduino.h"
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [subdir.mk:44: MQTToverWiFi.cpp.o] Error 1
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

18:32:41 Build Failed. 2 errors, 0 warnings. (took 655ms)
 
Thanks for your help,
Michael

[Hi to all.docx](https://github.com/Sloeber/arduino-eclipse-plugin/files/9690799/Hi.to.all.docx)

jantje, 2022-10-01:
The core and variant folder are not in the build command.
There should be a core folder in the root of your project and that core folder should contain a core folder (and very likely also a variant folder)
If those are not there something went wrong somewhere.
Can you provide a screenshot of the project with the core folder opened?
Can you provide the error log (window->show view->other-error log then in the error log view right click and export

Michael580519, 2022-10-01:
Hi Jantje,
i´m so happy for your help. Please find attached screenshots and an export oft the error log. 
Sometimes I create a variant of my project by copying it. It´s possible, that some informations are lost by that operation. The core folder in the directory path on the hard disk is empty.
Michael
[MySloeberProblem_V1.docx](https://github.com/Sloeber/arduino-eclipse-plugin/files/9691163/MySloeberProblem_V1.docx)

jantje, 2022-10-01:
Strange
I see no obvious reason in the info why the core and variant folders have not been added to the compile command.
> The core folder in the directory path on the hard disk is empty.

That is normal. There are only eclipse linked folders and these are not reflected on disk. The same counts for the Libraries folder.

>Sometimes I create a variant of my project by copying it. It´s possible, that some informations are lost by that operation.

Yes. But normally nothing project properties->sloeber->apply close can't fix. Have you tried this?

jantje, 2022-10-01:
can you check
project properties->C/C++general->Paths and symbols->GNU C++
It should look something like this
[image]

Michael580519, 2022-10-02:
"project properties->C/C++general->Paths and symbols->GNU C++" checked and lokks like this now:
[image]
       "But normally nothing project properties->sloeber->apply close can't fix. Have you tried this?"
Yes, this was most helpful. After a "Clean project" all goes well now: "14:59:07 Build Finished. 0 errors, 0 warnings. (took 1m:9s.67ms)" Unfortunately the cause could not be found, but still I will have a beer tonight ;-) Thanks very mouch!

[MySloeberProblem_V2.docx](https://github.com/Sloeber/arduino-eclipse-plugin/files/9692557/MySloeberProblem_V2.docx)

jantje, 2022-10-02:
What you are showing is
   "project properties->C/C++general->Paths and symbols->Assembly" 
not
   "project properties->C/C++general->Paths and symbols->GNU C++" 
It is unlikely but they may have been different

project properties->sloeber->apply
is the first thing to do when there are wierd problems. It is kind of a sloeber clean command

>Unfortunately the cause could not be found

I think it is related to copying the project. I never tested that and if I did I would have naturally used "project properties->sloeber->apply' and called it good if that fixes the problem.
It would be great if you could try to reproduce the problem so we can at least document how to get into this issue.
Enjoy the beer

Michael580519, 2022-10-02:
Please find attached my further experiments to reproduce the problem. It seems, that copying a project is ok, so far you comply some rules (project -> clean and project properties->sloeber->apply). But if you want to rename the project, trouble occurs. The new creation of sloeber.ini.cpp fails after I deleted it. See the error messages at the end of the document.
[MySloeberProblem_V3.docx](https://github.com/Sloeber/arduino-eclipse-plugin/files/9693074/MySloeberProblem_V3.docx)

jantje, 2022-10-03:
[image]
These have nothing to do with Sloeber.
[image]
I assume the download of the json was not successful causing the parsing to fail. I should add checksum validation to sloeber.
I also disable the download at start on my installs to reduce the impact of potential download fails
[image]
I'm going to try the rename of a project to see whether I can reproduce the issue.

jantje, 2023-01-16:
Closing this as a one of.

closed 2023-01-16 by jantje

6v6gt-duino, 2023-02-04:
I'm using Sloeber 4.4.1. I get this "Arduino.h: No such file or directory" error frequently, when I create a new version of a project based on a copy of an old one, and usually resort to some or all of the following "work arounds"

- cleaning the newly copied project
- deleting the "releases" folder
- the "project properties->sloeber->apply" trick
- closing sloeber and reopening it

After a few tries, the problem goes away and it does not reappear in the same project.

This is the error which occurs during the verify.
..\Main.h:14:10: fatal error: Arduino.h: No such file or directory

I'm using an ESP32 board LOLIN D32 PRO  and the project has over 20 files.

I'm reasonably confident I can re-create the problem if required.

Michael580519, 2023-02-04:
Hi 6v6gt-duino,
I can only describe what the real reason in my problem was: My backup software deletes every empty folder every night. So **the next day the core and libraries folders and subfolders (which are always empty) were gone and Arduino.h was not found.** I had to apply the workarounds and to reinstall the necessary libraries. So i created empty dummy.txt-files in the folders described above, again applied the "workarounds" and my problem was solved.
I´m not sure if this solves your problem. It is important to check all hints from jantje.

jantje, 2023-02-04:
@Michael580519 
Thanks for your input. I would never have thought of some software deleting empty folders as a root cause for the missing "Arduino.h" error .
But deleting the core and libraries folders (which are empty) will indeed create problems within Sloeber and missing "Arduino.h" error is guaranteed.
Maybe Sloeber should create a "doNotDeleteThisFolder.txt" files in these folders.

6v6gt-duino, 2023-02-05:
Thanks @Michael580519 and @jantje for the replies.

OK. Then that described in the problem report had a very specific cause. I don't rule out something related to my environment (Virus checker etc.) however I also think it could be something to do with the structure of my project.

When it is in this state and I copy Ardino.h from the core into my project then it starts complaining about another missing file. I would guess that it is probably something related to my project which has a sprawling collection of files (.h and .cpp)  statements with multiple references to each other. I could believe that this causes a problem when it tries to resolve out the redundant include statements probably having to build a huge table or path somewhere. 

Anyway, I will restructure the project at the next opportunity to see if the issue goes away.
It looks like it is not a common problem, and persistently applying the various work arounds eventually works, so it probably does not warrant any further action at the moment.

jantje, 2023-02-05:
Another tip: Do not just copy projects.
In the project you want to copy:
Use the "project properties->sloeber->other->share in version control" to put your compile options in a file called sloeber.cfg (if you have multiple configs do this for all configs you want to copy)

Copy your code (inclusive the sloeber.cfg file but exclusive the libraries and core) to the newly created project (not sloeber just a empty eclipse project like general)
Select the project in the project explorer; select file->new->other->Sloeber->convert to sloeber project