Sloeber/arduino-eclipse-plugin #1502: "Permission denied" on build...

MangyDogUK, 2022-10-04:
[image]

Ive recently started doing some more work again and accidently upgraded the IDE... And it stopped working...

I decided to download sloeber application and did a fresh install. 

I added my Arduino_Stm32 hardware list to sloebers hardware list and tried to build.

But I get this error.

I also get this error after I updated to the nightly build. 

I have a feeling ive had this problem before. And it might have been fixed by either adding some lines to the sloeber ini file or adding a line to the build argument list. 

Though im not sure...

Any ideas?

jantje, 2022-10-04:
Have you tried right clicking the project->properties->sloeber (verify if all is well) apply and close
This is needed after most upgrades and the first step to do in case of wierd problems

MangyDogUK, 2022-10-04:
Yes just double checked the board/core setup all seems fine there
(reapplied)

jantje, 2022-10-04:
Have you deleted the release folder?

MangyDogUK, 2022-10-04:
I didnt, just tried deleting and rebuilding. Still getting the error

jantje, 2022-10-04:
My eye just  droped on
>I added my Arduino_Stm32 hardware list to sloeber hardware list and tried to build.

I guess you mean this is a private hardware platform not installed by the boardsmanager. In which case there is no platform json and there is one (or more) environment variable missing in your environment.

To fix:
Look for the recipe.cpp.o.pattern in the platform.sloeber.txt file. it typically looks like
`recipe.cpp.o.pattern="${compiler.path}${compiler.cpp.cmd}" ${compiler}....`
The compiler.path In the esp8266 compiler.path is defined in the same platform.sloeber.txt as (which is common)
`compiler.path=${runtime.tools.xtensa-lx106-elf-gcc.path}/bin/`
The declaration of runtime.tools.*.path variables get resolved based on the json file which you do not have.
By creating this environment variable yourself you can tell sloeber where to find the compiler.

You can create the environment variable at the level of os, workspace (windows->preferences->C/C++->build->environment) , project configuration (right click project->properties->C/C++->build->environment). 

Based on the little info I have from you I would advice to do it at the level of the workspace
You can also modify the platform.txt and declare it there. Do not do it in platform.sloeber.txt as this is seen as a temporary file

MangyDogUK, 2022-10-04:
Forcing the path to the /bin/ for the embedded tool chain worked... 

Some more issues to deal with, but dealing :D 

Thanks

MangyDogUK, 2022-10-04:
Fixed. make sure the environmental path is correct

closed 2022-10-04 by MangyDogUK