Sloeber/arduino-eclipse-plugin #1504: Error Compiler message: fatal error: no input files

Alf-R, 2022-10-10:
Hello, 
my project failes to compile.
The message is:

C:/prg/Sloeber/arduinoPlugin/tools/make/make --no-print-directory main-build
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\prg\Sloeber\arduinoPlugin\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE= [ ...... ] -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"  --> all in all 8329 characters
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
make[1]: *** [subdir.mk:23: sloeber.ino.cpp.o] Error 1
make: *** [makefile:100: all] Error 2
"C:/prg/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

18:03:42 Build Failed. 2 errors, 0 warnings. (took 3s.368ms)

As other ESP32 projects are compiled and running fine I assume the problem is not my software.
The compare Project has 215 characters for compile instruction and runns fine.

Is the high amount of the 8329 characters for the compiler too much?
Is there a buffer to increase?

Thank you for your help :-)

Sloeber Version 4.4.1 (V4.4.1_win64.2022-09-05_06-48-31.zip) - downloaded as full instalation 
Windows Windows 10 Pro
Arduino 1.8.16 installed

ESP32 Dev Module project

jantje, 2022-10-10:
I agree it looks like a command line length issue.
Though the make the new sloeber uses is supposed not to have this issue.

Alf-R, 2022-10-11:
> I agree it looks like a command line length issue. Though the make the new sloeber uses is supposed not to have this issue.

Hi jantje, thank you for your feedback.
A question would be, if I'm the only one with this issue or is my project too big?
Is there anything I can do to?
What's the maximum length of the compile command line?

The command line for this project at Arduino IDE has 9662 Characters (even if I replace "\\" to "\" 8894 Characters remaining) and it works fine ....

jantje, 2022-10-11:
You are not the only one with this problem.
The root cause of the problem is a windows limitation https://learn.microsoft.com/en-us/troubleshoot/windows-client/shell-experience/command-line-string-limitation
However mostly the problem occurs at link/archiving time as the command line Sloeber generates for these actions is dependent on the number of files in the project.
I'm kind of surprised this is happening for the compile of 1 file. Are you having lots of libraries in your project?

Alf-R, 2022-10-11:
with only 1 File it works but my working project has all in all 50 Files and several libary's (OTA / SerialBT / Wire.h / ...)
Just to give a Idea about the size: Arduino say the scetch is using 1407978 Bytes .....

jantje, 2022-10-11:
As to your original post Sloeber.ino.cpp fails because of a to long command line.
Sloeber.ino.cpp compilation is fixed size + the includes. 
The max size of the command is 8191
Assuming the fixed size is 1191 that leaves 7000
The includes are +- equal to the number of libraries.
Assuming 200 for each include/library that makes 35 libraries.
If you have less than 35 libraries I would advice to verify the command line to see why it is so big.

Alf-R, 2022-10-11:
I will check this, thank you

Alf-R, 2022-10-12:
Hi jantje, I reduced the number of characters by changing the path (from C:\prg\Sloeber\  to c:\S\ ) to 7559 characters but it still doesn't work.
Is there a option to change the path for the hardware lib's ?
example:
"-IC:\S\arduinoPlugin\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp-face"
to
"-IC:\S\aP\pack\esp32\hw\esp32\1.0.6/tools/sdk/include/esp-face"
This could reduce the characters to 5948 but I have no idea if this would be enouhght

jantje, 2022-10-12:
>Hi jantje, I reduced the number of characters by changing the path to 7559 characters but it still doesn't work.

Strange. you are now below the windows limit which makes that hat should be fixed. I see 2 options
1) you are not using the latest make due to upgrading? stop sloeber ->delete []/arduinoPlugin/tools/make ->restart sloeber
2) there is a quoting issue -> to test move ` -x c++ "..\sloeber.ino.cpp"`

AS this is a command line you can always run the command from the command prompt of the os when you are in the release folder of the project

>Is there a option to change the path for the hardware lib's ?

I have been thinking about doing so for some time but I didn't do this one yet.
There is a option to define the environment variable SLOEBER_HOME (in eclipse or os) as c:\; then you will have c:\arduinoPlugin

Can you inform me on why the command is so long?

Alf-R, 2022-10-12:
Hi jantje, looks like the problem is in front of the screen ;-)
I tried a similar huge project with 7356 Characters and it was compiled fine.
I was just worried that I must include my *.cpp File into the *.h File - but it works.
The difference is that I have too much *.ino Files in my problem- project.
So for me the issue is closed, thank you for your help and ideas

closed 2022-10-12 by Alf-R