Sloeber/arduino-eclipse-plugin #1512: Can't open Sloeber on MacOS Ventura

0x3alex, 2022-11-07:
Hey there!
I wanted to use Sloeber V4.4.1 on my M2 MacBook Air. I already enabled "applications from everywhere" but Slober just wont launch. I either get, that the executable is damaged or that it just can't be opened. I installed java 11 already. Any ideas?

jantje, 2022-11-07:
Have you seen the mac specific install instructions here http://eclipse.baeyens.it/stable.php?OS=MacOS ?

> Due to mac new security approach a simple download and extract will not work.
> 
> Follow following steps to install.
> 
> 1)download with mac browser.
> 
> 2)unpacked the sloeber ide zip file with the macOS native Archive Utility. Not in the download folder!!!
> 
> 3)in the folder where you extracted Sloeber run: sudo xattr -r -d com.apple.quarantine Sloeber.app
> 
> Read the info on this site at: install->start here-> MacOS comments

0x3alex, 2022-11-07:
Yes, i tried that already. Same results sadly :/

0x3alex, 2022-11-07:
Alright. Thanks tho. Hoping for a response, since i need it for school

judge2005, 2022-11-08:
I had several issues before I finally got it running. One thing I remember is I installed Oracle's JDK17, then did _sudo xattr -r -d com.apple.quarantine Sloeber.app_. If you edited info.plist to specify the JDK to use, revert that change.

0x3alex, 2022-11-08:
So i did that, and it gives me this:

I installed openjdk17 with brew, since i dont want to create some oracle account. 
`java --version` gives me the correct result.

`The application /Users/alex/Documents/Sloeber.app cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x6000002c6e50 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}`

jantje, 2022-11-08:
As far as I know (I can't test as I don't have a mac) the Sloeber product no longer needs a preinstalled java on mac (and linux).
I'm no expert on this change but IMHO it works as follows: when eclipse starts it downloads java and goes with that java.
Maybe something in this process fails?

I googled `Error Domain=NSPOSIXErrorDomain Code=111` and it seems more people are struggling with a similar problem on mac. Is the app a script?

As we are thinking in java terms.. A good test my be to try to install "eclipse CDT" and add sloeber plugin. See where you get?

jantje, 2022-11-08:
The java addition happened in issue #1376

0x3alex, 2022-11-08:
Okay... i am now trying to install it with eclipse

0x3alex, 2022-11-08:
Found the solution for starting it:

I ran `chmod +x Sloeber.app/Contents/MacOS/sloeber-ide` and then `open -a Sloeber`. 
its now running but everything just throws errors that files dont exist and so on...

jantje, 2022-11-08:
These errors are because sloeber can not access the internet.

jantje, 2022-11-08:
Maybe it is disk write error and not internet.
Seems the archive has been downloadedbut can not be extracted

0x3alex, 2022-11-08:
Yeah so i gave up... i worked out support in arduino ide :/ sorry

closed 2022-11-10 by jantje