Sloeber/arduino-eclipse-plugin #1513: combiner syntax error

wolfib77, 2022-11-11:
Hello,
I'm trying to migrate from sloeber 4.4 to 4.4.1. With 4.4.1 the combiner shows a syntax error
`'Starting combiner'
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -O2 -Wl,--gc-sections,--relax,--defsym=__rtc_localtime= "-TC:/Program Files (x86)/Arduino/hardware/teensy/avr/cores/teensy3/mk20dx256.ld"  -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -o "C:\CerberusDoorAccessControl\Firmware441T\Cerberus\Release/Cerberus.elf"    .\CardMgmtThread.cpp.o .\CircularBuffer_TEST.cpp.o .\DRV8870.cpp.o .\EthernetThread.cpp.o .\InterThreadCom.cpp.o .\KonnektingThread.cpp.o .\MotorControl.cpp.o .\Peripherals.cpp.o .\RelayOutput.cpp.o .\sloeber.ino.cpp.o  .\libraries\Time\DateStrings.cpp.o .\libraries\Time\Time.cpp.o  .\libraries\TeensyThreads\TeensyThreads-asm.S.o .\libraries\TeensyThreads\TeensyThreads.cpp.o  .\libraries\TeensyMAC\TeensyMAC.cpp.o  .\libraries\SPI\SPI.cpp.o  .\libraries\LibTelnet\libtelnet.c.o  .\libraries\KONNEKTING_Device_Library\src\DebugUtil.cpp.o .\libraries\KONNEKTING_Device_Library\src\KnxComObject.cpp.o .\libraries\KONNEKTING_Device_Library\src\KnxDevice.cpp.o .\libraries\KONNEKTING_Device_Library\src\KnxTelegram.cpp.o .\libraries\KONNEKTING_Device_Library\src\KnxTpUart.cpp.o .\libraries\KONNEKTING_Device_Library\src\KonnektingDevice.cpp.o  .\libraries\Ethernet\src\utility\w5100.cpp.o  .\libraries\Ethernet\src\Dhcp.cpp.o .\libraries\Ethernet\src\Dns.cpp.o .\libraries\Ethernet\src\Ethernet.cpp.o .\libraries\Ethernet\src\EthernetClient.cpp.o .\libraries\Ethernet\src\EthernetServer.cpp.o .\libraries\Ethernet\src\EthernetUdp.cpp.o .\libraries\Ethernet\src\socket.cpp.o  .\libraries\Encoder\Encoder.cpp.o  .\libraries\EEPROM\EEPROM.cpp.o  .\libraries\DoorOpener_Library\AES128.cpp.o .\libraries\DoorOpener_Library\Classic.cpp.o .\libraries\DoorOpener_Library\DES.cpp.o .\libraries\DoorOpener_Library\Desfire.cpp.o .\libraries\DoorOpener_Library\PN532.cpp.o .\libraries\DoorOpener_Library\Utils.cpp.o  .\libraries\Bounce2\Bounce2.cpp.o    C:\CerberusDoorAccessControl\Firmware441T\Cerberus\Release/arduino.ar  "-LC:\CerberusDoorAccessControl\Firmware441T\Cerberus\Release" -larm_cortexM4l_math -lm -lstdc++
c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe:--defsym:1: syntax error
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:115: Cerberus.elf] Error 1
"C:/Sloeber441/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.`

In contrast with sloeber 4.4 there is a timestamp after `rtc_localtime=`:
`'Starting combiner'
"C:/Program Files (x86)/Arduino/hardware/teensy/../tools/arm/bin/arm-none-eabi-gcc" -O2 -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=1668111525 "-TC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/mk20dx256.ld"`

How can this be fixed?
Kind regards

jantje, 2022-11-11:
Strange.
I tried it here and here __rtc_localtime is defined. 
The value is provided by https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/toolchain/SloeberProjectVariableSupplier.java 
Which has not changed since december 2020. I think something is wrong in your project. I would start by creating a new project and see whether the value is know there. 
If so we know it is a project specific problem. Most likely a project setting in the environments variables that overwrites the value.

'Starting combiner'
"E:\arduino\arduuino-1.8.19\arduino-1.8.19\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -O3 -flto -fno-fat-lto-objects -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=1668182012 "-TE:/arduino/arduuino-1.8.19/arduino-1.8.19/hardware/teensy/avr/cores/teensy3/mk20dx128.ld" -fuse-linker-plugin -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -o "C:\workspaces\embedded\ggggg\Release/ggggg.elf"    .\ggggg.cpp.o    C:\workspaces\embedded\ggggg\Release/arduino.ar  "-LC:\workspaces\embedded\ggggg\Release" -larm_cortexM4l_math -lm -lstdc++
'Finished building: ggggg.elf'
' '

jantje, 2022-11-28:
Closing as one of

closed 2022-11-28 by jantje