Sloeber/arduino-eclipse-plugin #1514: Cant' check a checkbox in Platforms and Boards

freeride01, 2022-11-11:
Hi,
I downloaded the latest Sloeber V4.4.1 for Linux and installed it on Fedora. I want to add ESP32 support to it.
In Preferences-> Sloeber -> Third party index url's I entered " https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json"
Then in Platforms and boards I see the "package_esp32_index.json" option, but cant' select *any* option inside the package box. Any idea why?

jantje, 2022-11-11:
You have to click the > untill you get to versions.
[image]

freeride01, 2022-11-11:
Thanks, it works now.
I have another issue, however. The project I import uses the NimBLE stack, so I installed it, but when I try to build the project , I get `NimBLEDevice.h: No such file or directory`
The library shows as installed in Library Manager, but it seems it's not added to the project for some reason.

jantje, 2022-11-11:
great have fun

closed 2022-11-11 by jantje