Sloeber/arduino-eclipse-plugin #1515: Problem compiling Sloeber 4.4.1 and Teensy

CNathan74, 2022-11-21:
Hello everyone !
I have a problem with the new version of sloeber 4.4.1. When I want to compile a teensy project, I have this message :
Here is all the information you need:

> "C:\\Eclipse\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"{teensytools.path}precompile_helper" "{runtime.platform.path}/cores/{build.core}" "{build.path}" "{compiler.path}{build.toolchain}{build.command.g++}" -x c++-header {build.flags.optimize} {build.flags.common} {build.flags.dep} {build.flags.cpp} {build.flags.cpu} {build.flags.defs} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DF_CPU={build.fcpu} -D{build.usbtype} -DLAYOUT_{build.keylayout} "-I{runtime.platform.path}/cores/{build.core}" "{build.path}/pch/Arduino.h" -o "{build.path}/pch/Arduino.h.gch"
process_begin: CreateProcess(NULL, {teensytools.path}precompile_helper {runtime.platform.path}/cores/{build.core} {build.path} {compiler.path}{build.toolchain}{build.command.g++} -x c++-header {build.flags.optimize} {build.flags.common} {build.flags.dep} {build.flags.cpp} {build.flags.cpu} {build.flags.defs} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DF_CPU={build.fcpu} -D{build.usbtype} -DLAYOUT_{build.keylayout} -I{runtime.platform.path}/cores/{build.core} {build.path}/pch/Arduino.h -o {build.path}/pch/Arduino.h.gch, ...) failed.
make (e=2): Le fichier spécifié est introuvable.
make: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 2
"C:/Eclipse/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

OS : Windows 11
Sloeber : 4.4.1
Teensy target : 4.1
Tensyduino : 1.57

I feel like Sloeber can't override path variables.
Thanks

jantje, 2022-11-21:
How did you install the teensy platform?

CNathan74, 2022-11-21:
My procedure:
- I install Arduino ide
- I install teensyduino
- I install Sloeber
- I configure the Private Hardware Path to the teensy folder in Arduino (as in this [tutorial](https://eclipse.baeyens.it/how_to.shtml#/i))

I have already used the same procedure on another PC with Sloeber version 4.3 and it worked.

CNathan74, 2022-11-22:
Other information: I tested to compile the code with Arduino ide, and it works. But not on Sloeber

CNathan74, 2022-11-23:
Hi.
I tested your solution and it works!
I encountered a problem: when I want to use a teensy cores file, Sloeber does not detect it during the compilation (example I wanted to use the FS.h file which is in the teensy cores, and it’s not added to the project's core file at compile time).
To counter the problem, in addition to the json file, I install Adruino IDE 1.18, I install teensyduino and I link in Slober to C:\Program Files (x86)\Arduino\hardware\teensy\avr.
It's not the optimal solution, but for now I'll go with it.

Thank you for your help and your work, you are the king!

jantje, 2022-11-27:
I'm not sure what problem you ran into but the FS.h file is available here on this test system with teensy installed from the json

CNathan74, 2022-11-28:
Have you tried compiling a Teensy project that uses the FS.h file (or another in the same cores folder)? 
That's my problem, the file is present, but when compiling, Sloeber does not add it to the core folder of the project. Compilation failed because it cannot find the file.

jantje, 2022-11-28:
I've been running some tests with Teensy from json and the arduino libraries.
All seems to work fine.
I think we can close this issue. @CNathan74 what do you think?

CNathan74, 2022-11-28:
I just redid a full install with the json file only and it works!

Thank you very much for your help

closed 2022-11-28 by CNathan74