Sloeber/arduino-eclipse-plugin #1516: Major house keeping; Java 17 ; sloeber target; product; removal application

jantje, 2022-11-22:
I have been struggling to get Sloeber to compile in eclipse (maven works fine) on a new system and after I imported CDT plugins in my Sloeber build eclipse workspace all went banana's there as well.
So I decided to do some house keeping:
1) move to a consistent java version defined in the workspace. I used 17 as I had that one installed.
Some project had a java version different from the workspace version and that has cost me quite a lot of time to figure out why things didn't work.
I'd prefer the setting to be 1 for al projects in the repository but I do not know how to do that.
Still there is a java version in all manifest files that needs manual updates :-(
And also in the build path
2)I think I made the application plugin unneeded (so I removed it). I have always been confused about the difference between application and product and I now think the application is not needed.
Part of the confusion is because the product icons are in the application plugin so I moved them to the product plugin
3)I changed the sloeber.target so that compilation in eclipse no longer is dependent on installed plugins.
However this didn't go as easy as I wanted. I agree that my lack of knowledge is for sure part of my struggles but something had to be done.
Anyway I probably added way to much (amongst others because I had some CDT requirements in my head when doing this) but also because I needed to add org.junit for the test  plugins because it is a workbench wide setting.
Also finding out that eclipse.application contains org.eclipse.ui.ide.workbench turned out time consuming. FYI build is fine but as soon as you want to start a debug session ....

wimjongman, 2022-11-22:
You should work in a branch my friend, and only push to master when everything is working fine. That is why we have Github actions.

I suggest that you hard reset to the point where everything was still working and then force-push to the repo.

jantje, 2022-11-22:
I agree but the thing is: The locally maven build works; also eclipse build works. It is only the action maven build that fails.
And This was supposed to be a small change. I did all the work locally and then pushed; then I remembered the maven build ....

I already have 2 big branches :-( didn't feel like a third one for this small change :-(

wimjongman, 2022-11-22:
It could compile locally because the missing software (the application) gets installed in your local maven repo. The application is the starting point of the IDE. It determines the look and feel of the application and gets control when the IDE is started. A product is a set of features and some other goodies. Inside the product, you define the application.

If you think you can delete the application, also remove it from the module list in the master pom.

> I already have 2 big branches :-( didn't feel like a third one for this small change :-(

🏆 Congratulations, you won the smoesjes award!

jantje, 2022-11-22:
>If you think you can delete the application, also remove it from the module list in the master pom.

I did do that.
I changed the application to org.eclipse.ui.ide.workbench

closed 2022-11-27 by jantje