Sloeber/arduino-eclipse-plugin #1517: Actions fail to build

jantje, 2022-11-22:
Though the maven build works fine on my system now the actions fail on github.
@wimjongman would you please have a look as I have no idea what is going on

```
Error:  Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:2.7.5:validate-classpath (default-validate-classpath) on project io.sloeber.core.nl1: Execution default-validate-classpath of goal org.eclipse.tycho:tycho-compiler-plugin:2.7.5:validate-classpath failed: org.osgi.framework.BundleException: Bundle io.sloeber.core.nl1 cannot be resolved:io.sloeber.core.nl1 [113]
Error:    Unresolved requirement: Fragment-Host: io.sloeber.core
Error:      -> Bundle-SymbolicName: io.sloeber.core; bundle-version="4.4.1.qualifier"; singleton:="true"
Error:         io.sloeber.core [112]
Error:           Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=17))"
```

wimjongman, 2022-11-22:
Ok, I'm on it.

wimjongman, 2022-11-24:
Fixed

closed 2022-11-24 by wimjongman