Sloeber/arduino-eclipse-plugin #1525: Library problems

DocAlex, 2022-11-26:
First of all: Thank you for providing this envionment! It supports my developments with arduino in a very professional way!

There is one last thing:

**Describe your environment**   
Do you use the Sloeber product or plugin?   **Sloeber product**  
Sloeber version number    **Sloeber	4.4.1.202207231052	io.sloeber.product	null**
OS: [e.g. windows 10]  **Windows 10** 
The url of the json that you used to download the platform **https://eclipse.baeyens.it/**
A image of the project properties arduino showing the board settings   

[image]

**My intention ist to keep the most important libraries within the projectfolder to be able to recreate the project after years even if the arduino or any other library folder is missing.** 

To achive this I copied the important libs into the library folder of the project in sloeber workspace:

[image]

then I imported them into sloeber. 

My understanding is now, that the highest priority for including the lib into the project would be this D:\Benutzer\Alexander\Documents\sloeber-workspace\Raumfuehler\libraries\Adafruit_Si7021.h

unfortunately the rusult is this:

[image]

I used the following library manager settings to **prevent an automatic of copying the libraries somehow (Idid not understand this mechanism) into the project.**  

[image]

Side effect:
         If I enable the auto inclusion of the libraries into the project maybe it **compiles/builds once** but the I get a multiple definition 
         error such as
 
`'Starting combiner'
"C:\Sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc"
-w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p
-o
"D:\Benutzer\Alexander\Documents\sloeber-workspace\Raumfuehler\Release/Raumfuehler.elf"
.\sloeber.ino.cpp.o  .\libraries\libraries\LiquidCrystal_I2C.cpp.o
.\libraries\Wire\src\utility\twi.c.o .\libraries\Wire\src\Wire.cpp.o
.\libraries\TCA9548A\src\TCA9548A.cpp.o .\libraries\SPI\src\SPI.cpp.o
.\libraries\SHT2x.h\SHT2x.cpp.o
.\libraries\LiquidCrystal_I2C.h\TCA9548A.cpp.o
.\libraries\Adafruit_Si7021.h\Adafruit_Si7021.cpp.o
.\SHT2x.h\SHT2x.cpp.o  .\Adafruit_Si7021.h\Adafruit_Si7021.cpp.o
.\1.13.2\Adafruit_BusIO_Register.cpp.o .\1.13.2\Adafruit_I2CDevice.cpp.o
.\1.13.2\Adafruit_SPIDevice.cpp.o
D:\Benutzer\Alexander\Documents\sloeber-workspace\Raumfuehler\Release/arduino.ar
"-LD:\Benutzer\Alexander\Documents\sloeber-workspace\Raumfuehler\Release"
-lm
.\libraries\LiquidCrystal_I2C.h\TCA9548A.cpp.o (symbol from plugin): In
function `TCA9548A::TCA9548A(unsigned char)':
(.text+0x0): multiple definition of `TCA9548A::TCA9548A(unsigned char)'
.\libraries\TCA9548A\src\TCA9548A.cpp.o (symbol from
plugin):(.text+0x0): first defined here
.\libraries\LiquidCrystal_I2C.h\TCA9548A.cpp.o (symbol from plugin): In
function `TCA9548A::TCA9548A(unsigned char)':
(.text+0x0): multiple definition of `TCA9548A::TCA9548A(unsigned char)'
.\libraries\TCA9548A\src\TCA9548A.cpp.o (symbol from
plugin):(.text+0x0): first defined here
.\libraries\LiquidCrystal_I2C.h\TCA9548A.cpp.o (symbol from plugin): In
function `TCA9548A::TCA9548A(unsigned char)':
(.text+0x0): multiple definition of `TCA9548A::begin(TwoWire&)'
.\libraries\TCA9548A\src\TCA9548A.cpp.o (symbol from
plugin):(.text+0x0): first defined here
.\libraries\LiquidCrystal_I2C.h\TCA9548A.cpp.o (symbol from plugin): In
function `TCA9548A::TCA9548A(unsigned char)': `
... much more to follow

It even seems that sloeber copies the library files into the "Build" directory or in the 

I tried many approaches (even a complete reinstall of sloeber) but I end up always at the same issue.

Btw: its the same issue when I design a new class on my own. 

Here the complete view of the project 

[image]

the library exists here:

[image]

[image]

**So what's wrong with including the library?

Maybe I messed up the whole inclusion process. In this case: How do I include a library into my project in a priority so that this library is used and not any library from an arduino folder.  
**

Thanks a lot in advance

Alexander

jantje, 2022-11-26:
>My intention ist to keep the most important libraries within the projectfolder to be able to recreate the project after years even if the arduino or any other library folder is missing.

That is a good idea and works with sloeber.

>To achive this I copied the important libs into the library folder of the project in sloeber workspace:

I think you made 2 mistakes executing this step.
1) You used libraries folder. The libraries folder is maintained by Sloeber. You should not add stuff here yourself. Simply use a different folder like "myLibraries"
2)In the image you can only see header files. I would expect to see a folder for each library that contains all the source code for that library.

Note that the library folder is not part of the inclusion path so these headers will not be found and ignored by Sloeber.

I think a better way is to have 1 or more eclipse projects that contain the libraries you use and add these to the private libraries.
This way sloeber will link to them from the libraries folder and do all stuff like inclusion path and exclude stuff you do not want (like example code)

DocAlex, 2022-11-27:
Thank you for the help. 

It is compiling now. 

Would you think it makes sense to:

1. Mention in the documentation  hat the "libraries" folder is managed by sloeber
2. Add a folder "My_Libraries" or "Libraries_t_include" by default in the sloeber workspace and set the include path to this folder? This might it make easier for users. 

There is one last thing:

Here is the setting of the preferences I use now:

[image]

Is the setting for "automatically include libraries ... " correct? I am a bit confused about the difference between "import" and "install". Does one mean "copy" and the other one "do something magic in sloeber although the path is set correctly"?

I would appreciate an update in the documentation here to make it easier for beginners. 

Thanks again for your help!

Alexander

jantje, 2022-11-27:
>Mention in the documentation hat the "libraries" folder is managed by sloeber

Feel free. The website is here in the repositories or you can do it in the wiki's
Basically I'm already outnumbered by issues that documentation only gets fixed if it is really wrong or when someone else does it.

>Add a folder "My_Libraries" or "Libraries_t_include" by default in the sloeber workspace and set the include path to this folder?

Thing is: 
1)Sloeber automagically removes non existing include paths. 
2)The my_Libraries should contain a folder for each library and the include path should point to that folder or the subfolder src.
3)I think more users would be confused about the 2 folders then users who would benefit.
4)IMHO Using private libraries is the way to go.

Also I'm currently trying to rewrite the CDT managed build (less than 20% chance of success) which would allow for having a different project setup like
arduino/core
arduino/libraries
src
targets/Release

"import libraries" probably would be better named: "Automatically import libraries in the project based on unresolved includes in this project"
"install libraries" probably would be better named: "Let the library manager automatically search the library json files -provided in the third party index url's- for libraries that may resolve the unresolved includes for any project in the workspace and install them into the workspace if found"

But that is kind of getting long.
Maybe moving the install libraries to the library manager will make the distinction more clear.

jantje, 2022-11-27:
Can this issue be closed?

closed 2023-01-16 by jantje