Sloeber/arduino-eclipse-plugin #1526: Upload lem using diamex programmer

DocAlex, 2022-11-27:
Describe your environment
Do you use the Sloeber product or plugin? Sloeber product
Sloeber version number Sloeber 4.4.1.202207231052 io.sloeber.product null
OS: [e.g. windows 10] Windows 10
The url of the json that you used to download the platform https://eclipse.baeyens.it/
A image of the project properties arduino showing the board settings

[image]

My intention is to use this diamex programmer https://www.diamex.de/dxshop/USB-ISP-Programmer-fuer-Atmel-AVR with sloeber. 

I know the properties that this system is running. It did run 2 Weeks earlier with the "Not updated Version - as you install it from the download" of sloeber. 

Therefore I modified the file programmers.txt at the location C:\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.8.6

Here is the modified file: 

[programmers.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/10098035/programmers.txt)

Problem: it did not upload:

[image]

To check if this is really the file sloeber is using I modified the name of the diamex programmer to "DIAMEX-AVR-ISP-NEW", restarted sloeber ant to my suprise the old name was still present in the preferences menue (see above). 

For my understanding this means that sloeber, in the current version, does not use the programmers.txt file.

To my supprise I found a second file "programmers.sloeber.txt"

Here is the file

[programmers.sloeber.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/10098040/programmers.sloeber.txt)

When I do the name modification here It is showing up in the sloeber preference window as "DIAMEX-AVR-ISP-NEW". 

For my understanding this is the programmer file sloeber uses. I removed the "NEW" and discovered that this file seems to be a test since it stated at the beginning: "#Sloeber created workaound file V1.04.test 04" I assume that this came with the last update. 

Never the less the upload fails. 

To check the issue I looked at the upload command that sloeber uses: 

```
Starting upload
Uploading project "Raumfuehler" with "avrdude"

Launching: C:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:D:\Benutzer\Alexander\Documents\sloeber-workspace\Raumfuehler\Release/Raumfuehler.hex:i 
Output:
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x21
```

So the programmer is called with the options:

-b115200
- stk500_recv()

It is known that the diamex programmer needs the  stk500v2 driver and 19200 bps to work correctly. 

Therefore I have set these parameters in the programmer.txt

**Problem: sloeber uses different parameters for the upload. So the upload fails.** 

To veryfy if the diamex programmer is working correctly I used avrdudess to upload **the .hex file that sloeber generated** for the upload. Here is the result:

[image]

The upload is working fine with my board using the correct parameters.

So it seems to me that sloeber magically transforms my settings from the "programmer.txt" file to "programmers.sloeber.txt" which seems not to be updated at sloeber start. Then it seems that sloeber magically use different parameters (where ever they might origin from). 

------------- end bug decription ----------

Some comments: 

1. comment

In my old setting of sloeber I had to include the 

diamexavr.program.extra_params=-P COM3 

option since the "Port selector" Option in the programmer select menue did not have any effect on the com port. 

I see the adding of extra params is a good workaround but maybe you can have a look at this. 

remark: adding

diamexavr.program.extra_params=-P${serial.port} 

did not work. I found another issue where you mentioned that this variable is only set in default. 

I did not find another way to make use of the selection menue for the port. 

Any ideas?

2. comment

When you use the plain arduino IDE and hit the "upload button" by selecting "default" for upload. This diamex programmer is working without any modifications right out of the box.

For the sloeber environment I would have expected the same behaviour. But it did not work.

To get started with sloeber it took me a long time to fix this issue. Would it make sense to place a "Arduino IDE default Programmer" in the programmer.txt file when sloeber is installed? 

3. comment

For me as a user it is absolutely unclear what the data fields of a programmer I can modify. It is only "try and error" wit internet posts. 

I would recommend to provide a list of all parameters of the together with the default setting. A good place would be the "programmers.txt" file (of course commented out) or the documentation. This might make it easier for users to modify th settings correctly: 

```
ExampleProgrammer.name=USBasp     // can be any name for display
ExampleProgrammer.communication=usb // can be serial, USP .....
ExampleProgrammer.protocol=usbasp     // can be usbasp or dapa or ....
ExampleProgrammer.program.protocol=usbasp
ExampleProgrammer.program.tool=avrdude
ExampleProgrammer.program.tool.default=avrdude
ExampleProgrammer.program.extra_params=-Pusb
ExampleProgrammer.force=true
ExampleProgrammer.program.extra_params=-F    // they will appear in the programmer call

tbc.
```

---------- end comments ----------

Thanks a lot to provide this framework to the community. It is a real joy to work with it.

Best regards

Alexander

jantje, 2022-11-27:
### About the sloeber.txt files
>So it seems to me that sloeber magically transforms my settings from the "programmer.txt" file to "programmers.sloeber.txt" which seems not to be updated at sloeber start. Then it seems that sloeber magically use different parameters (where ever they might origin from).

The sloeber.txt versions of the .txt files are files that are created by sloeber at runtime.
If you modify the .txt file you should delete the corresponding sloeber.txt file for your change to be taken into account.
If you change any .txt file (and have deleted the sloeber.txt file) it is very likely these changes will not be taken into account until you have done the following action in Sloeber: project properties->sloeber->apply and close

The sloeber.txt files are files that contain workarounds that sloeber needs to do to work properly. As these are mostly regex expressions that take time creating these files is time consuming.
Due to the time cost (and for easier debugging reasons) I opted not to regenerate these files all the time. As a consequence there is an extra step of deleting the .sloeber.txt file. 
The comment line in the sloeber.txt is a hardcoded string that allows sloeber to see it has been upgraded to a version needing another workaround. Therefore the test string as I do test the workarounds.

### About txt file
txt files are provided by third parties. Sloeber reads them and starts from there. 
Sloeber interprets them as good as possible according to the arduino framework documentation.
History has proven that Sloeber has no influence on the arduino framework nor on specific platforms (that contain the .txt files)
Simply put: Nearly 100% of what is in the arduinoPlugin folder comes from third parties and is not under control of Sloeber.
Exceptions are the sloeber.txt which can be deleted at any time
anything under arduinoPlugin/tools folder (awk and make)

### About upload using a programmer
I recently made a change that makes Sloeber more compliant with the Arduino IDE on behalf of uploading/programming.
Arduino IDE has a "upload" menu item and a "program" menu item.
Sloeber only had a upload and when a programmer was selected the programming action was done.
If you look closely at the "program console" you provided you can see Sloeber is actually doing a upload and not a program.
[image]

There is a new toolbar item that you can use to do the programming
[image]

Note that all this is a consequence of adding burn bootloader to Sloeber.

### Programmers guru mode
After you programming you board using the programmer take a look in your release folder. You will see there is a target that looks familiar.
Open the makefile and you will see what the commands look like. It may make sense to you. If it does and you know the makefile includes "../makefile.target" which mean the file named makefile.target in the root of your project you are at guru level.
Congratulations 🎆 
[image]

### Programmers and quality of code implementation
As you may know, the quality of the first implementation of code is nearly always very bad. By testing and using it can improve over time. But here there is a problem with the programmers. I only own 1 programmer and I only used it about 2 times.
So the programmer code may be buggy.
Fortunately Sloeber uses the Arduino framework so Sloeber only needs to mimic the Arduino IDE but that code has it special behaviour (like anything else in this universe)

### About comment 1
If you open the project properties->c/c++ Build->environment you can see a lot of environment variables that mostly come from the .txt files.
You also see the port. You should be able to use any of these variables (with some wierdness though as they may expand at a different time you would expect).
[image]
>Any ideas?

Let everything I've written here sink in and retry.

### About comment 2
>When you use the plain arduino IDE and hit the "upload button" by selecting "default" for upload. This diamex programmer is working without any modifications right out of the box.

Unless Arduino IDE changed recently I would disagree. It woiuld be funny though that Arduino IDE switched to the Sloeber way at the same time Sloeber switched to the Arduino IDE way.

>To get started with sloeber it took me a long time to fix this issue. Would it make sense to place a "Arduino IDE default Programmer" in the programmer.txt file when sloeber is installed?

Funny how you mention this as this is how Sloeber did it before the bootloader was supported.
IMHO The arduino IDE is confusing because you have to select a programmer but the programmer is only used when you select "upload using programmer" which is consistent with what Sloeber does now.

### about comment 3
>For me as a user it is absolutely unclear what the data fields of a programmer I can modify. It is only "try and error" wit internet posts.

Modifying programmer.txt is advanced usage and you need to consult the arduino framework documentation, the programmer documentation and the board documentation.  
As I implemented the arduino framework I can help out with that. The others are unknown to me.

I probably missed something but enough rambling for now.

DocAlex, 2022-11-27:
Thank you very much for your help. 

Using the "new Uppload button" (Arrow up) it works perfectly.

jantje, 2022-11-27:
I guess we can close the issue.
If not please reopen

closed 2022-11-27 by jantje

theduskin, 2023-11-22:
not sure if it's only about diamex programmer.
when uploading, to at85 no bootloader, with burnOmat, code works as expected.
`-p t85 -P com11 -c stk500v2  -U flash:w:D:\name.hex:a `
when I use the new sloeber upload button, strange things happen, code is not working properly.
`-pattiny85 -cstk500v2 -PCOM11 "-Uflash:w:D:\name.hex:i"`
since the flash content differs from one to another, I would think the culprit is the format flag.
but there are also timing related differences in the .conf files.
I would try format:auto first if i had an idea where to set it.

jantje, 2023-11-22:
@theduskin 
Please create a new issue and assume I know nothing of your environment.
For example: The following words are not known/specific enough for me to help you
-at85
-burnOmat
-strange things happen
- code is not working properly.
- timing related differences in the .conf files.

Have a look at the original post of @DocAlex which links Sloeber content/actions to Sloeber generated command line output and expected output.