Sloeber/arduino-eclipse-plugin #1527: Sloeber missing support for the Serial UPDI upload method used in the DrAzzy megatiny core (ATtiny1614 etc.)

6v6gt-duino, 2022-11-29:
Sloeber does not appear to support the Serial UPDI upload method used on some of the newer ATTiny series boards/chips such as the ATtiny1614 using the DrAzzy megatiny core.

For this, I am using Sloeber 4.4.1 and have selected the Serial UPDI 230400 programmer method.

[image]

I get this when I attempt to upload a program: 

```
==============================================================

Starting upload
Uploading project "_test_ATTINY1614_V0_01" with "avrdude"

Launching: C:\sloeber_v4.4.1\arduinoPlugin\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\sloeber_v4.4.1\arduinoPlugin\packages\megaTinyCore\hardware\megaavr\2.6.4/avrdude.conf -pattiny1614 -cstk500v1 -PCOM4 -b19200 -Uflash:w:C:\Users\6v6gt\Documents\sloeber-workspace\_test_ATTINY1614_V0_01\Release/_test_ATTINY1614_V0_01.hex:i 
Output:
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20

avrdude done.  Thank you.

The execution of command "6.3.0-arduino17or18/bin/avrdude" is done.
==============================================================
```

If I try exactly the same settings in the Arduino IDE 1.8.19 it works and I get:

```
==============================================================
"C:\\Users\\6v6gt\\Documents\\ArduinoData\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino6/bin/avr-size" -A "C:\\Users\\6v6gt\\AppData\\Local\\Temp\\arduino_build_554987/sketch_nov29a.ino.elf"
Sketch uses 1758 bytes (10%) of program storage space. Maximum is 16384 bytes.
Global variables use 155 bytes (7%) of dynamic memory, leaving 1893 bytes for local variables. Maximum is 2048 bytes.
C:\Users\6v6gt\Documents\ArduinoData\packages\megaTinyCore\tools\python3\3.7.2-post1/python3 -u C:\Users\6v6gt\Documents\ArduinoData\packages\megaTinyCore\hardware\megaavr\2.6.4/tools/prog.py -t uart -u COM4 -b 230400 -d attiny1614 --fuses 0:0b00000000 2:0x02 6:0x04 7:0x00 8:0x00 -fC:\Users\6v6gt\AppData\Local\Temp\arduino_build_554987/sketch_nov29a.ino.hex -a write -v 
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port COM4 at 230400 baud.
Target: attiny1614
. . . 
. . .
this goes on to work
==============================================================
```

Clear is that, in the Arduino IDE,  a python script is used in the upload method ....\\megaTinyCore\hardware\megaavr\2.6.4/tools/prog.py  and in Sloeber attempts simply a direct call to AVRDUDE (which I believe cannot work)

jantje, 2022-11-29:
Since Sloeber 4.4.1sloeber has a upload and program button just like arduino IDE
You seem to have selected to upload and not to program

6v6gt-duino, 2022-11-30:
Well, I'm impressed. That did indeed work. 
It doesn't quite follow the same model as in the Arduino IDE (old 1.x.x) where the same upload button is pressed whether you are uploading a sketch to say a Uno over its UBS port or an ATtiny1614 using a UPDI programmer so I guess there may also be others confused by this. Anyway, this is the screen shot for anyone else attempting the same thing.

[image]

closed 2022-11-30 by jantje