Sloeber/arduino-eclipse-plugin #1529: Sloeber does not install in Eclipse 2022-12

Antonio-Raposo, 2022-12-08:
Hello.

I am evaluating Eclipse 2022-12 released yesterday and I am unable to install the Sloeber plugin.

My usual way of installing eclipse, as I develop both in java and C, is downloading eclipse C++ and eclipse java and extracting both on top of each other, creating this way a C/Java eclipse.

Then I tried to install Sloeber from V4 and from the nightly build on my Windows 10 and it didn't work.

For example for version 4.4.1.202209072349 (nightly) I get this error

> An error occurred while collecting items to be installed
> session context was:(profile=epp.package.java, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
> No repository found containing: osgi.bundle,org.eclipse.nebula.cwt,1.1.0.202211220849
> No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.cdatetime,1.5.0.202211220849
> No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope,1.4.0.202211220849
> No repository found containing: org.eclipse.update.feature,org.eclipse.nebula.widgets.oscilloscope.feature,1.4.0.202211220849
> No repository found containing: osgi.bundle,org.eclipse.nebula.widgets.oscilloscope.source,1.4.0.202211220849

I found an workarround
First from https://download.eclipse.org/nebula/snapshot/ install Nebula Oscilloscope Widget (1.4.0.202211220849 as the error indicates), then install Sloeber

Hope this description helps solving.

jantje, 2022-12-08:
Thanks for the input.

>For example for version 4.4.1.202209072349 (nightly) I get this error

That is an old nightly version. seems the nightly isn't updating anymore.

There are plenty of problems around updates not working. here are 2 issues #1361 #1492 
The set of update URL's and having the plugin or the project selected are some of the things we know give problems.

CarterSnich, 2022-12-25:
> Hello.
>
> I am evaluating Eclipse 2022-12 released yesterday and I am unable to install the Sloeber plugin.
>
> My usual way of installing eclipse, as I develop both in java and C, is downloading eclipse [...]

Do I install this on the Marketplace or "Install New Software"? I already tried installing Sloeber on "Install New Software" and it broke my Eclipse installation.

jantje, 2022-12-25:
>Do I install this on the Marketplace or "Install New Software"? I already tried installing Sloeber on "Install New Software" and it broke my Eclipse installation.

If you need to ask :Install the product from here http://eclipse.baeyens.it/

jantje, 2023-01-16:
Apart from the updates done by Wim; we learned CDT now needs Java.

jantje, 2023-01-19:
`org.hamcrest.core` Does not ring a bell.
 `org.junit,4.13.2.v20211018-1956`  JUnit is well know but not used in Sloeber. Maybe there is a dependency on the unit testing provided by CDT installed with Sloeber. The only thing is that the version requested seems old.
Absolutely do not understand why it works directly and not with the marketplace.
Might be related to the marketplace moving to https

dsptech, 2023-01-20:
Hi, just to understand the reason: The direct installation ask for "sloeber" and "sloeber plugin". I selected only the plugin (because this has been said in other github issues) . Is this the proper selection ?

The installation via markeplace asks no questions and now I'm wondering if it try to install both.

jantje, 2023-01-20:
>Hi, just to understand the reason: the direct installation ask for "sloeber" and "sloeber plugin". I selected only the plugin (because this has been said in other github issues) . Is this the proper selection ?

If you have a existing eclipse you can add the plugin. The product is the eclipse framework + the plugin.
So if you add the Sloeber product to a eclipse install you will overwrite the eclipse framework.
Probably with a older version.
Overwriting the eclipse framework is a required functionality to allow eclipse upgrades.

>The installation via markeplace asks no questions and now I'm wondering if it try to install both.

The marketplace should only install the plugin.

dsptech, 2023-01-21:
Hi,
I performed furter tests, unfortunatly with no success. The issues org.hamcrest.core and org.junit are already reported in the Sleber issues from 2021, but all the proposed solutions are not applicable here nor I'm able to find a proper repositories for the Eclipse configuration ( are they still exist ? ). 
Finally I just discovered that the link "https://download.eclipse.org/nebula/snapshot/" appear to be irrelevant (and not working). Only the non-secure version "http://download.eclipse.org/nebula/snapshot/" is required (but that may change in the future).
I have to desist now so I'll use the direct installation.

jantje, 2023-01-21:
We experienced plenty of strange behaviour with the update functionality of eclipse. Things we do not understand (like disappearing update site)
The move to https probably doesn't make things better.
Also note that my build server has been down (and I failed to get it up) for a long time. That means the update nightly sites hold the nightly from the last successful run.
In general I follow following rules when upgrading.
1) make a backup
2) try upgrade
3) if success be happy
4) if failed reinstall or go back to backup

dsptech, 2023-01-21:
Hi, 
It's exactly my philosophy and I don't want to be misunderstood. 
I decided to post the information here only in the hope that they will be useful to and, indeed, I am personally grateful to you for the hard work you are doing with this plugin. 
Regards,
Andrea.