Sloeber/arduino-eclipse-plugin #1532: usage of C++17 on STM32

MasteringTheMess, 2022-12-09:
I tried to get the compiler to use C++17 standard

Therefore I modified platform.txt and the corresponding platform.sloeber.txt in the C:\Sloeber\arduinoPlugin folder.
Although this works for Arduino IDE, it has no effect and __cplusplus still outputs 201402

Is compiler.cpp.std set somewhere else?

jantje, 2022-12-09:
How to modify a .txt file
1) Modify the .txt file
2) delete the corresponding *.sloeber.txt file.
3) for each project that should be impacted: goto project properties->sloeber->apply and close
3 check) if the sloeber.txt file is not there something went wrong and your change has not been taken into account in any project by Sloeber.
4) force Sloeber to recreate the makefile by deleting the build output folder
5) build

jantje, 2022-12-09:
Also see the newly created discussion for a better way 
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533

closed 2022-12-09 by jantje

MasteringTheMess, 2022-12-12:
Hooray! it works now!

Thanks for clarifying how to fiddle with the .txt-files :)