Sloeber/arduino-eclipse-plugin #1535: MiniCore-Master platform use

Mattpisti, 2022-12-21:
Hi there, I just installed version 4.4.1 and added the MiniCore-Master platform.  Unfortunately all my projects based on that cores do not compile any more showing syntax errors.

Everything works with version 4.3.3 

Any clue?
Thanks
Matt

jantje, 2022-12-21:
How did you install the MiniCore-Master platform.
What compile errors do you get?

Mattpisti, 2022-12-22:
I added the Mini-Core platform (I tried also with the newer release at https://github.com/MCUdude/MiniCore/releases ) with Sloeber\Preferences\Private Hardware Path and then in Project\Properties as in the attached picture 

[image]

This is the compile error

"C:\\Sloeber4.4.1\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=8000000L -DARDUINO=10812 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR     -I"D:\arduino-1.8.13\hardware\MiniCore-master\avr\cores\MCUdude_corefiles" -I"D:\arduino-1.8.13\hardware\MiniCore-master\avr\variants\standard" -I"D:\arduino-1.8.13\hardware\MiniCore-master\avr\libraries\EEPROM\src" -I"D:\arduino-1.8.13\libraries\jeelib-master" -I"D:\arduino-1.8.13\libraries\arduino-libs-manchester-master" -I"D:\arduino-1.8.13\libraries\truerandom" -I"C:\Users\vince\Documents\Arduino\libraries\OneWire" -I"C:\Users\vince\Documents\Arduino\libraries\DallasTemperature" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"   -o "sloeber.ino.cpp.o"
sh: /bin/avr-g++: Permission denied
make: *** [subdir.mk:20: sloeber.ino.cpp.o] Error 126
"C:/Sloeber4.4.1/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

Hope this helps

jantje, 2022-12-22:
When using private hardware there is platform.json file so not all environment variables are known.
You will need to look in the platform.txt file and find the environment variable in the compile command that points to the gcc install folder
Then create this environment variable in sloeber or the platform.txt or your OS (whatever suits your taste)

Mattpisti, 2022-12-23:
Sorry but I am a novice and don't fully understand what you mean. As I wrote, after using Arduino IDE, I used V4.3.3 that works just fine adding the MiniCore  platform in the same way I did for V4.4.1.
 I would appreciate if you are so kind to explain what I should do step by step.
For your reference I am attaching the platform.txt file contained in the MiniCore platform
Thanks

[platform.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/10293374/platform.txt)

jantje, 2022-12-25:
Novices should not use private hardware.
Why not use https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json ?

Mattpisti, 2022-12-25:
I tried using it but I get an error, see pictures, and I am stuck, any suggestion?
Why V4.3.3 works with private hw and 4.4.1 doesn't?

[image]

[image]

jantje, 2022-12-25:
It is unclear to me what you are trying to do but it doesn't comply with how Sloeber is supposed to work.
1) in the menu select Sloeber->preferences
2) add https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json to the "Third party index url's" page
[image]

3) select apply/close
4) in the menu select Sloeber->preferences
5) in Sloeber->Platform and boards check all the versions you want to install
[image]
6)Select apply and close
7)wait for Sloeber to download and install all needed stuff
[image]

You are ready to use the MicroCore Boards

Mattpisti, 2022-12-25:
Thanks, it works now and I am understanding a little bit more how Sloeber works.

jantje, 2022-12-25:
This is exactly the same behaviour as Arduino IDE.
Different places same behaviour.
Anyway I think this issue can be closed.

closed 2022-12-25 by jantje