Sloeber/arduino-eclipse-plugin #1542: Sloeber (features) wiped out after Eclipse "Check for updates"

dsptech, 2023-01-11:
Hi,
I have a Sloeber installation (Windows) done two year ago (4.2 or 4.3?). The original installation was done via the integrated package and was later updated via the Eclipse update system (Sloeber nightly too).
Over the time, "update sites" and java (now 15) have been updated according to the other Github issues and everything worked fine up to yesterday.
By applying the last update, Eclipse CDT appear to work, but Sloeber (and other extensions as the themes) seems to be wiped out and lots of errors are showing in the workspace log.
I suspect that the issue is someway related to: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1529.

Honestly, I have no idea how to deal the problem and also if this a Sloeber, Eclipse or Java version issue. Does anyone have any suggestion?.

Thank you.

P.S.: I attach the screen capture of:
1) Sloeber before the update
2) Configured update sites
3) Installation details
4) Proposed updates (performed with no update errors)
5) Final (bad) result with errors.

[image]
[image]

[image]
[image]
[image]
[image]
[image]

jantje, 2023-01-13:
From the log it seems some parts of the CDT ui are no longer available.
This is normal as you did a CDT upgrade that requires java 17 and you say you run java 15.
I would also recommend to clean up your update sites. Only keep the latest versions of the sites that have multiple versions.

dsptech, 2023-01-14:
Hi,
as you suggested I replaced Java 15 with Java 17 (JRE from https://adoptium.net/temurin/archive/) and now also assume that the info at http://eclipse.baeyens.it/stable.php?OS=Windows are outdated (it says max V.15)

With Java17, the update doesn't produce any final error but some plugins is still wiped. Focusing on Sloeber, I noticed that the "Sloeber" menu item is missing from the main menu, and now there is no icon for the Sloeber's perspective.
However the perspective seems to work (switching to Egit and Sloeber again restores the correct view) and the Sloeber entry also exists in the general preferences. Finally, all the Sloeber icons in the toolbars seems to work properly. 

Sadly, in the existing Arduino Projects the project context menu does no longer show the "Sloeber" entry so, for example, I'm no longer able to change the board.

Just for checking, has the last nightly the version number: 4.4.1.202209072349 ?
Thank you.
[image]

dsptech, 2023-01-15:
Hi,
as you suggest, I performed the update with Java 17 (JRE from https://adoptium.net) ... and now I also assume that the info at: http://eclipse.baeyens.it/stable.php?OS=Windows are outdated (it says Java max 15).

Now the update appear to work but themes continue do be wiped out.  
Focusing on Sloeber, the "Sloeber" entry in the main menu' is missings. Also the icon of the Sloeber perspective is missing but it appear to work properly, this mean that switching, for example, to the Egit perspective and again to Slober work as expected.

Sloeber entry is also present in the main preferences and it is also accessible via project context menu. Finally, the Sloeber toolbars appear to work fine. 

How to fix the missing items ? 

[image]

jantje, 2023-01-15:
>Focusing on Sloeber, the "Sloeber" entry in the main menu' is missings. 

This is only there when you are in the Sloeber or C/C++ perspective. Seems you are not in one of those perspectives.

>How to fix the missing items ?

I advice to download the latest Sloeber product and start from there.
This because an old eclipse install with many updates is prone to issues.

dsptech, 2023-01-19:
Hi @jantje,
I performed further test as remove the instlled plugins, update on a fresh workspace, change Java version etc, but but every attempt is failed so I place this thing in stand-by for a fresh installation.

By the way, in a fresh Windows 10 installation I downloded (from github) and installed the full package as you suggested. After a successfull update I found the that the list of the update sites has been totally emptied. Not sure if this is expected (and I have to reenter them) or I have to fill a new issue.

Finally, also the installation the Sloeber plugin in a fresh Eclipse installation has failed. Problems are similar to https://github.com/Sloeber/arduino-eclipse-plugin/issues/1529 and https://github.com/Sloeber/arduino-eclipse-plugin/issues/1492 but the proposed solutions do not work. I will post on a 1529 soon.

dsptech, 2023-01-21:
I'm no able to find a reason for the erasing of the update sites list (see previous message). I suspect now about a bug in the Eclipse update system. A further investigation should be required but this is over my skills so I will use this: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1529#issuecomment-1399234045.
Regards.

closed 2024-02-20 by jantje