Sloeber/arduino-eclipse-plugin #1543: Product bundle won't start on linux

woodz-, 2023-01-12:
My ubuntu's java version:

    $ java --version
    openjdk 11.0.17 2022-10-18 LTS
    OpenJDK Runtime Environment Zulu11.60+19-CA (build 11.0.17+8-LTS)
    OpenJDK 64-Bit Server VM Zulu11.60+19-CA (build 11.0.17+8-LTS, mixed mode)

After downloading your current [product bundle](https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_4_1_0/V4.4.1_linux64.2022-07-09_05-48-31.zip)
I encounter the following issues:

- the `sloeber-ide` file needed to be made executable: `$ chmod u+x sloeber-ide`
- running `$ ./sloeber-ide` gives an error:

    JVM terminated. Exit code=13 /home/woodz/Downloads/eclipse_22/Sloeber//plugins/org.eclipse.justj.openjd k.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/java
    -Dosgi.requiredJavaVersion=11
    -Dorg.eclipse.update.reconcile=false
    -Dlog4j.configuration=log4j/log4j.xml
    -Xms256m
    -Xmx1024m
    -jar 
/home/woodz/Downloads/eclipse_22/Sloeber//plugins/org.eclipse.equinox.lau ncher_1.6.400.v20210924-0641.jar
    -os linux
    -ws gtk
    -arch x86_64
    -showsplash
    -launcher /home/woodz/Downloads/eclipse_22/Sloeber/sloeber-ide
    -name Sloeber-ide
    --launcher.library 
/home/woodz/Downloads/eclipse_22/Sloeber//plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.2.500.v20220509-0833/eclipse_11701.so
    -startup 
/home/woodz/Downloads/eclipse_22/Sloeber//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
    --launcher.appendVmargs
    -exitdata 5e9004c
    -perspective io.sloeber.application.perspective
    -vm /home/woodz/Downloads/eclipse_22/Sloeber//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/java
    -vmargs
    -Dosgi.requiredJavaVersion=11
    -Dorg.eclipse.update.reconcile=false
    -Dlog4j.configuration=log4j/log4j.xml
    -Xms256m
    -Xmx1024m
    -jar /home/woodz/Downloads/eclipse_22/Sloeber//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar

Now I have a number of questions:

- Since it is a bundle, shouldn't it come with a fully featured java version?
- Why are you then requiring to install a java from https://adoptopenjdk.net/?
- And does it need to be exactly this type of java?
- I have zulu, would this cause problems?
- The vm option of the ini-file points to a somehow bundled java version 17.0.3, but the `osgi.requiredJavaVersion` of the vmargs wants version 11. Will this fit together?

jantje, 2023-01-13:
1)Since it is a bundle, shouldn't it come with a fully featured java version?

What do you mean with fully featured. As you are doing C++ development and not java a plain java runtime is enough.

2)Why are you then requiring to install a java from  https://adoptopenjdk.net/?

This is a new way for eclipse to bundle java without having to add files to the installation package. At startup the java runtime needed is identified and downloaded.

3)And does it need to be exactly this type of java?

No. If eclipse finds a java that java will be used. However if this java is not valid you will get startup issues. Currently Java 17 is needed for CDT. 

4) I have zulu, would this cause problems?

I don't know.

5)The vm option of the ini-file points to a somehow bundled java version 17.0.3, but the osgi.requiredJavaVersion of the vmargs wants version 11. Will this fit together?

Current Sloeber release is compiled and compliant with java 11. However it is also dependent on CDT who's newest version requires java 17.
As the build uses the latest version of CDT, eclipse and other plug-ins it is unpredictable which version will work.
I guess your results show java 11 is no longer working and I assume java 17 is needed.

woodz-, 2023-01-13:
I must admit _fully featured_ was unclear. I meant an included jre.
And of course we are not developing in java, it's CDT.

Meanwhile I got the secret lifted.
Beside it was required to `$ chmod +x sloeber-ide` it was also required to make all bundled jvm executables executable:

    $ chmod +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/*

What remains confusing is, that CDT needs java 17, but the sitting on top sloeber got compiled with java 11. 
The bundle [description](http://eclipse.baeyens.it/stable.php?OS=Linux) tells:
_Sloeber needs java 10 or higher and lower than 16 ! Other java versions will not work!_
and
_Download a java from https://adoptopenjdk.net/_

Both statements aren't true since sloeber runs with the bundled java 17 and
there is no need to update the machine to java 17. Once the bundled java 17 made executable (as shown above), sloeber runs out of the box.

jantje, 2023-01-14:
Good you got it towork.

>Sloeber needs java 10 or higher and lower than 16 ! Other java versions will not work!
Both statements aren't true

Indeed the lower than 16 requirement has been fixed and is no longer true. Where did you find that?

>Download a java from https://adoptopenjdk.net/

Again a "old statement" where did you find that (Honestly the documentation is getting more and more out of date)

jantje, 2023-01-14:
Workaround
$ chmod u+x sloeber-ide
Start Sloeber (will fail) (I'm not sure this step is needed; if you tried this workaround can you provide input)
$ chmod +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/*

woodz-, 2023-01-14:
I have given the link, where I found the confusing statements in my previous post. They are in the bundle [description](http://eclipse.baeyens.it/stable.php?OS=Linux)

woodz-, 2023-01-14:
The steps in your last comment are the same, I have given in my post and that is the solution. Just read my comment above.

jantje, 2023-01-14:
>The steps in your last comment are the same, I have given in my post and that is the solution that worked. Just read my comment above.

I know just putting the 2 commands together with the word workaround as I tagged this issue with "workaround documented"
I'm pretty sure a Sloeber start is needed in between as that one will download the files that need to be made executable.

quax1, 2023-01-15:
Hello this workaround does not help completely - I did this:

tried solution from  described here....
chmod +x 'arduinoPlugin/downloads' 
chmod +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/*

restarted sloeber

got this error:
_Failed to install Arduino AVR Boards
Failed to extract tar.bz2. "/home/matthias/Programme/Sloeber/arduinoPlugin/downloads/avr-1.8.6.tar.bz2".
Can't extract file "/home/matthias/Programme/Sloeber/arduinoPlugin/packages/arduino/hardware/avr/1.8.6/programmers.txt", file already exists!_

the directory 
/arduinoPlugin/packages/arduino/hardware/avr/1.8.6
contains only boards.txt  !!

extracted avr-1.8.6.tar.bz2  manually to /arduinoPlugin/packages/arduino/hardware/
created Sloeber/arduinoPlugin/packages/arduino/hardware/avr/avr-1.8.6
start sloeber again:  finisihing installation - wait

Failed to install Arduino AVR Boards
_Failed to extract tar.bz2. "/home/matthias/Programme/Sloeber/arduinoPlugin/downloads/avr-1.8.6.tar.bz2".
Cannot run program "chmod": error=13, Keine Berechtigung_

/arduinoPlugin/packages/arduino/hardware/ contains now two directories
1.8.6  avr-1.8.6

copied all from avr-1.8.6  to 1.8.6  expect now it should work 
start sloeber again:  "finisihing installation" - wait...
now it starts

created example sketch
AnalogReadSerial.ino
"Verify"..
got his error in eclipse
_Program "/bin/avr-g++" not found in PATH_

restarted sloeber
again:  "finisihing installation" - wait...
try again AnalogReadSerial.ino
"Verify"
same error
_Program "/bin/avr-g++" not found in PATH_

jantje, 2023-01-15:
Once things have gone bad it is best to clean up and retry.
So I would run following commands to cleanup 
1) goto the sloeber folder in a command line tool
2) sudo rm -r arduinoPlugin
3) sudo chmod -R 777 *
And retry
A full functional Sloeber requires full read/write/execute access.
This is required for the Sloeber boards manager that installs and runs executables (just like the Arduino IDE does).
If your OS disables running downloaded executables in one way or another you will have to find out how to disable this functionality to have a smooth working Sloeber.

quax1, 2023-01-15:
Thank you very -  much with this cleaning sloeber starts, I can compile a complex project with several libraries.  Perfect for me.
The only point - everytime sloeber starts it displays the window - finishing the installation - sit back and relax
the details show no other proceedings. 
I tried to find out what it is, resetted linux mint with the timeshift tool to a state after installation and removed the sloeber directory completely, but it is the same.   this is the terminal log:   

matthias@aragon:~/Programme$ sudo rm -r Sloeber
matthias@aragon:~/Programme$ cd Sloeber
matthias@aragon:~/Programme/Sloeber$ chmod u+x sloeber-ide
matthias@aragon:~/Programme/Sloeber$ chmod +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/*
matthias@aragon:~/Programme/Sloeber$ ./sloeber-ide
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
^[start installing platform Arduino AVR Boards avr(1.8.6)

(Sloeber:2784): Gdk-CRITICAL **: 18:18:07.740: gdk_threads_set_lock_functions: assertion 'gdk_threads_lock == NULL && gdk_threads_unlock == NULL' failed
matthias@aragon:~/Programme/Sloeber$ sudo rm -r arduinoPlugin
matthias@aragon:~/Programme/Sloeber$ sudo chmod -R 777 *
matthias@aragon:~/Programme/Sloeber$ ./sloeber-ide
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
start installing platform Arduino AVR Boards avr(1.8.6)
done installing platform Arduino AVR Boards avr(1.8.6)

(Sloeber:3531): Gdk-CRITICAL **: 18:36:46.703: gdk_threads_set_lock_functions: assertion 'gdk_threads_lock == NULL && gdk_threads_unlock == NULL' failed__

finishing the installation - sit back and relax

jantje, 2023-01-15:
> everytime sloeber starts it displays the window - finishing the installation - sit back and relax
the details show no other proceedings.

This is Sloeber updating the json files listed in "the third party index url's" in the preferences.
To stop Sloeber trying to update these disable the "update local json files"
[image]

quax1, 2023-01-16:
works fine !  thank you very much

jantje, 2023-01-17:
I updated the website so I'm closing this issue.

closed 2023-01-17 by jantje