Sloeber/arduino-eclipse-plugin #1544: BUILD FAILLURE io.sloeber.core 4.4.1-SNAPSHOT ..................... FAILURE

wolfgangwayland7, 2023-01-14:
Linux Mint        20.3 Cinnamon
Linux Kernel   5.4.0-136-generic
Processor        AMD Ryzen 9 5900X 12-Core Processor × 12

Apache Maven 3.6.3
Maven home: /usr/share/maven
Java version: 11.0.17, vendor: Eclipse Adoptium, runtime: /home/wolf/Downloads/Sloeber/jdk-11.0.17+8
Default locale: en_US, platform encoding: UTF-8
OS name: "linux", version: "5.4.0-136-generic", arch: "amd64", family: "unix"

/*****************************************************************************/
[INFO] Reactor Summary:
[INFO] 
[INFO] io.sloeber.parent 4.4.1-SNAPSHOT ................... SUCCESS [ 26.046 s]
[INFO] io.sloeber.core 4.4.1-SNAPSHOT ..................... FAILURE [  5.958 s]

/***********************************************************************/

 java --version
openjdk 11.0.17 2022-10-18
OpenJDK Runtime Environment Temurin-11.0.17+8 (build 11.0.17+8)
OpenJDK 64-Bit Server VM Temurin-11.0.17+8 (build 11.0.17+8, mixed mode)

/*****************************************************/

[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:2.7.5:validate-classpath (default-validate-classpath) on project io.sloeber.core: Execution default-validate-classpath of goal org.eclipse.tycho:tycho-compiler-plugin:2.7.5:validate-classpath failed: org.osgi.framework.BundleException: Bundle io.sloeber.core cannot be resolved:io.sloeber.core [107]
[ERROR]   Unresolved requirement: Require-Bundle: org.eclipse.cdt.managedbuilder.core
[ERROR]     -> Bundle-SymbolicName: org.eclipse.cdt.managedbuilder.core; bundle-version="9.5.0.202211211546"; singleton:="true"
[ERROR]        org.eclipse.cdt.managedbuilder.core [10]
[ERROR]          Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=17))"
[ERROR]          Unresolved requirement: Require-Bundle: org.eclipse.cdt.core; bundle-version="[8.0.0,9.0.0)"
[ERROR]            -> Bundle-SymbolicName: org.eclipse.cdt.core; bundle-version="8.0.0.202211292120"; singleton:="true"
[ERROR]               org.eclipse.cdt.core [12]
[ERROR]                 Unresolved requirement: Require-Bundle: org.eclipse.jdt.annotation; bundle-version="[2.0.0,3.0.0)"; resolution:="optional"
[ERROR]                 Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=17))"
[ERROR]                 Unresolved requirement: Require-Bundle: org.eclipse.cdt.core.native; bundle-version="[6.3.0,7.0.0)"; visibility:="reexport"
[ERROR]                   -> Bundle-SymbolicName: org.eclipse.cdt.core.native; bundle-version="6.3.0.202211062329"; singleton:="true"
[ERROR]                      org.eclipse.cdt.core.native [36]
[ERROR]                        Unresolved requirement: Require-Capability: osgi.ee; filter:="(&(osgi.ee=JavaSE)(version=17))"
[ERROR]          Unresolved requirement: Require-Bundle: org.eclipse.cdt.make.core; visibility:="reexport"
[ERROR]            -> Bundle-SymbolicName: org.eclipse.cdt.make.core; bundle-version="7.6.0.202211062329"; singleton:="true"
[ERROR]               org.eclipse.cdt.make.core [44]
[ERROR]                 Unresolved requirement: Require-Bundle: org.eclipse.cdt.core; bundle-version="[8.0.0,9.0.0)"
[ERROR]                   -> Bundle-SymbolicName: org.eclipse.cdt.core; bundle-version="8.0.0.202211292120"; singleton:="true"

/**********************************************************************/

The above happens on a  fresh install with : "mvn clean verify -DskipTests=true"

jantje, 2023-01-14:
Please try with java 17

Alex-Kent, 2023-08-08:
The requirement to use Java 17 should be added to the **Prerequisites** section of the top-level `readme.md` file.  (This is included in the [pull request](https://github.com/Sloeber/arduino-eclipse-plugin/pull/1579) I just made.)

(I had the same issue with Java 11 on Ubuntu Jammy.  Running `sudo apt install openjdk-17-jre:amd64 openjdk-17-jre-headless:amd64` fixed things.)

jantje, 2023-08-08:
Strange;
The latest version should add the correct java version itself.

Alex-Kent, 2023-08-08:
At least on Linux it the latest version will be switched to during the build.  On my system it looks like Java 11 is used by default and then the build process switches over to using Java 17 when needed; not sure what magic is going on under the hood.  Java 17 needs to be already installed though.  (I have Java 11, 17, and 9 IIRC installed)

jantje, 2023-08-09:
For building Sloeber you need java17 or later installed
Eclipse community made a plugin that downloads java and Sloeber is dependent on that plugin.
That is why the additional commands are needed in linux (to make the eclipse plugin installed java jre executable)

closed 2023-08-10 by jantje