Sloeber/arduino-eclipse-plugin #1545: Install Sloeber on  Linux Mint  ( 21.1) - product bundle does not install

quax1, 2023-01-14:
Hello Slantje,

you did a great job with sloeber - thank you so much - it is the first time I use something else than Arduino IDE and I love Eclipse/Sloeber.

 I have installed sloeber successfully on windows 10, but I cannot do it on Linux Mint - which I would prefer to use.  I think this can be solved easily - but I have tried for days and now I surrender.

I tried the following:

Install sloeber product as downloaded from your website, however the installer does not start - because java is missing, it gives errors like
JVM terminated. Exit code=13
/home/matthias/Downloads/Sloeber//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/java
-Dosgi.requiredJavaVersion=11 

I tried to copy the Java directory from a new arduino installation into sloeber, but it did not work either.

I have the fresh and newest version of linux mint 21.1 based on Ubuntu Jammy, which comes with openjdk version "11.0.17" (which should work as it is between 10 and 16 )
My system:
Distributor ID:	Linuxmint Cinnamon
Description:	Linux Mint 21.1
Release:	21.1
Codename:	vera
java -version
openjdk version "11.0.17" 2022-10-18
OpenJDK Runtime Environment (build 11.0.17+8-post-Ubuntu-1ubuntu222.04)
OpenJDK 64-Bit Server VM (build 11.0.17+8-post-Ubuntu-1ubuntu222.04, mixed mode, sharing)

I tried to install the newest version of Eclipse and sloeber as a plugin, but this did not work either. It worked in win10 but not in Linux:

In windows 10 I was successfull with the following
1. install eclipse 2022-12
2. install sloeber via marketplace -- gives the nebula error
3.  install new software, added the missing nebula repositories, https://download.eclipse.org/nebula/releases/latest/  and installed it.
4.  install sloeber via marketplace again - now runs fine

but this does not succeed with linux mint, here sloeber gives a lot of errors when compiling
 
One comment:  the instruction on github "Tutorial for Sloeber beginners" is excellent and would be very helpful on your website.
https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1342

jantje, 2023-01-14:
Sorry to hear about your problems.
Have you seen this issue: #1543
As to me these are the same problems you encountered.
IMHO these are new security issues with Linux
Here is how I see the workaround:
https://github.com/Sloeber/arduino-eclipse-plugin/issues/1543#issuecomment-1382781516

jantje, 2023-01-15:
closing as duplicate of https://github.com/Sloeber/arduino-eclipse-plugin/issues/1543

closed 2023-01-15 by jantje