Sloeber/arduino-eclipse-plugin #1551: Wrong avrdude parameter by upload to Atmega4809

stefanbbb, 2023-01-23:
My environment:   
Sloeber version number  4.4.1 (product)
OS: Windows 10  
Json for platform: https://mcudude.github.io/MegaCoreX/package_MCUdude_MegaCoreX_index.json
The following project preferences are set:
 
[image]

Hereby the following avrdude commandline results :

avrdude.conf -patmega4809 -cstk500v1 -PCOM7 -b115200 -D -Uflash:w:E:\HomeControl\SW\Sloeber_workspace2\test\Release/test.hex:i

As you can see, the preferences are ignored for creating the avrdude command string:

- wrong protocoll
- Parameter -D is wrong

Especially the parameter “-D” prevents the erase of eeprom and let upload fail

Thanks for helping

BR
Stefan

jantje, 2023-01-23:
I guess you want to upload over USB without a programmer.
In which case the programmer field is best left blank

stefanbbb, 2023-01-23:
I have a UPDI Programmer. The MCU is stand alone. From Arduino IDE it is working well. I managed to activate the protocol jtag2updi in Sloeber somehow, but I'm not able to remove the parameter "-D" from the command string.

At all, Sloeber is obviously ignoring the parameter set under preferences.
Can you help me out?

jantje, 2023-01-23:
As you use a programmer.. have you tried 
[image]
or the toolbar
[image]

stefanbbb, 2023-01-23:
Hi,

avrdude looks better now:
_avrdude.conf"   -patmega4809 -cjtag2updi -PCOM16 "-Uflash:w:E:\HomeControl\SW\Sloeber_workspace2\test\Release/test.with_bootloader.hex:i" "-Ufuse0:w:0x00:m" "-Ufuse1:w:0x54:m" "-Ufuse2:w:0x01:m" "-Ufuse4:w:0x00:m" "-Ufuse5:w:0b11001000:m" "-Ufuse6:w:0x06:m" "-Ufuse7:w:0x00:m" "-Ufuse8:w:0x00:m" "-Ulock:w:0xC5:m"_
But now, Sloeber is searching for **test.with_bootloader.hex**, but the file is named just **test.hex**:

_avrdude: reading input file "E:\HomeControl\SW\Sloeber_workspace2\test\Release/test.with_bootloader.hex"
avrdude: can't open input file E:\HomeControl\SW\Sloeber_workspace2\test\Release/test.with_bootloader.hex: No such file or directory
avrdude: read from file 'E:\HomeControl\SW\Sloeber_workspace2\test\Release/test.with_bootloader.hex' failed_

jantje, 2023-01-23:
I can confirm that this board has this behaviour.
I have not seen this before. This is something board specific.
```
21:06:55 **** Build of configuration Release for project test ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" uploadWithProgrammerWithoutBuild 
trying to upload with programmer avrdude without build
"C:\sloeber\arduinoPlugin\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude" "-CC:\sloeber\arduinoPlugin\packages\MegaCoreX\hardware\megaavr\1.0.10/avrdude.conf"   -patmega4809 -cjtag2updi -P "-Uflash:w:C:\workspaces\embedded2\test\Release/test.with_bootloader.hex:i" "-Ufuse0:w:0x00:m" "-Ufuse1:w:0x54:m" "-Ufuse2:w:0x01:m" "-Ufuse4:w:0x00:m" "-Ufuse5:w:0b11001000:m" "-Ufuse6:w:0x06:m" "-Ufuse7:w:0x00:m" "-Ufuse8:w:0x00:m" "-Ulock:w:0xC5:m"
avrdude: ser_open(): can't open device "-Uflash:w:C:\workspaces\embedded2\test\Release/test.with_bootloader.hex:i": The filename, directory name, or volume label syntax is incorrect.

avrdude done.  Thank you.

make: *** [makefile:83: uploadWithProgrammerWithoutBuild] Error 1
"C:/sloeber/arduinoPlugin/tools/make/make uploadWithProgrammerWithoutBuild" terminated with exit code 2. Build might be incomplete.

21:06:56 Build Failed. 1 errors, 0 warnings. (took 477ms)

```

stefanbbb, 2023-01-23:
By modifying your pre_processing_platform.txt I was able to get the protocoll jtag2updi to the commandstring:
[image]

But how can I remove the Parameter "-D"? If I would able to do that, I would have a workaround for this issue.

jantje, 2023-01-23:
I see the output is compliant with what I find in the platform.txt file
`tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i" {bootloader.fuse0} {bootloader.fuse1} {bootloader.fuse2} {bootloader.fuse4} {bootloader.fuse5} {bootloader.fuse6} {bootloader.fuse7} {bootloader.fuse8} {bootloader.lock}`

As you can see it states `/{build.project_name}.with_bootloader.hex:i`

Looking deeper I found that this board uses hooks Sloeber does not yet support
```
recipe.hooks.objcopy.postobjcopy.1.pattern.windows=cmd /C "{compiler.path}{compiler.objdump.cmd}" {compiler.objdump.flags} "{build.path}/{build.project_name}.elf" > "{build.path}/{build.project_name}_{build.mcu}_{build.f_cpu}.lst"
recipe.hooks.savehex.presavehex.1.pattern.windows="{runtime.platform.path}/scripts/delete_merged_output.bat" {build.export_merged_output} "{build.path}\{build.project_name}.with_bootloader.hex"
recipe.hooks.savehex.presavehex.2.pattern.windows=cmd /C copy "{build.path}\{build.project_name}_{build.mcu}_{build.f_cpu}.lst" "{sketch_path}"
```

I'm closing this as a duplicate of #927

As a workaround I tried to create a target to do the extra commands before the upload but that seems to fail on my system.
I attached a zipfile containing the makefile.targets. If you put that file in your project you can rightclick project/release/targets->add target name it "myUpload" 
[image]

When you double click the myUploadTarget and you should get

```
21:51:22 **** Build of configuration Release for project test ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" myUpload 
trying to upload with programmer avrdude without build
cmd /C echo. && "C:\sloeber\arduinoPlugin\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7/bin/avr-objdump" --disassemble --source --line-numbers --demangle --section=.text "C:\workspaces\embedded2\test\Release/test.elf" > "C:\workspaces\embedded2\test\Release/test_atmega4809_16000000L.lst"
 
"C:\sloeber\arduinoPlugin\packages\MegaCoreX\hardware\megaavr\1.1.1/scripts/delete_merged_output.bat" true "C:\workspaces\embedded2\test\Release\test.with_bootloader.hex"
cmd /C copy "C:\workspaces\embedded2\test\Release\test_atmega4809_16000000L.lst" ""
        0 file(s) copied.
"C:\sloeber\arduinoPlugin\packages\MegaCoreX\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\sloeber\arduinoPlugin\packages\MegaCoreX\hardware\megaavr\1.1.1/avrdude.conf"   -patmega4809 -cjtag2updi -P -Ufuses:w:0x00,0x54,0x01,0xff,0x00,0b11001000,0x06,0x00,0x02:m -Ulock:w:0xC5:m "-Uflash:w:C:\workspaces\embedded2\test\Release/test.with_bootloader.hex:i"
avrdude error: cannot open port -Ufuses:w:0x00,0x54,0x01,0xff,0x00,0b11001000,0x06,0x00,0x02:m: The filename, directory name, or volume label syntax is incorrect.

avrdude error: unable to open programmer jtag2updi on port -Ufuses:w:0x00,0x54,0x01,0xff,0x00,0b11001000,0x06,0x00,0x02:m

avrdude done.  Thank you.

make: *** [..\makefile.targets:6: myUpload] Error 1
"C:/sloeber/arduinoPlugin/tools/make/make myUpload" terminated with exit code 2. Build might be incomplete.

21:51:23 Build Failed. 1 errors, 0 warnings. (took 696ms)
```

[makefile.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/10483583/makefile.zip)

jantje, 2023-01-23:
Note that your workaround won't work as you use the upload parameters not the program parameters.
The question is: What do these commands do and is that needed to program the board

Also note that you can create any target you want and provide it any instruction you want. So that is an easy way to hardcode commands.

closed 2023-01-23 by jantje