Sloeber/arduino-eclipse-plugin #1552: openOCD failed to connect to target on STM32

MasteringTheMess, 2023-01-24:
I am trying to debug a Nucleo-L4R5

I had it working a while ago, but now I get the following error output

 ```
   xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-03-25-17:32)
    Licensed under GNU GPL v2
    For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
    Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
    Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
    Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
    Started by GNU MCU Eclipse
    Info : Listening on port 6666 for tcl connections
    Info : Listening on port 4444 for telnet connections
    Info : clock speed 500 kHz
    Info : STLINK V2J40M27 (API v2) VID:PID 0483:374B
    Info : Target voltage: 3.241253
    Error: init mode failed (unable to connect to the target)
```

And some additional info in a dialog:

```
    Error in final launch sequence
    Failed to execute MI command:
    -target-select remote localhost:3333
    Error message from debugger back end:
    localhost:3333: Connection timed out.
    Failed to execute MI command:
    -target-select remote localhost:3333
    Error message from debugger back end:
    localhost:3333: Connection timed out.
    localhost:3333: Connection timed out.
```

I followed the instructions here:
[](https://github.com/stm32duino/wiki/wiki/How-to-debug)

I can normally program the Nucleo via STM32CubeProgrammer / Arduino IDE / sloeber

The error is independent of the eclipse perspective I use

Is this hardware related or is something wrong with the GDB/openOCD part?
I am just starting to get into the subject...

Was it a bad idea to install the GNU-MCU plugin? I only read after installing that sloeber might have its own GDB/openOCD...

jantje, 2023-01-24:
### Context
As you may know Sloeber does not "really" provide debugging capabilities.
Let me explain what I mean by that.
**Eclipse** provides a framework for IDE's (integrated development environment) including compilation and debugging
**CDT** plug is into the eclipse framework to compile, debug test C/C++ code.
**Sloeber plugin** plugs into CDT to compile using the instructions provided by the Arduino framework. (not debug nor test)
**Sloeber plugin** provides ways to change the arduino framework provided compile options so that it is possible to make CDT create an executable with debug information in it.
**Sloeber plugin** -latest version- plugs into eclipse environment variables to make the environment variables of the current project active configuration available. These environment variables reflect the content of the *.txt files and can be used to make setting up a debug target easier.

**Sloeber plugin** is dependent on plugins that get installed when installing the plugin (some nebula/eclipse/CDT plugins) 
**Sloeber product** bundels the sloeber plugin with a selection of eclipse plugins and CDT plugins and some third party plugins (like git/GNU-MCU). Therefore the Sloeber product has all needed plugins to allow to start a debugging session and do version control.
Adding the Sloeber plugin to a eclipse install does not add the debugging/git capability to the eclipse install.
As far as I know the GNU-MCU plugin moved into the CDT space and therefore got (technically) renamed. Installing the old and new in the same eclipse install will at least cause confusion.

So if you installed the product version of Sloeber it contains a GDB/openOCD plugin from CDT/GNU-MCU
If you added the plugin version of Sloeber to a eclipse install; Sloeber did not add a GDB/openOCD plugin.
### Note 
1) the GDB/OpenOCD plugins only enable to communicate between gdb/openOCD and eclipse. 
You need an actual gdb and openOCD (which should be provided by the Arduino board provider)
2) There are several providers of GDB and OpenOCD eclipse plugins
3) the plugin model is very provider/version dependent. Installing a different version will/may cause problems. That is why we only support "Sloeber product out of the box".
4) Because debugging is hard to test and is dependent on third party plugins we do not test debugging capabilities before releasing.

### My advice
If you have problems with GDB/openOCD in a updated Sloeber product or the sloeber plugin :
1) de-install all GDB/openOCD plugins currently installed. Then install a consistent GDB/openOCD version and get support from there.
2) try with a brand new latest version out of the box Sloeber product. We will try to support you as good as we can.

I hope this helps

MasteringTheMess, 2023-01-25:
All right, I think I already know what may cause my problems!

I installed Sloeber product AND installed the GNU-MCU (renamed to "The Eclipse Embedded CDT") over it.
I will start again with a fresh install of Sloeber product.

Thanks so far, jantje!