Sloeber/arduino-eclipse-plugin #1553: [STM32] opt file/SrcWrapper.h workaround

tyllurius, 2023-02-03:
I still have issue #1538  with STM32 and SrcWrapper.h and can't get it to work even with the described workaround.
I get `arduino/Arduino.h:33:12: fatal error: SrcWrapper.h: No such file or directory` though I use nightly build with STM32 2.4.0 on fresh Windows 11 and applied the workaround in #1534 . Maybe it's because there's still an error in the platform.txt, because it's stated

> and
> replace
> recipe.hooks.prebuild.1.pattern="${runtime.tools.STM32Tools.path}/win/busybox.exe" sh "${extras.path}/prebuild.sh" "${build.path}" "${build.source.path}" "${runtime.platform.path}"
> with
> #recipe.hooks.prebuild.1.pattern="${runtime.tools.STM32Tools.path}/win/busybox.exe" sh "${extras.path}/prebuild.sh" "${build.path}" "${build.source.path}" "${runtime.platform.path}"

but both entries are same? Also, isn't this what should be the result in the platform.sloeber.txt (because of the $)?

Also, in my platform.txt there are two entries 
```
recipe.hooks.prebuild.1.pattern="{extras.path}/prebuild.sh" "{build.path}" "{build.source.path}" "{runtime.platform.path}"
recipe.hooks.prebuild.1.pattern.windows="{runtime.tools.STM32Tools.path}/win/busybox.exe" sh "{extras.path}/prebuild.sh" "{build.path}" "{build.source.path}" "{runtime.platform.path}"
```
which one should be changed? Read all related topics but can't get any step further... Hope this can be clarified as it seems there are some STM32 users running into this.

jantje, 2023-02-03:
>but both entries are same? 

note the # in the beginning which comments out the line

>Also, isn't this what should be the result in the platform.sloeber.txt (because of the $)?

You are absolutely right.

>Also, in my platform.txt there are two entries, which one should be changed?

At least the windows one when you are on windows.
At least the other one if you are not on windows
Or play safe and do both.

If the indexer is not working properly (due to a bad command line entry from Sloeber) it won't detect the missing SrcWrapper.h and won't add the library.
Manually adding the SrcWrapper library via the Sloeber menu "add libraries" can fix that but then you still have a wonky indexer and that makes lots of functionality is lost.

tyllurius, 2023-02-05:
Oh ok, didn't notice the #...
I commented out the line in platforms.txt, but still it couldn't find the SrcWrapper. What it changed it that Windows with which application it should open a .sh file.
Then I opened up the Sloeber menu "add libraries" - where SrcWrapper was already selected - and closed with Finish which causes some reindexing, and with that it worked, now I can compile every project successfully.
So, this workaround works for me, but what exacly did it? Only the reindexing? And how shall I continue with the #?

jantje, 2023-02-05:
> What it changed it that Windows with which application it should open a .sh file.

> >At least the windows one when you are on windows.
> At least the other one if you are not on windows
> Or play safe and do both.

I guess you didn't play safe. Now I think about it; you need to disable both.

>So, this workaround works for me, but what exacly did it? Only the reindexing? And how shall I continue with the #?

It is really complicated and a in depth understanding of sloeber and the arduino framework is needed to get to all details.
But in short: gcc allows to provide options via a file. If the file does not exists gcc fails.
Newer versions of STM and esp32 use this type of file to allow the end user to change the command line.
Due to gcc failing when the file is not there; a prebuild command makes sure the file is there.

But sloeber uses the gcc command to discover the compile options. 
The file will only be there after a first build. Discovery is run very rarely and is not easy to trigger.
Therefore discovery/indexing/adding libraries is all wonky (at best) in Sloeber
Leading to stuff like ScrWrapper.h not found (which has a workaround on it's own for Sloeber by the way)

The solution is 2 fold
1) remove the compile option file from the command in platform.txt
2) remove the prebuild command in platform.txt

OK; but what if I want to use command line extensions or when I have such options file?
1) Sloeber has it's own way of extending the command line in project properties->sloeber
2) You can use 1 to add the options file to the command line.

And yes this is really the short version.

tyllurius, 2023-02-05:
> 
> I guess you didn't play safe. Now I think about it; you need to disable both.
> 
Indeed I only disabled one. Now I disabled both entries and the message is gone.
> 
> And yes this is really the short version.
> 
Thank you for outlining this complicated issue. It's a good summary of the clue users get when reading related issues, like #1546 #1143 

So, to summarize the workaround, for a fresh sloeber install, nightly build, with newer STM32, and migrated projects/workspace from earlier sloeber versions:

- edit platform.txt

1. remove the compile option file

`compiler.extra_flags=-mcpu={build.mcu} {build.fpu} {build.float-abi} -DUSE_FULL_LL_DRIVER -mthumb`

2. remove the prebuild command
```
#recipe.hooks.prebuild.1.pattern="{extras.path}/prebuild.sh" "{build.path}" "{build.source.path}" "{runtime.platform.path}"
#recipe.hooks.prebuild.1.pattern.windows="{runtime.tools.STM32Tools.path}/win/busybox.exe" sh "{extras.path}/prebuild.sh" "{build.path}" "{build.source.path}" "{runtime.platform.path}"
```

- propagate changes #1533 
- "Add a library..", SrcWrapper should be checked, click Finish

Thanks again for clarifying this

closed 2023-02-05 by tyllurius