Sloeber/arduino-eclipse-plugin #1554: `{_id}` build property is empty

mcspr, 2023-02-04:
From https://github.com/esp8266/Arduino/commit/06f34ed566ae17c3ac3e1841b54caba44043df8f#commitcomment-97671105 by @kagouraki

> because of these changes, i cant compile with the 3.1.1 version on sloeber.
i get this error 
```
/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS/src/LEAmDNS.cpp: In member function 'const void* esp8266::MDNSImplementation::MDNSResponder::enableArduino(uint16_t, bool)':
/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS/src/LEAmDNS.cpp:1275:71: error: expected primary-expression before ')' token
 1275 |                 || (!addServiceTxt(hService, "board", ARDUINO_BOARD_ID))
      |                                                                       ^
```

> 
```
Building file: /home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS/src/LEAmDNS.cpp
Starting C++ compile
"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266  "-I/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/include" "-I/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/lwip2/include" "-I/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/home/labcnc/eclipse-workspaceNew/wifiToNmea183withServer/Release/core" -c "@/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/tools/warnings/none-g++"  -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -MMD -ffunction-sections -fdata-sections -fno-exceptions  -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000  -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10812 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DARDUINO_BOARD_ID="" -DLED_BUILTIN=2 -DFLASHMODE_DOUT     -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/cores/esp8266" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/variants/generic" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/libraries/ESP_EEPROM/2.1.2/src" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/DNSServer/src" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ArduinoOTA" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS/src" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266WebServer/src" -I"/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266WiFi/src" -I"/home/labcnc/Arduino/libraries/WiFiManager" -MMD -MP -MF"libraries/ESP8266mDNS/src/LEAmDNS.cpp.d" -MT"libraries/ESP8266mDNS/src/LEAmDNS.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS/src/LEAmDNS.cpp"   -o "libraries/ESP8266mDNS/src/LEAmDNS.cpp.o"
/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS/src/LEAmDNS.cpp: In member function 'const void* esp8266::MDNSImplementation::MDNSResponder::enableArduino(uint16_t, bool)':
/home/labcnc/eclipse/Sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/3.1.1/libraries/ESP8266mDNS/src/LEAmDNS.cpp:1275:71: error: expected primary-expression before ')' token
 1275 |                 || (!addServiceTxt(hService, "board", ARDUINO_BOARD_ID))
      |                                                                       ^
make[1]: *** [libraries/ESP8266mDNS/src/subdir.mk:42: libraries/ESP8266mDNS/src/LEAmDNS.cpp.o] Error 1
make: *** [makefile:97: all] Error 2
"make all" terminated with exit code 2. Build might be incomplete.
```

Sloeber generates `-DARDUINO_BOARD_ID=""` instead of the expected `-DARDUINO_BOARD_ID="generic"` (as boards.txt `generic.`... that is selected)
```console
> arduino-cli compile -b esp8266com:esp8266:generic:eesz=4M1M --show-properties | grep _id
_id=generic
```

From the platform spec
https://arduino.github.io/arduino-cli/0.29/platform-specification/#global-predefined-properties 
> {_id}: [board ID](https://arduino.github.io/arduino-cli/0.29/platform-specification/#boardstxt) of the board being compiled for

jantje, 2023-02-07:
I havn't had time to look into this but it seems very probable that this is true.

SturmWolfgang, 2023-05-07:
came across the same issue today. I am not the build expert. What I can confirm is:
building LEAmDNS.cpp in plattform 3.1.2 with ESP8266 "Huzzah" &  sloeber 4.4.0 does not work, as ARDUINO_BOARD_ID undefined, see error @mcspr 
in the console the file is complied with -DARDUINO_BOARD_ID=""
with platform 3.0.2 everything is fine
any proposal how to fix this in 3.1.2?

jantje, 2023-05-07:
The only workaround I see right now is to change the boards.txt for all boards you use with this library.
Below is the change I did for uno

```
uno.name=Arduino Uno
uno._id=uno
```
After executing the necessary actions to update the caching https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1533 you should see the _id in the environment variables

[image]

jantje, 2024-08-18:
Fixed as part of #1625

closed 2024-08-18 by jantje