Sloeber/arduino-eclipse-plugin #1557: Cannot install Sloeber V4.4.1 on Linux

RobertPatzke, 2023-03-19:
I have 2 PCs (64 Bit) running Sloeber V4.4.0 without problems: Linus Mint (Targa, based on Ubuntu 18.04) and Ubuntu 20.04.
Now I want to Update to version 4.4.1.
I downloaded 64 Bits Bundle from http://eclipse.baeyens.it/stable.php?OS=Linux and de-zipped it in a folder Sloeber in my home directory.
I followed the steps given with the latest version.
> chmod u+x sloeber-ide

Then I started sloeber-ide (via double mouse click in file manager, the same behavior via terminal) and a pop-up-window reports:
> VM terminated. Exit code=13
/home/rp/Sloeber//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/java
-Dosgi.requiredJavaVersion=11
-Dorg.eclipse.update.reconcile=false
-Dlog4j.configuration=log4j/log4j.xml
-Xms256m
-Xmx1024m
-jar /home/rp/Sloeber//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
-os linux
-ws gtk
-arch x86_64
-showsplash
-launcher /home/rp/Sloeber/sloeber-ide
-name Sloeber-ide
--launcher.library /home/rp/Sloeber//plugins/org.eclipse.equinox.launcher.gtk.linux.x86_64_1.2.500.v20220509-0833/eclipse_11701.so
-startup /home/rp/Sloeber//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
--launcher.appendVmargs
-exitdata 3c
-perspective io.sloeber.application.perspective
-vm /home/rp/Sloeber//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/bin/java
-vmargs
-Dosgi.requiredJavaVersion=11
-Dorg.eclipse.update.reconcile=false
-Dlog4j.configuration=log4j/log4j.xml
-Xms256m
-Xmx1024m
-jar /home/rp/Sloeber//plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar

Then I started

> chmod +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/bin/*

Everything seems OK, Eclipse/Sloeber starts and begins downloading and installing resources. But then it stops with the following error in a small Pop-up:
> Failed to install Arduino AVR Boards
Failed to extract tar.bz2. "/home/rp/Sloeber/arduinoPlugin/downloads/avr-1.8.6.tar.bz2".
Cannot run program "chmod": error=13, Keine Berechtigung

The click on Error Log gives the following information (copied lines to text editor):
> eclipse.buildId=unknown
java.version=17.0.3
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=de_DE
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os linux -ws gtk -arch x86_64 -perspective io.sloeber.application.perspective

> org.eclipse.urischeme
Error
Sun Mar 19 19:02:00 CET 2023
Cannot run program "xdg-mime": error=13, Keine Berechtigung

> java.io.IOException: Cannot run program "xdg-mime": error=13, Keine Berechtigung
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1143)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1073)
	at org.eclipse.urischeme.internal.registration.ProcessExecutor.execute(ProcessExecutor.java:36)
	at org.eclipse.urischeme.internal.registration.RegistrationLinux.getRegisteredDesktopFileForScheme(RegistrationLinux.java:145)
	at org.eclipse.urischeme.internal.registration.RegistrationLinux.determineHandlerLocation(RegistrationLinux.java:87)
	at org.eclipse.urischeme.internal.registration.RegistrationLinux.getSchemesInformation(RegistrationLinux.java:76)
	at org.eclipse.urischeme.AutoRegisterSchemeHandlersJob.run(AutoRegisterSchemeHandlersJob.java:87)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:63)
Caused by: java.io.IOException: error=13, Keine Berechtigung
	at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
	at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:314)
	at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:244)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1110)
	... 7 more

It is a pity, that my know-how is to less to have any idea for solving this problem.
At the time I have to continue with version 4.4.0.
I just try to install remote debugging (and compiling) on Raspberry Pi (3 B and 4 B) with my Eclipse/Sloeber and for this I wanted to use the latest version. I have that Raspberry Pi remote debugging already running with Eclipse CDT 2022-12 and wanted to test step by step, if it is possible with Sloeber. But now I think it is more import to have version 4.4.1 running at all.

jantje, 2023-03-19:
>Failed to extract tar.bz2. "/home/rp/Sloeber/arduinoPlugin/downloads/avr-1.8.6.tar.bz2".
Cannot run program "chmod": error=13, Keine Berechtigung

For safety reasons all osses are defaulting to as little as possible read/write acess. This causes plenty of issues for software installation. 
In this case it looks like sloeber lacks write access to the folder /home/rp/Sloeber/arduinoPlugin/packages
I say so because the first thing Sloeber does after install is downloading tools (make/awk) -only windows- and the latest Arduino avr platform.
avr-1.8.6.tar.bz2 clearly point to the download of the latest arduino avr platform.
Sloeber first downloads to the downloadsFolder and then extracts to the appropriate place. In this case that should be /home/rp/Sloeber/arduinoPlugin/packages/arduino/tools/avr-gcc/.....
During the extraction a chmod is executed https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/PackageManager.java#L259
Which (on linux) tries to run the chmod command https://github.com/Sloeber/arduino-eclipse-plugin/blob/0076c522e385a37af0f245e2354d158bb6389a91/io.sloeber.core/src/io/sloeber/core/tools/PackageManager.java#L336
Which seems to fail due to the os not entitling Sloeber to do so.
Note that Sloeber allows for setting the SLOEBER_HOME environment variable. In that case the Sloeber install folder will be ${SLOEBER_HOME}/arduinoPlugin
I hope this helps

davedeaven, 2023-03-19:
I just ran into the same issue. It appears from my initial look that the issue is with the JRE file:
`plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_17.0.3.v20220515-1416/jre/lib/jspawnhelper`
which is used by java as a wrapper around the failing commands forked by the java Process class. This needs the same treatment that the cheatsheet says the the `.../jre/bin/*` need, suggest adding the following to the pre-install checklist:
`chmod +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/lib/jspawnhelper`
My install is running now.

jantje, 2023-03-19:
Maybe a chmod -R plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre should be a better way. What do you think?

davedeaven, 2023-03-20:
That should work... I just confirmed a fresh install with
```
unzip V4.4.1_linux64.2022-07-09_05-48-31.zip
cd Sloeber
chmod +x sloeber-ide 
chmod -R +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre
./sloeber-ide
```
Just to make sure no typos I copied and pasted from the window I used.

jantje, 2023-03-20:
> because some parameters seem to have changed with 4.4.1

I'm not sure what you are referring to but mostly when people say "parameters changed in Sloeber" they actually installed a different platform version.

RobertPatzke, 2023-03-24:
Yes, that is true. For a short test I used the latest versions of Arduino tools and in my projects I seldom change the tool version.
I will continue testing, but that will not have a relation to the opened issue.
So I will close this issue.
Again @jantje  and @davedeaven : Thank You very much.

closed 2023-03-24 by RobertPatzke

jcp-sd, 2023-04-21:
Same problem here.

I would **strongly** urge the developers to distribute this as a compressed tar archive in the future. This problem then becomes a complete non-issue because tar (short for tape _archiver_) properly restores directory/file permissions and ownership, hard/soft links, sockets, etc.  Linux users will certainly know how to use tar, so it becomes a simple untar, and run situation.

Yes, ZIP is more portable but it's totally not required here (as nobody installs the Linux version on MacOS or Windows).

I determined the specific permission fixes required by looking at the executable files in the Debian (Bullseye 11.6) packaged version of JDK/JRE 17.  Here is a work-around I tested (and it worked perfectly for me);

```
$ cd Sloeber

$ chmod u+x sloeber-ide

$ chmod u+x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/bin/*

$ chmod u+x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/lib/jexec 

$ chmod u+x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/lib/jspawnhelper 

```

uzi18, 2023-07-10:
@jantje  7zip application on Windows also knowns tar archive :)
or provide script inside Sloeber folder to recreate permissions.

jantje, 2023-07-10:
@uzi18 We tried with tar an 7zip.
Lots of complaints for "additional software requirements"
My 2 cent: Just build the damn thing locally; after all; you want to do software development; don't you?

Alex-Kent, 2023-08-08:
This is fixed in the [pull request](https://github.com/Sloeber/arduino-eclipse-plugin/pull/1579) that I just made.  (Changed package format on Linux from `zip` to `tar.gz`)