Sloeber/arduino-eclipse-plugin #1563: Application does not launch on MacOS Ventura, on M1 chip.

baudhuina, 2023-05-11:
**Environment**   
Sloeber product v4.4.1
OS: MacOS Ventura (13.3.1) on Apple M1 MacBook Air.

**The issue**   
After following all MacOS installation instructions from http://eclipse.baeyens.it/stable.php?OS=MacOS, launching the app always causes the "Cannot open Sloeber.app" error.

**To Reproduce**   
1. Download from https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/V4_4_1_0/V4.4.1_mac64.2022-09-05_06-48-31.zip
2. Copy zip file to /Applications and extract there with MacOS Archive Utitlity
3. In Terminal: sudo xattr -r -d com.apple.quarantine Sloeber.app
4. Double-click on Sloeber.app in Application folder. 
Dialog (from Gatekeeper): Cannot open "Sloeber.app", with Finder icon. 
(reproduced many different times, no partial download issue). 

**Investigation**   
- The MacOS Gatekeeper has changed in MacOS Ventura, and some sources claim that stripping the quarantine flag as stated in the installation procedure won't work anymore, as the checks previously performed only on the first launch are now performed at every launch. See e.g. https://eclecticlight.co/2022/09/24/why-some-apps-wont-run-in-ventura-and-how-to-fix-it/  Given this change, is Sloeber.app expected to run on MacOS Ventura? 
- Trying to understand why the Gatekeeper refuses to run Sloeber, I can confirm that the application has a properly identified developer (confirmed in the System Information report, tab Applications), but trying to check whether the application is recognzied as notarized with `spctl -a -vvv -t install Sloeber.app` returns the following error: 
`Sloeber.app: code has no resources but signature indicates they must be present`
- Thought of veryfing the app checksum but I couldn't find the expected value. 

**Status**
I'm very willing to perform additional tests, but I'm currently out of ideas... Thanks in advance for any suggestion!

baudhuina, 2023-05-11:
I'll try that. Nevertheless, my tests were about the product, not the plugin and those instructions seem to be for the plugin?

baudhuina, 2023-05-11:
Well, the build fails: apparently not happy with my Java environment:
`[ERROR] Internal error: org.eclipse.tycho.core.ee.UnknownEnvironmentException: Unknown OSGi execution environment: 'JavaSE-20' -> [Help 1]`
Any prerequisite I should be aware of regarding the installed Java version? 
Note that the version of Java used, seems to be the one installed by `brew install mvn`. It is located in /usr/local/Cellar/openjdk:
`ls /usr/local/Cellar/openjdk`
`20.0.1/`
 I'm not clear why this JavaSE-20 is used rather than another  installed version:
result of `which java`:
`/usr/bin/java`
Result of 'java --version' :
'java 15.0.2 2021-01-19
Java(TM) SE Runtime Environment (build 15.0.2+7-27)
Java HotSpot(TM) 64-Bit Server VM (build 15.0.2+7-27, mixed mode, sharing)'
Result of `/usr/libexec/java_home -V`:
```
Matching Java Virtual Machines (3):
    15.0.2 (x86_64) "Oracle Corporation" - "Java SE 15.0.2" /Library/Java/JavaVirtualMachines/jdk-15.0.2.jdk/Contents/Home
    1.8.341.10 (x86_64) "Oracle Corporation" - "Java" /Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home
    1.8.0_101 (x86_64) "Oracle Corporation" - "Java SE 8" /Library/Java/JavaVirtualMachines/jdk1.8.0_101.jdk/Contents/Home
/Library/Java/JavaVirtualMachines/jdk-15.0.2.jdk/Contents/Home
------------------------------
Output of `>mvn clean verify -DskipTests=true` command:
(large number of successful downloads not copied)
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[ERROR] Internal error: org.eclipse.tycho.core.ee.UnknownEnvironmentException: Unknown OSGi execution environment: 'JavaSE-20' -> [Help 1]
org.apache.maven.InternalErrorException: Internal error: org.eclipse.tycho.core.ee.UnknownEnvironmentException: Unknown OSGi execution environment: 'JavaSE-20'
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:109)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:910)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:283)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:206)
    at jdk.internal.reflect.DirectMethodHandleAccessor.invoke (DirectMethodHandleAccessor.java:104)
    at java.lang.reflect.Method.invoke (Method.java:578)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:283)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:226)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:407)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:348)
Caused by: org.eclipse.tycho.core.ee.UnknownEnvironmentException: Unknown OSGi execution environment: 'JavaSE-20'
    at org.eclipse.tycho.core.ee.ExecutionEnvironmentUtils.getExecutionEnvironment (ExecutionEnvironmentUtils.java:96)
    at org.eclipse.tycho.core.osgitools.OsgiBundleProject.applyBestOfCurrentOrConfiguredProfile (OsgiBundleProject.java:778)
    at org.eclipse.tycho.core.osgitools.OsgiBundleProject.readExecutionEnvironmentConfiguration (OsgiBundleProject.java:748)
    at org.eclipse.tycho.core.resolver.DefaultTychoResolver.setupProject (DefaultTychoResolver.java:103)
    at org.eclipse.tycho.core.maven.TychoMavenLifecycleParticipant.afterProjectsRead (TychoMavenLifecycleParticipant.java:111)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:223)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:173)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:101)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:910)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:283)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:206)
    at jdk.internal.reflect.DirectMethodHandleAccessor.invoke (DirectMethodHandleAccessor.java:104)
    at java.lang.reflect.Method.invoke (Method.java:578)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:283)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:226)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:407)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:348)
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/InternalErrorException

jantje, 2023-05-11:
To run/build of Sloeber latest version needs at least Java 17 (because CDT does) so the only java version you have installed that matches that requirement is version 20.
I fear there is a problem with java version 20 and tycho 2.7.5 specified in the pom file https://github.com/Sloeber/arduino-eclipse-plugin/blob/0076c522e385a37af0f245e2354d158bb6389a91/io.sloeber.parent/pom.xml#LL12C18-L12C24
Can you try with java 17 installed?

@wimjongman (my usual saviour in these matters) can you shine your light on this?
As a side thought: Maybe we should specify a git command to get a release version and not the latest. Not that I think it would be better in this case as the build will bundle the latest CDT which will require java 17.

baudhuina, 2023-05-12:
Progressing....
Installed java17 with `sdk install java 17.0.7-amzn`
Build now succeeds, but: 
Running `./build_then_launch.sh` gives: everything ok until:
`[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  58.807 s
[INFO] Finished at: 2023-05-12T10:28:49+02:00
[INFO] ------------------------------------------------------------------------
[WARNING] 
[WARNING] Plugin validation issues were detected in 13 plugin(s)
[WARNING] 
[WARNING]  * org.eclipse.tycho:tycho-versions-plugin:2.7.5
[WARNING]  * org.apache.maven.plugins:maven-site-plugin:3.12.1
[WARNING]  * org.eclipse.tycho:tycho-p2-repository-plugin:2.7.5
[WARNING]  * org.eclipse.tycho:tycho-p2-publisher-plugin:2.7.5
[WARNING]  * org.apache.maven.plugins:maven-surefire-plugin:2.22.2
[WARNING]  * org.apache.maven.plugins:maven-failsafe-plugin:2.22.2
[WARNING]  * org.apache.maven.plugins:maven-enforcer-plugin:1.2
[WARNING]  * org.apache.maven.plugins:maven-enforcer-plugin:1.0
[WARNING]  * org.eclipse.tycho:tycho-ds-plugin:2.7.5
[WARNING]  * org.eclipse.tycho:tycho-surefire-plugin:2.7.5
[WARNING]  * org.apache.maven.plugins:maven-resources-plugin:2.4.3
[WARNING]  * org.eclipse.tycho:tycho-packaging-plugin:2.7.5
[WARNING]  * org.eclipse.tycho:tycho-compiler-plugin:2.7.5
[WARNING] 
[WARNING] For more or less details, use 'maven.plugin.validation' property with one of the values (case insensitive): [BRIEF, DEFAULT, VERBOSE]
[WARNING] 
Searching for the Eclipse executable (with our plugin pre-packaged) to launch
Did not find an Eclipse Arduino IDE executable built.....`

**NEVERTHELESS**:
There is a Sloeber.app in folder `sloeber/io.sloeber.product/target/products/io.sloeber.product/macosx/cocoa/aarch64` and double-clicking on it ***launches the app smoothly***. 

So:

- there is definitely a security issue with the distributed build on MacOs Ventura
- there should be some additional instructions added for the build on MacOs (Ventura) currently: 
Install java17 (not higher) with: 
`curl -s "https://get.sdkman.io" | bash  ` (to install sdkman if not installed already)
`sdk install java 17.0.7-amzn` (in a new shell if you just installed sdkman)
- there is a tweek to add in `build_then_launch.sh` because it does not correctly identify the executable to run on MacOs (Ventura ? )

but I'm temporarily out of the woods. Thanks for your help, I'll try not to bother you anymore :-)

jantje, 2023-05-12:
Thanks for the input

>there is definitely a security issue with the distributed build on MacOs Ventura

I'm not sure about that. I think downloaded apps are treated more harshly than locally created apps.

I'll try to properly integrate your advice in the read.me file

baudhuina, 2023-05-12:
Ok, my bad: of course, there isn't an **actual** security issue, just a security check which fails.  I just meant I can download apps and install them (e.g. Eclipse CDT), while I cannot do the same with Sloeber. So, there is something to be done to be able to have the app recognized as safe, if executable distribution remains the objective.

jantje, 2023-05-12:
That being said I fear problems will arise one day with the Arduino framework where compilers and uploaders are downloaded and run. I'm not really sure why I haven't seen problems yet in this area.
I'm probably overlooking something

wimjongman, 2023-05-12:
Jan, there are now TWO (2) downloads for the mac. One is the classic 64 bits and the other is the new builds for the M1 or ARM architecture. 

I have already changed the build a while ago to also produce the M1/ARM builds but they cannot be found in the release record and also not in the baayens website.

Try this: https://github.com/Sloeber/arduino-eclipse-plugin/suites/12857480725/artifacts/693183718

wimjongman, 2023-05-12:
If you want to build locally for macm1 try this:

* git clone https://github.com/Sloeber/arduino-eclipse-plugin.git
* cd arduino-eclipse-plugin
* mvn clean verify -Pmacm1,latest,NOSDK -DskipTests=true

After this you will find the m1 builds in the product/target directory

baudhuina, 2023-05-12:
mmh, I just modified 'build_then_launch.sh' to properly run the MacOS app after build, and in the last 30 minutes, the build with `mvn clean verify -DskipTests=true`stopped working... Could it be linked to the changes for the M1 build mentioned above? 

Hereunder the updated version (just updated to properly detect a MacOS application bundle and launch it). 
[build_then_launch.sh.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/11465194/build_then_launch.sh.zip)

wimjongman, 2023-05-12:
Thanks for the feedback.

closed 2023-05-12 by wimjongman

jantje, 2023-05-12:
I just love it when people work together to fix an issue.

>Jan, there are now TWO (2) downloads for the mac. One is the classic 64 bits and the other is the new builds for the M1 or ARM architecture.

I wasn't aware of that. I'll take it into account when doing the next release (I hope I will remember 🙄 )