Sloeber/arduino-eclipse-plugin #1567: Renaming a perfectly operational project fails, and causes next build to fail

baudhuina, 2023-05-16:
**Environment**
Sloeber product on MacOS Ventura (M1)
Bug encountered and reproduced  with a project name "aaa", on board "Adafruit Feather Huzza ESP8266".

**Description**
Renaming the project from 'aaa' to 'bbb' leaves include paths with the old name, and causes an error during the rename and during every subsequent build.

**How to reproduce**
1. Rename project from "aaa" to "bbb" (using the contextual menu on the project entry in the project explorer).
`Error:
Cannot invoke "org.eclipse.core.runtime.IPath.toOSString()" because the return value of "org.eclipse.core.resources.IProject.getLocation()" is null
Cannot invoke "org.eclipse.core.runtime.IPath.toOSString()" because the return value of "org.eclipse.core.resources.IProject.getLocation()" is null`

Nevertheless: the project is renamed, the project folder is renamed.
NB: The project location seems perfectly OK before the rename: generation of file sloeber.ino.cpp works perfectly and happens in the project folder, after detection of the ino file in the project folder. 

2. Build project bbb. It now fails with error:
`Errors occurred during the build.
Errors running builder 'Scanner Configuration Builder' on project 'test_ESP8266_RF_Ljink'.
Cannot invoke "org.eclipse.core.runtime.IPath.toOSString()" because the return value of "org.eclipse.core.resources.IProject.getLocation()" is null`

**Analysis and workaround**
Apparently, this is because the include paths, visible in `properties/C/C++ build/Settings/ArduinoC++ Compiler/include folders` still contain the previous name of the project. 
Those paths are:
`"${workspace_loc:/aaa/core/variant}"
"${workspace_loc:/aaa/core/core}"`

Manually updating each include path, replacing 'aaa' with 'bbb' fixes the build error.

jantje, 2023-05-24:
It seems there is a regression test for renaming a configuration but not for renaming a project. 😞

jantje, 2023-05-24:
I got a stacktrace during the rename.
Setting a breakpoint I could see the code is only triggered once and tries to rename one of the include directories
[image]

```
!ENTRY org.eclipse.ltk.ui.refactoring 4 10000 2023-05-25 00:57:25.868
!MESSAGE Internal Error
!STACK 0
org.eclipse.core.runtime.AssertionFailedException: null argument:
	at org.eclipse.core.runtime.Assert.isNotNull(Assert.java:88)
	at org.eclipse.core.runtime.Assert.isNotNull(Assert.java:76)
	at org.eclipse.core.filesystem.URIUtil.toPath(URIUtil.java:63)
	at org.eclipse.core.internal.resources.Workspace.copyVariable(Workspace.java:1205)
	at org.eclipse.core.internal.resources.Workspace.transferVariableDefinition(Workspace.java:1181)
	at org.eclipse.core.internal.resources.Workspace.copyTree(Workspace.java:1117)
	at org.eclipse.core.internal.resources.Workspace.copyTree(Workspace.java:1169)
	at org.eclipse.core.internal.resources.Workspace.copyTree(Workspace.java:1169)
	at org.eclipse.core.internal.resources.Workspace.move(Workspace.java:2111)
	at org.eclipse.core.internal.resources.ResourceTree.movedProjectSubtree(ResourceTree.java:685)
	at org.eclipse.core.internal.resources.ResourceTree.standardMoveProject(ResourceTree.java:1108)
	at org.eclipse.core.internal.resources.Resource.unprotectedMove(Resource.java:1850)
	at org.eclipse.core.internal.resources.Resource.move(Resource.java:1513)
	at org.eclipse.ltk.core.refactoring.resource.RenameResourceChange.perform(RenameResourceChange.java:118)
	at org.eclipse.ltk.core.refactoring.CompositeChange.perform(CompositeChange.java:281)
	at org.eclipse.ltk.core.refactoring.PerformChangeOperation.lambda$0(PerformChangeOperation.java:258)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2380)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2405)
	at org.eclipse.ltk.core.refactoring.PerformChangeOperation.executeChange(PerformChangeOperation.java:295)
	at org.eclipse.ltk.internal.ui.refactoring.UIPerformChangeOperation.executeChange(UIPerformChangeOperation.java:94)
	at org.eclipse.ltk.core.refactoring.PerformChangeOperation.run(PerformChangeOperation.java:219)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2380)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2405)
	at org.eclipse.ltk.internal.ui.refactoring.WorkbenchRunnableAdapter.run(WorkbenchRunnableAdapter.java:89)
	at org.eclipse.jface.operation.ModalContext$ModalContextThread.run(ModalContext.java:122)

!ENTRY org.eclipse.jface 4 4 2023-05-25 00:57:25.869
!MESSAGE Rename Resource : An exception has been caught while processing the refactoring 'Rename Resource'.
```

I tried the same in a plain managed build project.
Strangely enough I did not get the stracktrace

further investigation lead me to this code in workspace class 
```
	public URI transferVariableDefinition(IResource source, IResource dest, URI sourceURI) throws CoreException {
		IPath srcLoc = source.getLocation();
		IPath srcRawLoc = source.getRawLocation();
		if ((srcLoc != null) && (srcRawLoc != null) && !srcLoc.equals(srcRawLoc)) {
			// the location is variable relative
			if (!source.getProject().equals(dest.getProject())) {
				String variable = srcRawLoc.segment(0);
				variable = copyVariable(source, dest, variable);
				IPath newLocation = Path.fromPortableString(variable).append(srcRawLoc.removeFirstSegments(1));
				sourceURI = toURI(newLocation);
			} else {
				sourceURI = toURI(srcRawLoc);
			}
		}
```
source is F/Sloeber_renamed/core/core
dest is F/Sloeber_renamed_again/core/core
sourceURI is file:/E:/platform/arduinoPlugin/packages/esp32/hardware/esp32/1.0.6/cores/esp32
srcRawLoc  becomes e:/platform/arduinoPlugin/packages/esp32/hardware/esp32/1.0.6/cores/esp32 which seems right as this is a esp32 project and the core is there
String variable = srcRawLoc.segment(0); makes variable project
[image]

Which then leads to 
```
String copyVariable(IResource source, IResource dest, String variable) throws CoreException {
		IPathVariableManager destPathVariableManager = dest.getPathVariableManager();
		IPathVariableManager srcPathVariableManager = source.getPathVariableManager();

		IPath srcValue = URIUtil.toPath(srcPathVariableManager.getURIValue(variable));
```
variable =project
srcPathVariableManager.getURIValue(variable) returns null resulting in  URIUtil.toPath complaining.

jantje, 2024-02-20:
Should be better now the underlying CDT issue is (partially) fixed

closed 2024-02-20 by jantje

dwittenb, 2024-12-31:
Hi, there is no hint how to Fix the issue? I have the same Problem still now. Have i to patch something?

jantje, 2025-01-02:
I have provided a partial(?)fix in CDT. So to have the fix you need to upgrade CDT.