Sloeber/arduino-eclipse-plugin #1568: Serial monitor: not readable when eclipse Dark theme is active

baudhuina, 2023-05-17:
**Environment**
Sloeber product on MacOS Ventura (M1)
Bug encountered and reproduced with a project name "aaa", on board "Adafruit Feather Huzza ESP8266".

**Description**
When Eclipse used the Dark theme (properties/General/appearance: Enable theming + Theme = Dark, the content of the Serial Monitor is in black on dark-grey which cannot be read. 

**How to reproduce**
In eclipse, properties/General/appearance: Enable theming and set Theme = Dark. Apply
Open the terminal monitor and connect to any board sending some text on the USB Serial interface. Text is black on dark grey. Other standard themes (Classic and Light) are ok. (see images)
[image]
[image]

jantje, 2023-05-17:
I don't know how these themes work but I do know you can change the text colour and font to anything you like
[image]

baudhuina, 2023-05-17:
Sure. I don’t know much about the themes either,except that:
- Font colors in CDT components (editor etc.) are not defined in Color and Fonts and  change with themes (I guess such components somehow obtain their colors from the theme).
- Font colors defined in the Appearance settings for the Serial Monitor are applied whatever the theme. Not many colors will be readable both on white and on black background. 
- I didn’t select a theme in Eclipse, so either Dark is the default, or is the default on MacOS, or it is selected based on OS display settings.. 

Thought I would register it for the long term: perhaps Sloeber can force a default theme in the product install which is compatible with its default color palette ? Or even some day manage colors like the CDT editors do ?  
Very minor minor issue, anyway.  

 <https://user-images.githubusercontent.com/1175841/239051472-619082e7-998f-4f2d-bd36-91d72a64920a.png>

jantje, 2023-05-17:
>Or even some day manage colors like the CDT editors do ?

How is that different to setting the colors in the preferences? https://github.com/Sloeber/arduino-eclipse-plugin/issues/1568#issuecomment-1551904212

baudhuina, 2023-05-18:
I’m just guessing here, but since CDT editor does NOT define the font colors in the preferences/Fonts & Colors (just the font size, see below), and the font color changes when the theme is changed, I assume the editor get’s the color to use from the current theme, not from the preferences.

This seems to be confirmed by https://help.eclipse.org/latest/index.jsp?topic=%2Forg.eclipse.platform.doc.isv%2Freference%2Fextension-points%2Forg_eclipse_ui_themes.html: 
API Information: 

The org.eclipse.ui.IWorkbench.getThemeManager() provides an instance of org.eclipse.ui.themes.IThemeManager that may be used to obtain a named theme (by id, including the default theme which has an id of IThemeManager.DEFAULT_THEME) or the current theme. From an org.eclipse.ui.themes.ITheme you may obtain a org.eclipse.jface.resources.ColorRegistry, an org.eclipse.jface.resources.FontRegistry and the arbitrary data associated with a theme.

￼

How is that different to setting the colors in the preferences? #1568 (comment) <https://github.com/Sloeber/arduino-eclipse-plugin/issues/1568#issuecomment-1551904212>

jantje, 2023-05-18:
My2cents
The way it works is: if you do not specify the font/colour then the "default" font colour is used.
As most gui classes are derived from others the default is the one of the first parent class that defines it.
So to change the font of the CDT editors you would change the font of one of the parent editors.
This allows you to change the font for all editors in one go. Which seems desired behaviour to me.

Sloeber can not use this method as Sloeber uses a different colour for each serial connection (maximum 6 connections)
So Sloeber serial monitor needs 6 colours where a "normal" editor only needs one.
To support "themes" I assume sloeber should define some xml stuff in the plugin.xml so that for each "sloeber supported theme" the colours are specified. As I seem to be pretty bad at picking colours I do not really believe it would be time well spent if I made this xml section.

Contributions are welcomed.

closed 2024-02-20 by jantje