Sloeber/arduino-eclipse-plugin #1570: serial monitor won'r release port

jonbarril, 2023-06-08:
I am having a devil of a time trying to use the serial monitor after anything happens to the arduino (e.g. program crashes) or USB connected to it (e.g. disconnect then reconnect the connector). Something in Sloeber is not releasing/resetting the port connection for serial monitor use. I can kill the serial monitor, then restart it, selecting the one available com port. The program will upload fine, as evidence from the console, but when the program in the arduino starts and tries to send a message to the serial monitor I get an error message from sloeber saying to select the correct port. There is only one port to select and it is already selected. There is no way I have found to release/reset the port for serial monitor use other than to restart eclipse. This is obviously an unacceptable solution.

As far as I know I am using the latest eclipse and sloeber plugins.

jantje, 2023-06-08:
It is hard for me to understand what you are trying to say.
A couple of things I learned with com ports, Sloeber and windows.
If the com port is "corrupted" disconnect the USB cable, disconnect in Sloeber serial monitor reconnect the USB cable.
Note that the Sloeber serial monitor can connect to multiple serial ports at the same time. You need to select the right port and disconnect.

If you can reproduce the problem. Can you document how with screenschots/video?

jonbarril, 2023-06-08:
Steps to reproduce this problem in eclipse (latest) running on Windows (latest) using the sloeber plugin (latest):

1. Open a serial monitor, connect it to the one available USB port.
2. Upload a program to arduino (Pro Micro). Upload succeeds.
3. The program runs and sends status messages to the serial monitor.
4. Disconnect the USB cable from the arduino.
5. Reconnect the USB cable.
6. Upload the same program to the arduino. Upload succeeds.
7. The program runs but shortly a sloeber error message appears indicating that the serial port (the one that used to work) cannot be found. (see attached)
8. In the serial port view, remove the port, then close the view.
9. Reopen the serial monitor view, add the one available port (same port as before).
10. Upload the same program to the arduino. Upload succeeds.
11. The program runs but shortly the same sloeber error message appears indicating that the serial port (the one that used to work) cannot be found. (see attached)  
12. (repeat 8 through 11 as often as you like)
13. Exit eclipse.
14. Restart eclipse.
15. Open a serial monitor view and connect to the one available USB port (the same as before).
16. Upload the same program as before to arduino. Upload succeeds.
17. The program runs and sends status messages to the serial monitor.

Instead of disconnecting the USB cable, the same problem occurs if the arduino program "crashes". As before, restarting Eclipse allows the serial monitor to use the the same port as before.

--jon
[image]

jantje, 2023-06-08:
At step 5 things are broken
If you disconnect the USB cable the com port is physically gone and the monitor holds the com number "locked"
So when the USB is reinserted Windows assigns another com port to the board.
Given your experiences I conclude that the special upload procedure the Pro micro (ATmega32U4)t uses still works regardless the locked port
Basically you need to disconnect the serial connection on the PC before you plug the USB back in.
Note that this is typical windows behaviour and not a specific Sloeber issue.

jonbarril, 2023-06-08:
I don't understand what you are suggesting. What do you mean by "disconnect the serial connection on the PC". Is this physically disconnecting the USB cable then re-inserting it into the PC? If so that is what I am doing. Should I be doing something else?

In Windows, the normal and expected behavior is for an app to release the port when the USB is disconnected from the PC, and to re-acquire a/the port when the USB is reconnected. It seems that sloeber is the only software I have used that won't let go of the USB port.

--jon

jantje, 2023-06-08:
>In Windows, the normal and expected behavior is for an app to release the port when the USB is disconnected from the PC, and to re-acquire a/the port when the USB is reconnected. It seems that sloeber is the only software I have used that won't let go of the USB port.

That may be the rule but for a very long time applications on windows were not alerted on windows when the serial connection breaks.
Sloeber is build on java which has very little support for serial. So I use a custom library which has to be compatible with Windows/linux/mac and tested with Sloeber on all these oses. (Lots of manual labour)
So I'm not looking for new/better versions.

jonbarril, 2023-06-08:
OK. Understood, but this is a real PITA when doing testing and such as restart of eclipse takes a while and breaks the work flow.
Is there some way for the serial monitor to force the port reset when the "remove port" red X is pressed on the serial view? Seems that even a crappy library should have something of this ilk.

BTW: For another project I am interested in accessing USB ports from Java. What library are you using? Perhaps I can do some research for you on how to release the USB port.

--jon

jonbarril, 2023-06-08:
FWIW: For my other project I am using library software by a company called Phidget to connect Java to their USB devices. Probably all proprietary but the USB ports act as they should on Windows. I can try asking them what they use if you are interested:-)

jantje, 2023-06-08:
There is no need to restart.
You need to disconnect in the serial monitor
disconnect usb reconnect usb

jonbarril, 2023-06-08:
As i described in my detailed steps to reproduce the problem, what you suggest definitely does not work.
--jon

jantje, 2023-06-09:
Have you tried disconnecting between step 3 and 4
The error you get at 7 is because comport 3 was in use by the serial monitor after step 5 so windows assigned another port
So Sloeber disconnects but can not reconnect and now you are in "what is connected to which port" which is very confusing and even more with the upload protocol used with the ATmega32U4

jonbarril, 2023-06-09:
Unfortunately the disconnect is by accident, and the same problem occurs if the arduino program crashes. Both of these use cases are unpredictable so disconnecting before ot happens is not an option.

Is there no way for slowber to gorce a disconnect of all current and padt ports? Sort of a nuclear option but still better than restarting eclipse.

jantje, 2023-06-09:
If you close the serial monitor view and the plotter view all serial connections should be removed from memory.
Note that the leonardo (also a ATmega32U4) is a very special beast as to com port usage and I had a very hard time to get it even to upload.
I'll have a look to see whether I can recreate the issue (with a leonardo as I do not own a Pro micro) but I'm currently focussed on something really big and exhausting so I can't spend time on this right now.