Sloeber/arduino-eclipse-plugin #1572: duplicate file tabs in Eclipse

jonbarril, 2023-06-15:
I'm still unclear how it happens or how to reproduce it, but it happens often enough to have caused me to loose file edits. It seems that the same project source file sometimes gets opened in different tabs in Eclipse. If there are bunch of tabs already open it goes unnoticed while I'm editing the file in one of the tabs. Then, trying to save the edits, eclipse complains that the file has been changed on the file system, and do you want to overwrite the changes. Looking through the tabs I realize there is a duplicate tab. Now it is a crap shoot as to what to do to save the edits. Half the time I guess wrong and loose the edits.

When this occurs it seems that both tabs point to the same file, but selecting one syncs the tree view, but selecting the other does not. I've only noticed this problem when using Sloeber, so I presume it is a Sloeber specific problem and not an Eclipse problem. 

In Eclipse, when not using Sloeber, opening a file that is already open in a tab opens and selects the already open tab -- Eclipse does not allow a duplicate tab to be opened. If you want to see a second view of an open file/tab you have to "split" the view.

--jon

jantje, 2023-06-16:
>I'm still unclear how it happens or how to reproduce it, but it happens often enough to have caused me to loose file edits. It seems that the same project source file sometimes gets opened in different tabs in Eclipse. 

This is caused by the file linking feature of eclipse.
Eclipse considers an eclipse resource (file project or folder) to be something the workspace points to. As such 2 eclipse files can point to the same physical location but are distinct from eclipse point of view.

Sloeber uses file linking intensively (to link to libraries and cores) and as such this behaviour is very visible in Sloeber (as soon as you start using multiple projects and editing libraries/cores).
To reproduce: create 2 projects for the same board.
In each project open the Arduino.h file.
You will have 2 file editors both pointing to the same Arduino.h. If you have "sync with project manager" on you can see eclipse will change the selection in the project manager when you change editor.

Once you are aware of this behaviour it will likely happen less. Things/hints to improve your experience:
1)Files that contain unsaved changes are marked with a *
[image]
2)Use save all (schift+ctrl+s) regularly. This will update all editors with your edits (or generate a error if you updated the same file in 2 editors).

jonbarril, 2023-06-17:
Hmmmm. What I have found is that, when this happens, only one of the tabs will sync with the tree view. Selecting the other tab does not. And, save all results in the "file has been changed on the file system" message, and now it is a crap shoot as to which is the real file containing the edits. Getting it wrong loses your edits.

So, taking this one step at a time. You say, I believe, that both tabs should sync the tree view. This is definitely not happening for me. Why is mine different? is it the same problem you describe, or a new/different one?

And, saving all file, or just each individual one, the "file changed" message occurs. In your response you seem not to have this problem. As before, am I seeing a new/different issue than the one you describe?

Thanks
--jon

jantje, 2023-06-17:
>As before, am I seeing a new/different issue than the one you describe?

Maybe. When things start making unexpected moves humans have a hard time seeing what actually is happening.
Understanding a case close to what you perceived may make it easier to reproduce your experience.
Lets hope you find a way to reproduce.

That being said. If you can reproduce I think it will turn out not to be a Sloeber issue. This because Sloeber feeds CDT/eclipse files and build commands but does not provide file editors or plays any role in opening closing saving.

jonbarril, 2023-06-17:
What you say about sloeber built on eclipse makes sense, which is why this is perplexing. I agree that my trying to consistently reproduce the problem is the next step.
Thanks
--jon

closed 2024-02-20 by jantje