Sloeber/arduino-eclipse-plugin #1574: Failed to upload to Teensy2 with two Teensy boards.

Raduq91, 2023-07-04:
I have two teensy boards connected to the pc, T4.0(as keyboard) and T2.0 .  [  Sloeber	4.4.1.202207231052	io.sloeber.product	null ]
When uploading with arduino ide it works to upload to T2, but when uploading with sloeber it fails to upload and I have to replug T4 for it to work again.

Arduino ide 1.8.19:
0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2.ino -path=C:\Users\Radu\AppData\Local\Temp\arduino_build_905193 -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=usb:0/140000/0/C -portlabel=hid#vid_16c0&pid_0478 Bootloader -portprotocol=Teensy 

Sloeber:
Launching: E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2_teensy2_0 -path=C:\Users\Radu\Documents\sloeber-workspace\433mhz_2_teensy2_0\Release -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=COM10 -portlabel= -portprotocol= 
Output:
Found 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
The execution of command "teensy/../tools/teensy_post_compile" is done.

Sloeber, wrote in the port field "usb:0/140000/0/C":
Launching: E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2_teensy2_0 -path=C:\Users\Radu\Documents\sloeber-workspace\433mhz_2_teensy2_0\Release -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=usb:0/140000/0/C -portlabel= -portprotocol= 
Output:
Found 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
The execution of command "teensy/../tools/teensy_post_compile" is done.

Sloeber, if only teens2.0 is connected uploading works:
Launching: E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2_teensy2_0 -path=C:\Users\Radu\Documents\sloeber-workspace\433mhz_2_teensy2_0\Release -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=COM10 -portlabel= -portprotocol= 
Output:
Teensy should be selected from "teensy ports" rather
than "Serial ports" in Arduino's Tools > Port menuThe execution of command "teensy/../tools/teensy_post_compile" is done.

[image]

Side note, add configurable feature:
When selecting in the 'opened files tab area' file.ino from project a, switch to project a. When selecting file.ino from project b, switch to project be.
So I don't have to do extra clicks to switch projects for upload when working with two file.ino opened from two projects, when doing increment changes.

jantje, 2023-07-04:
These commands Sloeber generates seem strange
IMHO you use the private hardware feature to point to a Teensy Install.
Have you tried with Teensy installed into Sloeber using the boards manager?

Raduq91, 2023-07-04:
> These commands Sloeber generates seem strange IMHO you use the private hardware feature to point to a Teensy Install. Have you tried with Teensy installed into Sloeber using the boards manager?

Yes private hardware path was used(removed it).
Now I added https://www.pjrc.com/teensy/package_teensy_index.json, selected teensy 1.58.1 from platform and boards.
I click upload and I get the same issue:

Launching: E:\Arduino_compile\Sloeber\arduinoPlugin\packages\teensy\tools\teensy-tools\1.58.0/teensy_post_compile -file=433mhz_2_teensy2_0 -path=C:\Users\Radu\Documents\sloeber-workspace\433mhz_2_teensy2_0\Release -tools=E:\Arduino_compile\Sloeber\arduinoPlugin\packages\teensy\tools\teensy-tools\1.58.0 -board=TEENSY2 -reboot -port=COM10 -portlabel= -portprotocol= 
Output:
Opening Teensy Loader...
Teensy should be selected from "teensy ports" rather
than "Serial ports" in Arduino's Tools > Port menuFound 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
The execution of command "1.58.0/teensy_post_compile" is done.

[image]

jantje, 2023-07-04:
Seems upload has changed a bit and Sloeber doesn't execute the correct command.
I say so because the com port was not used to upload in t"the old days" and now it seems to be part of the upload command.
I'm not sure why the port protocol is missing.
I'll take a look but I'm busy on other stuff right now.

Raduq91, 2023-07-04:
> Seems upload has changed a bit and Sloeber doesn't execute the correct command. I say so because the com port was not used to upload in t"the old days" and now it seems to be part of the upload [...]

Well in arduino ide I can select the teensy port: -port=usb:0/140000/0/C and it uploads without problem:
`0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2.ino -path=C:\Users\Radu\AppData\Local\Temp\arduino_build_905193 -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=usb:0/140000/0/C -portlabel=hid#vid_16c0&pid_0478 Bootloader -portprotocol=Teensy`

I now tried in arduino ide using to upload COM10 instead of teensy port, and it worked, no issue if both teensy boards are plugged in:

```
E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools/teensy_post_compile -file=433mhz_2.ino -path=C:\Users\Radu\AppData\Local\Temp\arduino_build_599706 -tools=E:\Arduino_compile\arduinoide4.0\hardware\teensy/../tools -board=TEENSY2 -reboot -port=COM10 -portlabel=COM10 -portprotocol=serial 
Teensy should be selected from "teensy ports" rather
than "Serial ports" in Arduino's Tools > Port menuFound 2 Teensy boards, but using auto-search to find board for upload.
Please use Tools > Ports(Teensy) to select the specific board.
```