Sloeber/arduino-eclipse-plugin #1577: Add notes about jexec and jspawnhelper to download page

Alex-Kent, 2023-08-02:
On the Linux download page the following is mentioned as needing to be run:

```
chmod +x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/bin/*
```

For e.g. upload programs to work the following is also needed:

```
$ chmod u+x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/lib/jexec 
$ chmod u+x plugins/org.eclipse.justj.openjdk.hotspot.jre.full.linux.x86_64_*/jre/lib/jspawnhelper 
```

Without doing this one gets the following when trying to upload:

```
java.io.IOException: Cannot run program "..." (in directory "..."): error=13, Permission denied
```

_Originally posted by @jcp-sd in https://github.com/Sloeber/arduino-eclipse-plugin/issues/1557#issuecomment-1518244651_

Alex-Kent, 2023-08-04:
FWIW I agree with @jcp-sd that tarballs should be used (instead of zip) for Linux.  Doing so would obviate the permissions issue.

jantje, 2023-08-05:
Hi
Thanks for the input.
As to the linux instructions on the website:
There used to be a build server who copied the website from github to my web server. Unfortunately I never go the build server to run after the last upgrade of the server :-(.
So I think I did the updates to the  website but they have not yet been published because I got spoiled by the build server auto copy.

As to the tarballs
Please feel free to provide a pull request. It should be a simple change here https://github.com/Sloeber/arduino-eclipse-plugin/blob/7596d4358344209aa720c1ac8042ea27464f6072/io.sloeber.product/pom.xml#L36

As I have no real linux experience I can't test and appreciate the change.

Alex-Kent, 2023-08-08:
I just put in a [pull request](https://github.com/Sloeber/arduino-eclipse-plugin/pull/1579).

closed 2023-08-10 by jantje